espressif/esp-idf #13990: 'MAX_RFC_PORTS' inaccessible for Bluetooth SPP API

LucasSaintrain, 2024-06-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm using the SPP profile with Bluedroid in my project.
There, I need the maximum number of serial ports allowed, to allocate a fixed-size array.

The constant I need is MAX_RFC_PORTS in the "common/bt_target.h" include header of the 'bt' component:
    /* The maximum number of ports supported. */
    #ifndef MAX_RFC_PORTS
    #define MAX_RFC_PORTS               8 /*max is 30*/
    #endif

Unfortunately, this header is a private include to the bt component, and so I'm unable to access it.
I see two possible solutions to the issue:
    - Adding the path "host/bluedroid/common/include" to the public includes in the cmake file.
    - Exposing the constant using the "esp_spp_api.c" and "esp_spp_api.h" files.

ESP-YTGerd, 2024-06-17:
Hi, @LucasSaintrain 
I suggest defining a constant with the same value as `MAX_RFC_PORTS` in the header or source files of your project. For now, we don't plan to change this value. If it becomes variable in the future, we will consider making it configurable and exposed.

LucasSaintrain, 2024-06-17:
Hi, @ESP-YTGerd,
Thanks for the suggestion. As you would expect, this is already the current implementation in my project, though it's nice to know this value won't be changed during development.
Thank you for the reply.

closed 2024-07-01 by LucasSaintrain