espressif/esp-idf #13995: ESP32C6 Variables Set by HP Core are 0 in LP Core

patrickslarson8, 2024-06-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.1

### Espressif SoC revision.

ESP32C6

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

Variables declared in LP Core and linked into HP core are then set by HP core and are available by LP core.

### What is the actual behavior?

Variables when read by LP core are all 0.

### Steps to reproduce.

## Steps
1. Build and flash
2. Monitor serial consoles. A FTDI is attached to the LP core to enable UART output for monitoring.

## Overview
The variables are declared and linked following the instructions here: https://docs.espressif.com/projects/esp-idf/en/stable/esp32c6/api-reference/system/ulp-lp-core.html#accessing-the-ulp-lp-core-program-variables
The HP sets the values at the variable addresses that are defined by some defines in a header file. After these are all set, the LP core is started and the HP core does not access these variables again (to prevent race conditions).

## Code snippet in LP core:
#### Variables declared
```C
volatile uint32_t shared_deadman_threshold;
```

#### bmi_160_register struct for reference
```C
typedef struct {
    uint8_t reg;
    uint8_t value;
} bmi_160_register;
```

#### Variable usage in LP core
```C
    lp_core_printf("dogvalue:%d, dogaddress:%d\n\r",
       (int) shared_deadman_threshold,
       (int) &shared_deadman_threshold);
```

## Code snippet in HP core
```C
    volatile uint32_t *shared_watchdog_threshold = &ulp_shared_deadman_threshold;
    *shared_watchdog_threshold = LP_WATCHDOG_THRESH;
    printf("dogaddress:%d, shareddogvalue:%d\n\r",
       (int) &ulp_shared_deadman_threshold,
       (int) ulp_shared_deadman_threshold);
```

### Debug Logs.

```plain
## Output
#### HP Core

dogaddress:1342183744, shareddogvalue:5128

#### LP Core
```terminal
dogvalue:0, dogaddress:1342183744
```
The address translates to 0x50001940, which is in the LP SRAM according to the technical manual (to be expected).
```

### More Information.

I have other variables and they are all reported as 0. The variable chosen to demonstrate is specifically a uint32_t to rule out any casting behavior.

SoucheSouche, 2024-06-17:
Hello @patrickslarson8, thanks for reporting this issue. We will take a look a tit a come back to you as soon as possible.

sudeep-mohanty, 2024-06-17:
Hello @patrickslarson8,
Thanks for approaching us with this query. I see that the variable `ulp_shared_deadman_threshold` is initialized by the HP core in the function `load_bmi_defaults()` _before_ the LP core is initialized. Once the LP core starts, it re-initializes the same variable `shared_deadman_threshold` to 0 since it is a `volatile` variable declared in global scope. Hence, you see this variable reported as 0 once the LP core starts to print. You should be able to see the value being set by the HP core if you set it _after_ the LP core is up and running.

patrickslarson8, 2024-06-17:
I see, so it is expected behavior then. Thanks for the help. I am closing the issue.

closed 2024-06-17 by patrickslarson8