espressif/esp-idf #13997: Registering mcpwm timer event callbacks after enable

askuric, 2024-06-17:
### Is your feature request related to a problem?

I was wondering what are the reasons behind the choice that the mcpwm timer event callbacks (i.e. function `mcpwm_timer_register_event_callbacks`) can be set only before calling `mcpwm_timer_enable`. 
I am asking because the earlier version of the mcpwm api (v4.x) did not have this constraint so I guess that the hardware timers are capable of initializing the interrupts even after the init.
Is there a hardware limitation that is addressed by this constraint or was this a convenience choice?

### Describe the solution you'd like.

Would you consider making this function callable after the timer enable, the line of code that would have to be removed is this one:
https://github.com/espressif/esp-idf/blob/8760e6d2a7e19913bc40675dd71f374bcd51b0ae/components/esp_driver_mcpwm/src/mcpwm_timer.c#L220

Then people could enable the callbacks after the init and call the `esp_intr_enable(timer)` once that's done.

### Describe alternatives you've considered.

An alternative could be to make the `mcpwm_timer_t` publicly available (not private declared in `mcpwm_private.h` - [here](https://github.com/espressif/esp-idf/blob/d7b0a45ddbddbac53afb4fc28168f9f9259dbb79/components/driver/mcpwm/mcpwm_private.h#L81-L95)). That way people who really need it could still do a hack like this:

```cpp
  // set the timer state to init (so that we can call the `mcpwm_timer_register_event_callbacks` )
  // this is a hack, as this function is not supposed to be called when the timer is running
  // the timer does not really go to the init state!
  timer->fsm = MCPWM_TIMER_FSM_INIT;
  // set the callback
  mcpwm_timer_register_event_callbacks(timer, &cbs, user_data);
  // set the timer state to enabled again
  timer->fsm = MCPWM_TIMER_FSM_ENABLE;
  esp_intr_enable(timer->isr);
```

### Additional context.

The issue of adding the timer callbacks after the timer enable are important for us at [simplefoc](https://github.com/simplefoc/Arduino-FOC) and we'll be happy to contribute if you're in for changing this.

suda-morris, 2024-06-17:
Only the first call of `mcpwm_timer_register_event_callbacks` should be before the enable function. 

The reason is, in MCPWM driver, the interrupt is lazy installed. If you don't register the event callback, no interrupt service will be installed. The enable function like `esp_intr_enable` will enable system-like services, e.g. interrupt, power management lock and so on. So if you didn't register any callback function and then call the enable function, the driver would think you don't want any feature offered by the interrupt service.

You can change the callback function to others later BTW. There's no restriction.

May I know why you wish to register the interrupt callback after the enable?

Or have you can even register empty callback function first. like

```c
mcpwm_timer_event_callbacks_t cbs = {}; // empty callback
    TEST_ESP_OK(mcpwm_timer_register_event_callbacks(timer, &cbs, NULL));
```
Then after mcpwm_timer_enable, you can again call this function to regsiter a real callback.

suda-morris, 2024-06-18:
Yes, it will get executed. If there's no user callback function being registered, `mcpwm_timer_default_isr` should exit quickly.