espressif/esp-idf #13999: DHCP server to support MAC filtering

grisharav, 2024-06-17:
### Is your feature request related to a problem?

I am using ESP32 module to have other devices connect to it, and I would like to allow only specific list of MAC addresses to get an IP address from the built-in DHCP server

### Describe the solution you'd like.

I would suggest an option in the dhcp server to optionally set a list of MAC addresses that will be checked before replying with an offer, if the requesting MAC isn't in that list, no offer shall be sent

### Describe alternatives you've considered.

Writing my own dhcp server based on the one provided at https://github.com/espressif/esp-idf/blob/89cb1d10d621266677ff1785f270e760ddd014a6/components/lwip/apps/dhcpserver/dhcpserver.c

grisharav, 2024-06-18:
It turns out it's not feasible to base the dhcp server on https://github.com/espressif/esp-idf/blob/master/components/lwip/apps/dhcpserver/dhcpserver.c since it uses the lwip API which is not accessible to user code according to https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_netif.html#esp-netif-architecture.
So I will either have to roll my own custom implementation of the entire server using the application level APIs or patch the existing dhcp server within the idf.

david-cermak, 2024-06-21:
Hi @grisharav 

I agree that the MAC filtering at DHCP server side would be a very useful feature. If you have your own implementation that you can share, you can post a PR in https://github.com/espressif/esp-idf/pulls or just point us to your code.

> it uses the lwip API which is not accessible to user code

The raw lwip API is still public; not recommended in general, but could be used, see https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/lwip.html#adapted-apis 
It's actually easier now, as you can use [CONFIG_LWIP_CHECK_THREAD_SAFETY](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/kconfig.html#config-lwip-check-thread-safety) option to test if you're using the raw API correctly.

grisharav, 2024-06-23:
Unfortunately I won't have time to implement this properly internally within IDF, it should be tested thoroughly.
It seems it's not simple to use the lwip API from user code, even including the right headers was not trvial when I tried (I am working using platformio...)