espressif/esp-idf #14001: `targets` section in `idf_component.yml` improperly registers component for invalid target

finger563, 2024-06-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2-163-gf81b598d10-dirty

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I can build the project for esp32, even if the components directory contains a component which does not support esp32, such as `tinyusb`. `tinyusb` is not in the `REQUIRES` section of the component.

### What is the actual behavior?

The build fails, complaining that `tinyusb` does not support target esp32.

[image]

### Steps to reproduce.

1. Add `tinyusb` to components directory
2. Make a simple project which points to the components directory and builds for esp32, but does not require or include any part of tinyusb.
3. Try to build, see failure.

### Build or installation Logs.

```plain
-- Building ESP-IDF components for target esp32
CMake Error at /Users/bob/esp/esp-idf/tools/cmake/build.cmake:552 (message):
  ERROR: Component "tinyusb" does not support target esp32

Call Stack (most recent call first):
  /Users/bob/esp/esp-idf/tools/cmake/project.cmake:605 (idf_build_process)
  CMakeLists.txt:20 (project)

-- Configuring incomplete, errors occurred!
```

### More Information.

This was working on 5.2.1 and before, but has broken on 5.2.2

finger563, 2024-06-17:
If I change the `targets` section to instead be:

```yaml
matches:
  - if: "target in [esp32s2, esp32s3, esp32p4]"
```

Then the builds work just fine.

sudeep-mohanty, 2024-06-18:
Hi @finger563, 
`tinyusb` is not supported on `esp32`. In case you want to use this component anyway, I suggest you follow the steps [here](https://components.espressif.com/components/espressif/tinyusb/versions/0.15.0~10). Then, you shouldn't run into an issue where the build system builds `tinyusb` for `esp32`. You could also refer to the examples [here](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device).

SoucheSouche, 2024-06-27:
@finger563, I am closing the issue. If you have any troubles with the proposed solution, fell free to reopen it.

closed 2024-06-27 by SoucheSouche

finger563, 2024-06-27:
We don't use the component registry, but instead use git submodules. In this case, we need to change the targets section as mentioned above, so I don't think the information you shared is a viable way around the problem.

sudeep-mohanty, 2024-07-09:
> We don't use the component registry, but instead use git submodules. In this case, we need to change the targets section as mentioned above, so I don't think the information you shared is a viable way around the problem.

Hello @finger563, Apologies for a delayed response. Unfortunately, TinyUSB isn't currently supported as a submodule in IDF. In case, you still would like to build your project for multiple targets and with TinyUSB as part of your project, they you would have to update the `idf_component.yml` for TinyUSB in your project.