espressif/esp-idf #14002: gptimer and nvs utilities do not get along: crash if timer occurs during some nvs calls

mbratch, 2024-06-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.1

### Espressif SoC revision.

ESP32-PICO-D4

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32 Dev Kit 1

### Power Supply used.

USB

### What is the expected behavior?

I would expect that the gptimer interrupt handlers, if very simple, would not cause a system failure or crash if they occur during calls the the NVS API.

### What is the actual behavior?

Crash can occur

### Steps to reproduce.

1. Start program which executes simple GPIO timer interrupts every 200us and with 20us intervals (see details in #13892)
2. Issue commands to system that could cause reads from NVS
3. Crash can occur

### Debug Logs.

```plain
Stack trace:

_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE8_M_eraseEjj
C:/Users/mdbra/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/basic_string.tcc:355
gptimer_default_isr
C:/Users/mdbra/.espressif/frameworks/esp-idf-v5.2.1/components/driver/gptimer/gptimer.c:543
_xt_lowint1
C:/Users/mdbra/.espressif/frameworks/esp-idf-v5.2.1/components/xtensa/xtensa_vectors.S:1240
cache_ll_l1_get_enabled_bus
C:/Users/mdbra/.espressif/frameworks/esp-idf-v5.2.1/components/hal/esp32/include/hal/cache_ll.h:192 (discriminator 4)
spi_flash_disable_cache
C:/Users/mdbra/.espressif/frameworks/esp-idf-v5.2.1/components/spi_flash/cache_utils.c:363
spi_flash_disable_interrupts_caches_and_other_cpu
C:/Users/mdbra/.espressif/frameworks/esp-idf-v5.2.1/components/spi_flash/cache_utils.c:209
cache_disable
C:/Users/mdbra/.espressif/frameworks/esp-idf-v5.2.1/components/spi_flash/spi_flash_os_func_app.c:70
spi1_start
C:/Users/mdbra/.espressif/frameworks/esp-idf-v5.2.1/components/spi_flash/spi_flash_os_func_app.c:122
spiflash_start_default
C:/Users/mdbra/.espressif/frameworks/esp-idf-v5.2.1/components/spi_flash/esp_flash_api.c:188
esp_flash_read
C:/Users/mdbra/.espressif/frameworks/esp-idf-v5.2.1/components/spi_flash/esp_flash_api.c:888
esp_partition_read
C:/Users/mdbra/.espressif/frameworks/esp-idf-v5.2.1/components/esp_partition/partition_target.c:38
_ZN3nvs12NVSPartition4readEjPvj
C:/Users/mdbra/.espressif/frameworks/esp-idf-v5.2.1/components/nvs_flash/src/nvs_partition.cpp:37
_ZNK3nvs4Page9readEntryEjRNS_4ItemE
C:/Users/mdbra/.espressif/frameworks/esp-idf-v5.2.1/components/nvs_flash/src/nvs_page.cpp:854
_ZN3nvs4Page8findItemEhNS_8ItemTypeEPKcRjRNS_4ItemEhNS_9VerOffsetE
C:/Users/mdbra/.espressif/frameworks/esp-idf-v5.2.1/components/nvs_flash/src/nvs_page.cpp:904
_ZN3nvs7Storage8findItemEhNS_8ItemTypeEPKcRPNS_4PageERNS_4ItemEhNS_9VerOffsetE
C:/Users/mdbra/.espressif/frameworks/esp-idf-v5.2.1/components/nvs_flash/src/nvs_storage.cpp:167 (discriminator 1)
_ZN3nvs7Storage9writeItemEhNS_8ItemTypeEPKcPKvj
C:/Users/mdbra/.espressif/frameworks/esp-idf-v5.2.1/components/nvs_flash/src/nvs_storage.cpp:293
_ZN3nvs15NVSHandleSimple8set_blobEPKcPKvj
C:/Users/mdbra/.espressif/frameworks/esp-idf-v5.2.1/components/nvs_flash/src/nvs_handle_simple.cpp:44
_ZN3nvs15NVSHandleLocked8set_blobEPKcPKvj
C:/Users/mdbra/.espressif/frameworks/esp-idf-v5.2.1/components/nvs_flash/src/nvs_handle_locked.cpp:24
_ZN13Configuration12saveNvsParamERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKN10AccessCard12accessCard_tE
C:/Users/mdbra/Documents/Projects/ESP32/BD-DTB/include/Configuration.hpp:70 (discriminator 2)
_ZN6Access8sendCardER10AccessCard
C:/Users/mdbra/Documents/Projects/ESP32/BD-DTB/main/Access.cpp:62
_ZN9WebServer13restSetAccessEP9httpd_req
C:/Users/mdbra/Documents/Projects/ESP32/BD-DTB/main/web_server/WebServerRestAccess.cpp:255
httpd_uri
C:/Users/mdbra/.espressif/frameworks/esp-idf-v5.2.1/components/esp_http_server/src/httpd_uri.c:329
httpd_parse_req
C:/Users/mdbra/.espressif/frameworks/esp-idf-v5.2.1/components/esp_http_server/src/httpd_parse.c:667
httpd_req_new
C:/Users/mdbra/.espressif/frameworks/esp-idf-v5.2.1/components/esp_http_server/src/httpd_parse.c:795
httpd_sess_process
C:/Users/mdbra/.espressif/frameworks/esp-idf-v5.2.1/components/esp_http_server/src/httpd_sess.c:426
httpd_process_session
C:/Users/mdbra/.espressif/frameworks/esp-idf-v5.2.1/components/esp_http_server/src/httpd_main.c:255
httpd_sess_enum
C:/Users/mdbra/.espressif/frameworks/esp-idf-v5.2.1/components/esp_http_server/src/httpd_sess.c:50 (discriminator 1)
httpd_server
C:/Users/mdbra/.espressif/frameworks/esp-idf-v5.2.1/components/esp_http_server/src/httpd_main.c:305
httpd_thread
C:/Users/mdbra/.espressif/frameworks/esp-idf-v5.2.1/components/esp_http_server/src/httpd_main.c:327
vPortTaskWrapper
C:/Users/mdbra/.espressif/frameworks/esp-idf-v5.2.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134
```
```

suda-morris, 2024-06-18:
> C:/Users/mdbra/.espressif/frameworks/esp-idf-v5.2.1/components/driver/gptimer/gptimer.c:543

```c
if (on_alarm_cb(timer, &edata, timer->user_ctx)) {
```

This implied the gptimer ISR is calling a user callback function while the cache is disabled. The root cause can be, your callback function is not in the IRAM, or the user context is not in the internal RAM.

mbratch, 2024-06-18:
Thank you @suda-morris . The details are in the other ticket I linked, but I do have the callback configured to be in IRAM.

My sdkconfig settings:
```
CONFIG_GPTIMER_ISR_HANDLER_IN_IRAM=y
CONFIG_GPTIMER_CTRL_FUNC_IN_IRAM=y
CONFIG_GPTIMER_ISR_IRAM_SAFE=y
```
I'll investigate the user context.

closed 2025-03-07 by Alvin1Zhang