espressif/esp-idf #14006: When compiling without builtin dhcp server there is a link error

grisharav, 2024-06-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Code compiles & links

### What is the actual behavior?

Code compiles, but doesn't link:
```
C:/Users/GregRavikovich/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio\build\gpt_pcb_gen1\esp-idf\esp_wifi\libesp_wifi.a(wifi_default.c.o):(.literal.esp_netif_create_default_wifi_mesh_netifs+0x0): undefined reference to `esp_netif_dhcps_stop'
C:/Users/GregRavikovich/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio\build\gpt_pcb_gen1\esp-idf\esp_wifi\libesp_wifi.a(wifi_default.c.o): in function `esp_wifi_set_default_wifi_ap_handlers':
C:/Users/GregRavikovich/.platformio/packages/framework-espidf/components/esp_wifi/src/wifi_default.c:263:(.text.esp_netif_create_default_wifi_mesh_netifs+0x53): undefined reference to `esp_netif_dhcps_stop'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\gpt_pcb_gen1\firmware.elf] Error 1
```

### Steps to reproduce.

Use the default wifi interface and don't configure DHCP Server in menuconfig:
```
#
# DHCP server
#
# CONFIG_LWIP_DHCPS is not set
```

### Build or installation Logs.

```plain
C:/Users/GregRavikovich/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio\build\gpt_pcb_gen1\esp-idf\esp_wifi\libesp_wifi.a(wifi_default.c.o):(.literal.esp_netif_create_default_wifi_mesh_netifs+0x0): undefined reference to `esp_netif_dhcps_stop'
C:/Users/GregRavikovich/.platformio/packages/toolchain-xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: .pio\build\gpt_pcb_gen1\esp-idf\esp_wifi\libesp_wifi.a(wifi_default.c.o): in function `esp_wifi_set_default_wifi_ap_handlers':
C:/Users/GregRavikovich/.platformio/packages/framework-espidf/components/esp_wifi/src/wifi_default.c:263:(.text.esp_netif_create_default_wifi_mesh_netifs+0x53): undefined reference to `esp_netif_dhcps_stop'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\gpt_pcb_gen1\firmware.elf] Error 1
```
```

### More Information.

I suspect this is due to https://github.com/espressif/esp-idf/blob/89cb1d10d621266677ff1785f270e760ddd014a6/components/esp_wifi/src/wifi_default.c#L469

david-cermak, 2024-06-21:
Are you trying to build this example: https://github.com/espressif/esp-idf/tree/master/examples/mesh/ip_internal_network or some project derived from that?
Or why would we need a definition of 

https://github.com/espressif/esp-idf/blob/0479494e7abe5aef71393fba2e184b3a78ea488f/components/esp_wifi/src/wifi_default.c#L449-L453

(or does platformio builds it with `WHOLE_ARCHIVE`?)

I'm able to reproduce the issue when building that mesh example, but the soft-ap builds and works okay without DHCP server (it would perhaps log an error if trying to use the *very* default config)

grisharav, 2024-06-21:
I am building a software we develop using mesh indeed.

david-cermak, 2024-06-21:
Oh, I see, this is really a bug. Thanks for reporting, will fix
(probably simply by enclosing the statement withing `#ifdef`s)
```patch
--- a/components/esp_wifi/src/wifi_default.c
+++ b/components/esp_wifi/src/wifi_default.c
@@ -480,8 +480,10 @@ esp_err_t esp_netif_create_default_wifi_mesh_netifs(esp_netif_t **p_netif_sta, e
     ESP_ERROR_CHECK(esp_netif_attach_wifi_station(netif_sta));
     ESP_ERROR_CHECK(esp_wifi_set_default_wifi_sta_handlers());
 
+#ifdef CONFIG_LWIP_DHCPS
     // ...and stop DHCP client (to be started separately if the station were promoted to root)
     ESP_ERROR_CHECK(esp_netif_dhcpc_stop(netif_sta));
+#endif
 
     if (p_netif_sta) {
```

grisharav, 2024-06-23:
Thanks, sounds like the right fix

closed 2024-07-04 by espressif-bot