espressif/esp-idf #14008: esp32s3 join AP failure if the AP channel is 1

AxelLin, 2024-06-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2-163-gf81b598d10

### Espressif SoC revision.

esp32s3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32S3-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

Join AP should work.

### What is the actual behavior?

Test with tp-link AX1800 AP, security: WPS2-PSK[AES].

When the AP 2.4G setting is "20/40MHz" with channel "auto", it was working fine.
However, after I switch channel "auto" to channel 1, the esp32s3 behavior abnormal.
ps. I also test esp32 and esp32 does not has such issue.

Below show the log to join AP , it keep retry if got disconnect, but always failure.

### Steps to reproduce.

Just make device join the AP.

### Debug Logs.

```plain
I (1985) pp: pp rom version: e7ae62f
I (1987) net80211: net80211 rom version: e7ae62f
I (1993) wifi:wifi driver task: 3fcbc0bc, prio:23, stack:6656, core=0
I (2002) wifi:wifi firmware version: 803d274
I (2002) wifi:wifi certification version: v7.0
I (2006) wifi:config NVS flash: enabled
I (2010) wifi:config nano formating: disabled
I (2014) wifi:Init data frame dynamic rx buffer num: 32
I (2019) wifi:Init static rx mgmt buffer num: 5
I (2023) wifi:Init management short buffer num: 32
I (2028) wifi:Init static tx buffer num: 16
I (2032) wifi:Init tx cache buffer num: 32
I (2035) wifi:Init static tx FG buffer num: 2
I (2039) wifi:Init static rx buffer size: 1600
I (2044) wifi:Init static rx buffer num: 16
[... 175 lines cut ...]
I (101885) wifi:state: init -> auth (b0)
I (101896) wifi:state: auth -> assoc (0)
I (102452) wifi:state: assoc -> run (10)
I (104572) wpa: EAPOL1 received for 4 times, sending deauth
I (104573) wifi:state: run -> init (f00)
I (104573) wifi:new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,1>, prof:1
I (104579) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (104584) wifi:state: init -> auth (b0)
I (105587) wifi:state: auth -> init (200)
I (105588) wifi:new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,1>, prof:1
I (105590) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (105593) wifi:state: init -> auth (b0)
I (106585) wifi:state: auth -> assoc (0)
I (106620) wifi:state: assoc -> run (10)
E (108725) wpa: Eapol message 4/4 tx failure, not installing keys
E (109249) wpa: Eapol message 4/4 tx failure, not installing keys
E (109774) wpa: Eapol message 4/4 tx failure, not installing keys
E (110298) wpa: Eapol message 4/4 tx failure, not installing keys
I (116621) wifi:state: run -> init (cc00)
I (116622) wifi:new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,1>, prof:1
I (116624) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (116627) wifi:state: init -> auth (b0)
I (117630) wifi:state: auth -> init (200)
I (117631) wifi:new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,1>, prof:1
```

AxelLin, 2024-06-19:
I can also observe the esp32s3 abnormal if I change AP from "20/40MHz" to "20MHz" and then switch back to "20/40MHz".
Sometimes it keep received "Disconnect reason: 34" after connected.
Sometimes it can join AP but the wifi transfer becomes slow and easy to hit failure.
This issue does not happen on esp32 when test at the same time.

below shows the "Disconnect reason: 34"  log
```

D (73695) wifi:Start wifi connect
D (73698) wifi:connect status 4 -> 0
D (73701) wifi:connect chan=0
D (73704) wifi:first chan=3
D (73706) wifi:connect status 0 -> 1
D (73709) wifi:filter: set rx policy=3
D (73713) wifi:clear scan ap list
D (73716) wifi:start scan: type=0x50f, priority=2, cb=0x42094e18, arg=0x0, ss_state=0x1, time=73721706, index=0
0x42094e18: cnx_start_handoff_cb at ??:?

D (73726) wifi:perform scan: ss_state=0x9, chan<3,0>, dur<0,120>
D (73762) wifi:rsn valid: gcipher=3 ucipher=3 akm=5 mac=ac:15:a2:4e:6b:47
D (73763) wifi:profile match: ss_state=0x7
D (73852) wifi:scan end: arg=0x0, status=0, ss_state=0x7
D (73852) wifi:perform scan: ss_state=0xd, chan<1,0>, dur<0,120>
D (73973) wifi:scan end: arg=0x0, status=0, ss_state=0x7
D (73974) wifi:perform scan: ss_state=0xd, chan<2,0>, dur<0,120>
D (74095) wifi:scan end: arg=0x0, status=0, ss_state=0x7
D (74095) wifi:perform scan: ss_state=0xd, chan<4,0>, dur<0,120>
D (74216) wifi:scan end: arg=0x0, status=0, ss_state=0x7
D (74217) wifi:perform scan: ss_state=0xd, chan<5,0>, dur<0,120>
D (74337) wifi:scan end: arg=0x0, status=0, ss_state=0x7
D (74338) wifi:perform scan: ss_state=0xd, chan<6,0>, dur<0,120>
D (74459) wifi:scan end: arg=0x0, status=0, ss_state=0x7
D (74459) wifi:perform scan: ss_state=0xd, chan<7,0>, dur<0,120>
D (74580) wifi:scan end: arg=0x0, status=0, ss_state=0x7
D (74581) wifi:perform scan: ss_state=0xd, chan<8,0>, dur<0,120>
D (74702) wifi:scan end: arg=0x0, status=0, ss_state=0x7
D (74702) wifi:perform scan: ss_state=0xd, chan<9,0>, dur<0,120>
D (74823) wifi:scan end: arg=0x0, status=0, ss_state=0x7
D (74824) wifi:perform scan: ss_state=0xd, chan<10,0>, dur<0,120>
D (74945) wifi:scan end: arg=0x0, status=0, ss_state=0x7
D (74945) wifi:perform scan: ss_state=0xd, chan<11,0>, dur<0,120>
D (75066) wifi:scan end: arg=0x0, status=0, ss_state=0x7
D (75067) wifi:perform scan: ss_state=0xd, chan<12,0>, dur<360,360>
D (75427) wifi:scan end: arg=0x0, status=0, ss_state=0x7
D (75428) wifi:perform scan: ss_state=0xd, chan<13,0>, dur<360,360>
D (75788) wifi:scan end: arg=0x0, status=0, ss_state=0x7
D (75789) wifi:perform scan: ss_state=0xd, chan<14,0>, dur<360,360>
D (76149) wifi:scan end: arg=0x0, status=0, ss_state=0x7
D (76150) wifi:filter: set rx policy=4
D (76150) wifi:first chan=1
D (76151) wifi:handoff_cb: status=0
D (76153) wifi:ap found, mac=ac:15:a2:4e:6b:47
D (76157) wifi:going for connection with bssid=ac:15:a2:4e:6b:47
D (76163) wifi:new_bss=0x3fca34e0, cur_bss=0x0, new_chan=<3,1>, cur_chan=3
D (76169) wifi:filter: set rx policy=5
D (76173) wpa: WPA: set AP RSNXE - hexdump(len=0):
D (76178) wpa:  [NULL]
I (76180) wifi:new:<3,1>, old:<3,0>, ap:<255,255>, sta:<3,1>, prof:1
D (76187) wifi:connect_op: status=0, auth=5, cipher=3 
D (76191) wifi:auth mode is not none
D (76194) wifi:connect_bss: auth=1, reconnect=0
I (76199) wifi:state: init -> auth (b0)
D (76202) wifi:start 1s AUTH timer
D (76205) wifi:clear scan ap list
D (76208) event: running post WIFI_EVENT:41 with handler 0x420114c4 and context 0x3fcba3c4 on loop 0x3fcb38dc

I (76218) net_utils: WIFI_EVENT_HOME_CHANNEL_CHANGE: old_chan=3/old_snd=0, new_chan=3/new_snd=1
D (76216) wifi:recv auth: seq=2, status=0
I (76231) wifi:state: auth -> assoc (0)
D (76235) wifi:restart connect 1s timer for assoc
[... 1603 lines cut ...]
D (111337) esp_netif_lwip: esp_netif_dhcpc_start_api esp_netif:0x3fcba290
D (111344) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3fcba290
D (111351) esp_netif_lwip: if0x3fcba290 start ip lost tmr: already started
D (111358) esp_netif_lwip: starting dhcp client
D (111363) wifi:D (111363) esp_netif_lwip: call api in lwip: ret=0x0, give sem
eb is dhcp or dns sport = 68, dport = 67
D (111373) event: running post WIFI_EVENT:4 with handler 0x420114c4 and context 0x3fcba3c4 on loop 0x3fcb38dc

I (111383) net_utils: STA connected
I (111619) wifi:<ba-add>idx:0 (ifx:0, ac:15:a2:4e:6b:47), tid:7, ssn:0, winSize:64
D (111863) wifi:eb is dhcp or dns sport = 68, dport = 67
I (111871) wifi:<ba-del>idx:0, tid:7
I (111998) wifi:<ba-add>idx:0 (ifx:0, ac:15:a2:4e:6b:47), tid:7, ssn:0, winSize:64
D (112863) wifi:eb is dhcp or dns sport = 68, dport = 67
I (112869) wifi:<ba-del>idx:0, tid:7
I (113015) wifi:<ba-add>idx:0 (ifx:0, ac:15:a2:4e:6b:47), tid:7, ssn:0, winSize:64
D (114863) wifi:eb is dhcp or dns sport = 68, dport = 67
I (114869) wifi:<ba-del>idx:0, tid:7
I (115021) wifi:<ba-add>idx:0 (ifx:0, ac:15:a2:4e:6b:47), tid:7, ssn:0, winSize:64
I (118483) wifi:<ba-del>idx:0, tid:7
I (118489) wifi:<ba-add>idx:0 (ifx:0, ac:15:a2:4e:6b:47), tid:7, ssn:0, winSize:64
D (118863) wifi:eb is dhcp or dns sport = 68, dport = 67
D (122863) wifi:eb is dhcp or dns sport = 68, dport = 67

```

AxelLin, 2024-06-19:
I can reproduce only when the AP's RSSI is -5X ~ -7X.
If I move the device closer to the AP, (RSSI is -2X), I cannot reproduce this issue.
However, the question is ESP32 works fine even in all my test cases.

MaxwellAlan, 2024-06-19:
Hi @AxelLin 

> below shows the "Disconnect reason: 34" log

This means AP can not receive S3 reply 802.11 acks, also from below comments:

>I can reproduce only when the AP's RSSI is -5X ~ -7X.
If I move the device closer to the AP, (RSSI is -2X), I cannot reproduce this issue.
However, the question is ESP32 works fine in all my test cases.

seems this issue related with S3 tx power is abnormal.

Could you please upload the sdkconfig?
And you can try below config before test:
1. idf.py erase-flash // erease s3 flash
2. re-test

Or:

1. disable Component config -> PHY -> Store phy calibration data in NVS
2. re-test

Thanks.

AxelLin, 2024-06-19:
Hi @MaxwellAlan 

I have much less chance to see "Disconnect reason: 34" after setting CONFIG_ESP_PHY_CALIBRATION_AND_DATA_STORAGE=n,
(I just found disconnect reason 34 once , see below log)

However, it may take quite long time to upload a 40KB file (sometimes it took 20 ~ 50 sec), the AP RSSI is -67.
Sometimes I got below failure (RSSI is -68):
W (46979) esp-tls: Failed to open new connection in specified timeout
E (46979) transport_base: Failed to open a new connection
Or below failure:
E (41298) esp-tls: [sock=49] select() timeout
E (41299) esp-tls: Failed to open new connection
E (41299) transport_base: Failed to open a new connection
E (41303) HTTP_CLIENT: Connection failed, sock < 0

But the ESP32 test is stable and just needs less than 5 sec when test at the same place.

Below is the long when I got disconnect reason 34, (just once).
```
I (41072) wifi:<ba-add>idx:0 (ifx:0, ac:15:a2:4e:6b:47), tid:7, ssn:0, winSize:64
D (41551) wifi:eb is dhcp or dns sport = 68, dport = 67
D (42551) wifi:eb is dhcp or dns sport = 68, dport = 67
D (44074) wifi:recv deauth, reason=0x22
I (44074) wifi:state: run -> init (22c0)
D (44075) wifi:stop beacon and connect timers
D (44075) wifi:recv deauth/disassoc
D (44078) wifi:connect status 5 -> 4
D (44081) wifi:sta leave
I (44083) wifi:pm stop, total sleep time: 0 us / 3143159 us

D (44089) wifi:stop CSA timer
I (44091) wifi:<ba-del>idx:0, tid:7
D (44095) wifi:remove ac:15:a2:4e:6b:47 from rc list
I (44099) wifi:new:<2,0>, old:<2,1>, ap:<255,255>, sta:<2,1>, prof:1
D (44106) wifi:filter: set rx policy=8
D (44109) wifi:Send disconnect event, reason=34, AP number=0
D (44114) event: running post WIFI_EVENT:41 with handler 0x420114c4 and context 0x3fcba3c4 on loop 0x3fcb38dc
0x420114c4: net_utils_wifi_event_handler at /home/axel/esp/esp-idf-dev/components/ingics/net_utils/net_utils.c:544

I (44124) net_utils: WIFI_EVENT_HOME_CHANNEL_CHANGE: old_chan=2/old_snd=1, new_chan=2/new_snd=0
D (44134) event: running post WIFI_EVENT:5 with handler 0x4205d770 and context 0x3fcba19c on loop 0x3fcb38dc
0x4205d770: wifi_default_action_sta_disconnected at /home/axel/esp/esp-idf/components/esp_wifi/src/wifi_default.c:104

D (44144) esp_netif_handlers: esp_netif action disconnected with netif0x3fcba290 from event_id=5
D (44152) esp_netif_lwip: check: remote, if=0x3fcba290 fn=0x4203b8f8
0x4203b8f8: esp_netif_down_api at /home/axel/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1674

D (44159) esp_netif_lwip: esp_netif_down_api esp_netif:0x3fcba290
D (44165) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3fcba290
D (44172) esp_netif_lwip: if0x3fcba290 start ip lost tmr: already started
D (44179) esp_netif_lwip: check: local, if=0x3fcba290 fn=0x4203be60
0x4203be60: esp_netif_update_default_netif_lwip at /home/axel/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:317

D (44185) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3fcba290
D (44192) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (44198) event: running post WIFI_EVENT:5 with handler 0x420114c4 and context 0x3fcba3c4 on loop 0x3fcb38dc
0x420114c4: net_utils_wifi_event_handler at /home/axel/esp/esp-idf-dev/components/ingics/net_utils/net_utils.c:544

I (44208) net_utils: Disconnect reason: 34, rssi=-70

```

MaxwellAlan, 2024-06-19:
Hi @AxelLin 

>I have much less chance to see "Disconnect reason: 34" after setting CONFIG_ESP_PHY_CALIBRATION_AND_DATA_STORAGE=n,
(I just found disconnect reason 34 once , see below log)

seems PHY recalibration can make TX power to normal, but there still have some issues, maybe related with Bandwidth, did you test with AP bandwidth HT20-only and HT40-only to check whether HT40 issue.
PS: did you test with other S3 boards?

AxelLin, 2024-06-20:
We have observed several wifi issue with S3, but not every device is debug build.
So at the moment, I'm not sure if this is a single case or other boards have the same issue.
I will update to you when I have more test result.
But overall, when the AP's rssi is < -6X, the S3 (ESP32S3-WROOM-1 module) has very high chance fail to upload data.
e.g.
W (49531) transport_base: Poll timeout or error, errno=Connection already in progress, fd=49, timeout_ms=15000
This is very strange because I think -6X is not very weak.

Also it looks like using esp_camera has impact on wifi transfer.

AxelLin, 2024-06-20:
BTW, is there any API I can use to check if TX power is good or not?

AxelLin, 2024-06-20:
I notice ESP32S3 may switch channel multiple times when join an AP.

```
I (2100) net_utils: Joining AP: IGS-MESH
I (2102) phy_init: phy_version 670,b7bc9b9,Apr 30 2024,10:54:13
I (2137) wifi:mode : sta (dc:da:0c:7c:d9:0c)
I (2138) wifi:enable tsf
I (2138) net_utils: WIFI_EVENT_HOME_CHANNEL_CHANGE: old_chan=0/old_snd=0, new_chan=1/new_snd=0
I (2143) net_utils: STA started
I (4919) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (4920) wifi:state: init -> auth (b0)
I (4921) net_utils: WIFI_EVENT_HOME_CHANNEL_CHANGE: old_chan=1/old_snd=0, new_chan=1/new_snd=1
I (4937) wifi:state: auth -> assoc (0)
I (4949) wifi:state: assoc -> run (10)
I (7958) wpa: EAPOL1 received for 4 times, sending deauth
I (7959) wifi:state: run -> init (f00)
I (7960) wifi:new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,1>, prof:1
I (7964) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (7970) wifi:state: init -> auth (b0)
I (7973) net_utils: WIFI_EVENT_HOME_CHANNEL_CHANGE: old_chan=1/old_snd=1, new_chan=1/new_snd=0
I (7982) net_utils: WIFI_EVENT_HOME_CHANNEL_CHANGE: old_chan=1/old_snd=0, new_chan=1/new_snd=1
I (7996) wifi:state: auth -> assoc (0)
I (8028) wifi:state: assoc -> run (10)
I (11038) wpa: EAPOL1 received for 4 times, sending deauth
I (11039) wifi:state: run -> init (f00)
I (11039) wifi:new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,1>, prof:1
I (11045) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (11050) wifi:state: init -> auth (b0)
I (11054) net_utils: WIFI_EVENT_HOME_CHANNEL_CHANGE: old_chan=1/old_snd=1, new_chan=1/new_snd=0
I (11062) net_utils: WIFI_EVENT_HOME_CHANNEL_CHANGE: old_chan=1/old_snd=0, new_chan=1/new_snd=1
I (11071) wifi:state: auth -> assoc (0)
I (12076) wifi:state: assoc -> init (2700)
I (12077) wifi:new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,1>, prof:1
I (12078) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (12082) wifi:state: init -> auth (b0)
I (12085) net_utils: WIFI_EVENT_HOME_CHANNEL_CHANGE: old_chan=1/old_snd=1, new_chan=1/new_snd=0
I (12094) net_utils: WIFI_EVENT_HOME_CHANNEL_CHANGE: old_chan=1/old_snd=0, new_chan=1/new_snd=1
I (13085) wifi:state: auth -> init (200)
I (13086) wifi:new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,1>, prof:1
I (13087) wifi:new:<8,0>, old:<1,0>, ap:<255,255>, sta:<8,0>, prof:1
I (13091) wifi:state: init -> auth (b0)
I (13094) net_utils: WIFI_EVENT_HOME_CHANNEL_CHANGE: old_chan=1/old_snd=1, new_chan=1/new_snd=0
I (13098) wifi:state: auth -> assoc (0)
I (13107) net_utils: WIFI_EVENT_HOME_CHANNEL_CHANGE: old_chan=1/old_snd=0, new_chan=8/new_snd=0
I (13113) wifi:state: assoc -> run (10)
I (13134) wifi:connected with IGS-MESH, aid = 2, channel 8, BW20, bssid = ac:15:a2:4e:6b:47
I (13135) wifi:security: WPA2-PSK, phy: bgn, rssi: -70
I (13136) wifi:pm start, type: 1

I (13139) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (13148) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 25000, mt_pti: 0, mt_time: 10000
I (13156) wifi:AP's beacon interval = 102400 us, DTIM period = 1

```

Below is ESP32 log for comparison:
```
I (1958) phy_init: phy_version 4791,2c4672b,Dec 20 2023,16:06:06
I (2027) wifi:mode : sta (08:d1:f9:ca:35:f4)
I (2028) wifi:enable tsf
I (2028) net_utils: WIFI_EVENT_HOME_CHANNEL_CHANGE: old_chan=0/old_snd=0, new_chan=1/new_snd=0
I (2033) net_utils: STA started
I (4609) wifi:new:<8,0>, old:<1,0>, ap:<255,255>, sta:<8,0>, prof:1
I (4610) wifi:state: init -> auth (b0)
I (4611) net_utils: WIFI_EVENT_HOME_CHANNEL_CHANGE: old_chan=1/old_snd=0, new_chan=8/new_snd=0
I (4622) wifi:state: auth -> assoc (0)
I (4629) wifi:state: assoc -> run (10)
I (4644) wifi:connected with IGS-MESH, aid = 2, channel 8, BW20, bssid = ac:15:a2:4e:6b:47
I (4644) wifi:security: WPA2-PSK, phy: bgn, rssi: -62
I (4646) wifi:pm start, type: 1

I (4649) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (4663) wifi:<ba-add>idx:0 (ifx:0, ac:15:a2:4e:6b:47), tid:7, ssn:0, winSize:64
I (4720) wifi:AP's beacon interval = 102400 us, DTIM period = 1

```

AxelLin, 2024-06-21:
I also observed below symptom when test with freenove s3 dev board:
 1. Sometimes I got "disconnect reason:34" or "39"
 2. If the AP's rssi is week, I can observe the upload failure.
    If move the device closer to AP, it works fine.
    However, even in the upload success case, the upload time is much longer than esp32.

 ping the device:
 $ ping 192.168.0.160
PING 192.168.0.160 (192.168.0.160) 56(84) bytes of data.
64 bytes from 192.168.0.160: icmp_seq=1 ttl=64 time=22.6 ms
64 bytes from 192.168.0.160: icmp_seq=6 ttl=64 time=1628 ms
64 bytes from 192.168.0.160: icmp_seq=13 ttl=64 time=1863 ms
64 bytes from 192.168.0.160: icmp_seq=14 ttl=64 time=839 ms
64 bytes from 192.168.0.160: icmp_seq=23 ttl=64 time=768 ms
64 bytes from 192.168.0.160: icmp_seq=24 ttl=64 time=1408 ms
64 bytes from 192.168.0.160: icmp_seq=25 ttl=64 time=391 ms

AxelLin, 2024-06-21:
@MaxwellAlan 
Is this a known issue for esp32s3? https://www.esp32.com/viewtopic.php?f=2&t=40111&sid=283a77286dd295dcfdd896e2c0d6e3f2#p132803

hansw123, 2024-06-21:
hi @AxelLin 
we do not have tp-link AX1800 ,could you provide the capture in channel  1 ，thank you very much !

AxelLin, 2024-06-21:
@hansw123 
Please check attached log.

The AP mac: ac:15:a2:4e:6b:47
ESP32S3 mac: dc:da:0c:7c:d9:0c
[bug_report.pcapng.txt](https://github.com/user-attachments/files/15925501/bug_report.pcapng.txt)

hansw123, 2024-06-24:
@AxelLin 
do you use  USB , you can disable USB module and retest ,thank you !

AxelLin, 2024-06-24:
> @AxelLin do you use USB , you can disable USB module and retest ,thank you !

@hansw123 
What's your finding in the wireshark log? Any thing unusual?

According to this comment https://www.esp32.com/viewtopic.php?f=2&t=40111&sid=283a77286dd295dcfdd896e2c0d6e3f2#p133443 , the impact of usb is very limited.
So why you suggest to disable usb usage?

hansw123, 2024-06-24:
@AxelLin 
we find the station recv ack from ap may loss , so the eapol process sometimes will tinmeout . 
We're guessing that it's probably due to the influence of the USB

AxelLin, 2024-06-25:
@hansw123 @MaxwellAlan 

As I said in prvious comment, your colleague says the impact of USB is limited.
Howver, in my test, this issue seems happens *after* I **init camera**.
See below log:

I (2345) net_utils: Joining AP: IGS-MESH
I (2348) phy_init: phy_version 670,b7bc9b9,Apr 30 2024,10:54:13
I (2390) wifi:mode : sta (dc:da:0c:7c:d9:0c)
I (2390) wifi:enable tsf
I (2391) net_utils: WIFI_EVENT_HOME_CHANNEL_CHANGE: old_chan=0/old_snd=0, new_chan=1/new_snd=0
I (2395) net_utils: STA started
I (5168) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (5169) wifi:state: init -> auth (b0)
I (5169) net_utils: WIFI_EVENT_HOME_CHANNEL_CHANGE: old_chan=1/old_snd=0, new_chan=1/new_snd=1
I (5171) wifi:state: auth -> init (8a0)
I (5179) wifi:new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,1>, prof:1
I (5187) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (5192) wifi:state: init -> auth (b0)
I (5195) net_utils: WIFI_EVENT_HOME_CHANNEL_CHANGE: old_chan=1/old_snd=1, new_chan=1/new_snd=0
I (5199) wifi:state: auth -> assoc (0)
I (5208) net_utils: WIFI_EVENT_HOME_CHANNEL_CHANGE: old_chan=1/old_snd=0, new_chan=1/new_snd=1
I (5214) wifi:state: assoc -> run (10)
I (5242) wifi:connected with IGS-MESH, aid = 2, channel 1, 40U, bssid = ac:15:a2:4e:6b:47
I (5242) wifi:security: WPA2-PSK, phy: bgn, rssi: -72
I (5244) wifi:pm start, type: 1

I (5247) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (5255) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 25000, mt_pti: 0, mt_time: 10000
I (5264) net_utils: STA connected
I (5272) wifi:<ba-add>idx:0 (ifx:0, ac:15:a2:4e:6b:47), tid:7, ssn:0, winSize:64
I (5339) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (6264) esp_netif_handlers: sta ip: 192.168.0.244, mask: 255.255.255.0, gw: 192.168.0.1
I (6339) wifi:<ba-add>idx:1 (ifx:0, ac:15:a2:4e:6b:47), tid:0, ssn:0, winSize:64
**// Start init esp_camera here**
I (16041) wifi:state: run -> init (22c0)
I (16043) wifi:pm stop, total sleep time: 5845865 us / 10795879 us

I (16043) wifi:<ba-del>idx:1, tid:0
I (16045) wifi:<ba-del>idx:0, tid:7
I (16048) wifi:new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,1>, prof:1
I (16055) net_utils: WIFI_EVENT_HOME_CHANNEL_CHANGE: old_chan=1/old_snd=1, new_chan=1/new_snd=0
I (16064) net_utils: Disconnect reason: 34, rssi=-73

Could you help to check if this is related to esp_camera usage? (This happens after lcd_camera_controller is enabled)
Also please check my feedback here: https://github.com/espressif/esp32-camera/issues/662#issuecomment-2182278303

AxelLin, 2024-06-26:
@hansw123 @MaxwellAlan 

Attached 2 wireshark log that uploading data to http server: 1 ok case, 1 fail case.
S3 has much higher chance to hit the failure case.

[s3-http-ok1.pcapng.txt](https://github.com/user-attachments/files/15981920/s3-http-ok1.pcapng.txt)
[s3-http-fail1.pcapng.txt](https://github.com/user-attachments/files/15981921/s3-http-fail1.pcapng.txt)

hansw123, 2024-06-28:
@AxelLin 
Can you provide the model number and example code of the camera development board you are using? do ESP32-s3-EYE

hansw123, 2024-07-04:
@AxelLin 
please check your  camera task  priorities, if  MAXpriorities -2 (which == wifi task) ,maybe cause the issue

AxelLin, 2024-07-05:
> @AxelLin please check your camera task priorities, if MAXpriorities -2 (which == wifi task) ,maybe cause the issue

If the cam_task task priority is a concern, you should talk to the esp32-cam maintainer @me-no-dev 
because the cam_task task is not configurable.
However, even if the task priority is configurable, how to decide which priority is a proper priority?

BTW, I tried lower cam_task priority by modifying esp32-cam code, which seems not related to this issue.
But this issue might be related to the interference caused by esp32-cam usage.

hansw123, 2024-07-05:
@AxelLin 
thanks for your test ,but we can not reproduce  the issue ,so could you provide the example code and the Specific hardware models used

AxelLin, 2024-07-05:
@hansw123 

The interference between CPU, camera and antenna seems to be a common issue on esp32/esp32s3 as it has been reported by several users.

see
https://github.com/espressif/arduino-esp32/issues/4655
https://github.com/espressif/arduino-esp32/issues/5834

It would be helpful if espressif can provide some s/w and h/w design guideline
to avoid such issue. (https://github.com/espressif/esp32-camera/issues/662#issuecomment-2204966252)

Zmmfly, 2024-08-29:
@AxelLin @MaxwellAlan @hansw123 @methodmissing 
Hello everyone, I may have the same problem in `IDFv5.3` with `ESP32S3`; Here is a brief description of the operating environment:
1. `BLE` is turned on and `scanning` is in progress, with a cycle of 1000ms and a window of 100ms
2. `LVGL` is running and using the `RGB screen`, and the buffer is located in PSRAM
3. `WiFi & LWIP` are turned on to `allocate memory to PSRAM first`
4. A `4MBytes font` file is loaded `in PSRAM`
5. `XIP` in PSRAM is turned on
6. Both the `bss` segment and the `noinit` segment are placed `in PSRAM`

In addition to these, I tried to run the station example. Except that Bluetooth and the RGB screen are completely disabled, other conditions are the same, and the example can successfully connect to the AP.

I hope the above information can help you locate the problem.

Zmmfly, 2024-08-29:
I tried to reorder of WiFi configuration calls, and it worked.

The following is the current call order:

- Call `esp_wifi_init` during initialization
- If `esp_wifi_start` has been called, call `esp_wifi_stop` to stop WiFi first
- Call `esp_wifi_set_mode` to set the mode
- Call `esp_wifi_set_config` to set the STA configuration
- Call `esp_wifi_start` to start WiFi
- Call `esp_wifi_connect` in the event `WIFI_EVENT_STA_START`
- Successfully connected to STA and obtained IP

AxelLin, 2024-08-29:
@Zmmfly

AxelLin, 2024-08-29:
@Zmmfly 
For comparison, could you also show your WiFi configuration calls that was not working?

Nezaemmy, 2024-08-30:
I have the same problem in IDFv5.1 with ESP32S3. esp32s3 takes time to connect to the internet while esp32 doesn't have this issue

AxelLin, 2024-08-30:
> I have the same problem in IDFv5.1 with ESP32S3. esp32s3 takes time to connect to the internet while esp32 doesn't have this issue

@Nezaemmy 
Could you enable CONFIG_ESP_WIFI_DEBUG_PRINT=y, and share the device boot log?

AxelLin, 2025-02-14:
@hansw123 @MaxwellAlan 
This is marked as resolution done since Jan 10, what was fixed?

hansw123, 2025-02-14:
@AxelLin 
we can not reproduce the issue and the github issue have no response a long time

closed 2026-03-01 by AxelLin