espressif/esp-idf #14014: [gatt_server] Cannot set bigger packet size when communicating with Linux client

lucic71, 2024-06-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I run gatt_server example on a esp32c3 and from Debian 12 I connect to it using gatttool which is using version 5.66-1+deb12u1 of bluez. My final goal is to send 2MB of data over BLE, however this is quite slow because only sending a packet worth 1024 bytes using gatttool takes around 1.3 seconds. That means sending 2MB would require around 45 minutes.

To send the data I use the following command:
```sh
(export PKT_SZ=1024; gatttool -b 34:85:18:F7:F7:3A --mtu 500 --char-write-req -a 0x002a -n $(printf "ff%.0s" `seq 1 $PKT_SZ`))
```

To make sure that the MTU gatttool is trying to negotiate is accepted, I also added a cal to ``esp_ble_gatts_send_response(gatts_if, param->mtu.conn_id, ESP_GATT_OK, 0, NULL);`` inside the ESP_GATTS_MTU_EVT event.

To make this faster I thought on using the packet length extension functionality of BLE which should get me at most 251 bytes of data in a packet instead of 20, which is the default. I tried to increase the size using a call to ``esp_ble_gap_set_pkt_data_len(param->connect.remote_bda, 251)`` inside ESP_GATTS_CONNECT_EVT event handler.

However, that does not seem to work. When I send a 21 byte  packet using the following command:
```sh
(export PKT_SZ=21; gatttool -b 34:85:18:F7:F7:3A --mtu 500 --char-write-req -a 0x002a -n $(printf "ff%.0s" `seq 1 $PKT_SZ`))
```

I get the following output from gatt_server:
```
I (3414311) GATTS_DEMO: GATT_WRITE_EVT, conn_id 0, trans_id 1, handle 42
I (3414341) GATTS_DEMO: GATT_WRITE_EVT, conn_id 0, trans_id 2, handle 42
I (3414351) GATTS_DEMO: ESP_GATTS_EXEC_WRITE_EVT
I (3414351) GATTS_DEMO: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
I (3414351) GATTS_DEMO: ff ff ff ff ff 
```

Compared to sending a 20 byte packet which looks like this:
```
I (3423851) GATTS_DEMO: GATT_WRITE_EVT, conn_id 0, trans_id 1, handle 42
I (3423851) GATTS_DEMO: GATT_WRITE_EVT, A value len 20, value :
I (3423861) GATTS_DEMO: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
I (3423871) GATTS_DEMO: ff ff ff ff 
```

It looks like the 21 byte packet is split in two packets and the 20 byte packet is not, which indicates that the call to esp_ble_gap_set_pkt_data_len had no effect at all. I also see a lot of GATT_WRITE_EVT when writing the 1024 bytes packet, when in fact there should be just 5-6 of them (1024/251).

In summary, the BLE stack inside esp-idf does not seem able to use the packet length extension functionality. Am I doing something wrong or is this a known limitation?

eriksl, 2024-06-18:
My $0.02 here. It appeared to me that when using Nimble (what stack are you using?) you can stretch the MTU to something like 2048 (2k) BUT then the BLE stack on Linux start moaning and doesn't work (see dmesg on your Linux machine). I could only get it to work with MTU sizes smaller than 560 bytes (which seems to be "the standard").

BLE is really awkward if you want to send large amounts of data, but for example the ESP32-S3 only has BLE, not BT classic, I found out the hard way. If your ESP32-C3 does support BT classic, I'd suggest using that instead. Otherwise I am curious if it can be solved on the Espressif side.

lucic71, 2024-06-19:
@eriksl, I used bluedroid in the experiment from above but I also tried nimble with the throughput_app/bleprph_throughput example. For a 500 bytes packet with nimble I get around 450ms, that's about 1Kbps. Fortunately, the nimble example already sets the packet length using ble_hs_hci_util_set_data_len, as opposed to gatt_server which does not.

However, nimble fails to set such a big MTU as you proposed. ble_att_set_preferred_mtu has a hard limit at 527.

eriksl, 2024-06-19:
You can set the MTU in the sdk menuconfig. IIRC you also need to set it in either Nimble or on the client side.

Anyway, I have it now working if I use a MTU smaller than 560 bytes. On the client side I send a specific command string to the HCI that sets the MTU and then the server (esp32) follows, given you have configured a big enough MTU size in the sdkconfig.

With that configuration I can reach a speed of 3-4 kbytes per second. Which is awfully slow, but it's sufficient to do an emergency firmware upload if nothing else works. Upload takes a few minutes that way, it's doable.

My experience is that the examples aren't very good actually. It pays off to compare the API calls they're doing to the official documents of the ESP-ISD and the Nimble documentation. Sometimes some API calls aren't really required, sometimes some are missing and sometimes they need to be slightly different according to your own needs.

It took me A LOT of time to get a simple SPP-like thing working in BLE because of the not-so-great documentation and examples (both from Espressif and Nimble). I also had the bad luck I had to implement the client as well, on Linux, and here the documentation is even worse (Bluez). If you don't want to submit to the dbus nightmare, you're on your own there. Even though (I found out) talking to the HCI really isn't difficult, if only it were documented...

lucic71, 2024-06-26:
It's possible that there's a problem with gatttools. I established a connection from the computer with the Bleak python client and I got the desired MTU.

In this case there seems to be nothing wrong with esp-idf. I will close the issue.

closed 2024-06-26 by lucic71

eriksl, 2024-06-27:
No I don't think gatttools is the cause. I got messages from the kernel (= low level, in-kernel part of the BT). I searched the kernel source and found the offending line. It just simply assumes no one will ever need to use a bigger MTU in this transfer mode. This is detected at a point way too far in the process to correct it, or have the user informed in an orderly way, that's why it's reported in the dmesg and just fails with no clear reason (packet is dropped).