espressif/esp-idf #14015: Netif interface not yet in "up" state when sta is associated event is called.

eriksl, 2024-06-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.2

### Espressif SoC revision.

ESP32-S3 v0.1

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

Wemos Lolin S3-mini

### Power Supply used.

USB

### What is the expected behavior?

I would like to add a ipv6 "link local" address to the wifi sta interface as soon as it's associated. And possibly also a static ipv6 address.

### What is the actual behavior?

esp_netif_create_ip6_linklocal(netif_sta)) fails with ESP_FAILURE. From the source it turns out that the interface hasn't been set up at this stage. Apparently that only happens at some point after the dhcp ipv4 client is started and before the ipv4 address has been received. The problem is that I need this to work even if the dhcp server is not responding.

### Steps to reproduce.

```
static void wlan_event_handler(void *arg, esp_event_base_t event_base, int32_t event_id, void *event_data)
{
	switch(event_id)
	{
// ...
		case(WIFI_EVENT_STA_CONNECTED):
		{
			util_warn_on_esp_err("esp_netif_up", esp_netif_up(netif_sta)); // FIXME at this point the interface is not up, linklocal returns error
			util_warn_on_esp_err("esp_netif_create_ip6_linklocal", esp_netif_create_ip6_linklocal(netif_sta));
    			break;
		}
// ...
```

### More Information.

I worked around this issue by calling esp_netif_up explicitly, which is a private function. I'd rather have this fixed one way or another. The goal is to be able to add IP addresses at this point (ipv6), so I guess the netif should be simple set "up"before this point instead of at a later point. I don't see why that should be a problem. If not possible or preferred, then please make this function public.

david-cermak, 2024-06-21:
At which point do you register your handler: `wlan_event_handler()` ? Note that event handlers are called in the same order as they were registered and the same `event_id=WIFI_EVENT_STA_CONNECTED` is used to turn the netif up.
So maybe just moving your `esp_event_handler_register()` below the `esp_netif_create_default_wifi_sta()` would solve the problem?
One possible workaround would also be to call `esp_netif_action_connected(wifi_netif, 0, 0, 0);` (which is a public API)

david-cermak, 2024-06-21:
yes, the order matters, unfortunately, as creating the default wifi interface(s) also registers the (system) handlers and these should be called before user handlers. should be:
1) create default event loop
2) create default wifi netifs
3) register app handlers

```cpp
// 1)
	util_abort_on_esp_err("esp_event_loop_create_default", esp_event_loop_create_default());
// 2) 
	netif_sta = esp_netif_create_default_wifi_sta();
	netif_ap = esp_netif_create_default_wifi_ap();
// 3) 
	util_abort_on_esp_err("esp_event_handler_instance_register 1",
			esp_event_handler_instance_register(WIFI_EVENT, ESP_EVENT_ANY_ID, wlan_event_handler, (void *)0, (esp_event_handler_instance_t *)0));
	util_abort_on_esp_err("esp_event_handler_instance_register 2",
			esp_event_handler_instance_register(IP_EVENT, ESP_EVENT_ANY_ID, ip_event_handler, (void *)0, (esp_event_handler_instance_t *)0));
```

eriksl, 2024-06-21:
When I move the creation of the event loop after esp_netif_create_default_wifi_sta(), I get a panic from this line:

/**
 * @brief User init default station (official API)
 */
esp_netif_t* esp_netif_create_default_wifi_sta(void)
{
    esp_netif_config_t cfg = ESP_NETIF_DEFAULT_WIFI_STA();
    esp_netif_t *netif = esp_netif_new(&cfg);
    assert(netif);
    ESP_ERROR_CHECK(esp_netif_attach_wifi_station(netif));
    **ESP_ERROR_CHECK(esp_wifi_set_default_wifi_sta_handlers());**
    return netif;
}

So I guess that's not working. I am not sure that's documented, that you're required to first initialise the default event loop before calling esp_netif_create_default_wifi_sta()?

eriksl, 2024-06-21:
And now tried as per your suggestion above and yes, that does the trick! Thanks!

Maybe something for the documentation ;-)

Or add a feature to explicitly prepend/append handlers... Or even prevent to ever have user event handlers prepended before system handlers.

eriksl, 2024-06-21:
Now this is something you might know too: I am now struggling with the situation that sometimes a cached SLAAC address is used, instead of one based on the currently broadcast router advertisements. Is there a way to clear this cached version?

david-cermak, 2024-06-21:
> Maybe something for the documentation ;-)

Yeah, this needs to be (better) documented, already working on it.
Thanks for the report, this is really not a nice way of tying to init and moving things around. I'll think about making the API more robust...

> the situation that sometimes a cached SLAAC address is used

we currently don't have a good solution to invalidate older IPv6 addresses automatically, we generally suggest that users configure more IP addresses in `CONFIG_LWIP_IPV6_NUM_ADDRESSES`.
You can try to invalidate the cached address manually by calling `esp_netif_add_ip6_address()` with the previous IPv6 address and with *`preferred=false`*. This should find the previous address index in lwip and mark it `IP6_ADDR_DEPRECATED`.

eriksl, 2024-06-21:
Great news and thanks for the interest and effort!

The problem I am having with SLAAC is persistent. I mean that an obsolete SLAAC address is set after a restart! When I restart again, often the correct address is set. It looks like a router advertisement is solicited and when the (in my case) switch doesn't supply it quick enough, some old/cached information is used.

Would you by chance happen to know how LWIP determines  gateway and netmask for IPv6 static address? It looks like:
- netmask is assumed to be always /64 (bad...)
- gw is determined from either: the first non-local IPv6 packet arriving or sending to all-routers (ff02::2). I do not particularly like both of them.

I thought static IPv6 wouldn't work without SLAAC for prefix/gw, but to my surprise it does!

I already have an open issue about this in https://github.com/espressif/esp-idf/issues/14012 BTW.

david-cermak, 2024-06-21:
> I mean that an obsolete SLAAC address is set after a restart!

You mean after the ESP32 restart? This is weird, there's no way the address would persist any address after reset (unless it got an obsolete address from some RA message)?

> Would you by chance happen to know how LWIP determines gateway and netmask for IPv6 static address?

Not sure about that, but think that the prefix length is set by SLAAC (RA packet from `radvd`) and the default GW is actually your LL address (equals to your default netif), if you're asking how lwip chooses the source address, it's basically trying to find the best match with the prefix, I think.

eriksl, 2024-06-21:
Yes after a restart. I am pretty sure I configured it correctly, but unfortunately I cannot check it (the RA runs on the control plane of a switch). The interesting thing is that most of the time it works correctly...  But if you say the information isn't cached somewhere non-volatile, I am puzzled. Luckily I don't really need the SLAAC address (I am also assigning a static address and that works).

What I was saying is that using a static ipv6 address works even without RA (router) present and without SLAAC (ESP32) enabled. So the ESP32 doesn't have access to the router's address (not global nor link-local) or prefix length. I am reaching the ESP32 from outside it's network, so a gateway is definitely required. 

So I was under the impression static ipv6 would only work with SLAAC also enabled (for gw and prefix length) but apparently it also works without SLAAC.

So that kind of puzzles me too.

david-cermak, 2024-06-21:
> I am puzzled. Luckily I don't really need the SLAAC address (I am also assigning a static address and that works).

If you're interested, you can enable LWIP_DEBUG options with IPV6 in menuconfig, it will tell you if it got the IP from an RA message.

> What I was saying is that using a static ipv6 address works even without RA 

Oh, I see, if this is just statically assigned IP, then I think lwip doesn't need to know the prefix nor the GW, it just sends the packet, IMO.

> So I was under the impression static ipv6 would only work with SLAAC also enabled

As said, I think the lwip would just output the packet to the related netif (based on the link local) and, to the HW address previously cached by ND (if it doesn't know it, it just queues the packet and waits until the *next-hop* neighbor is available).
Then, it's up to the router to forward the packet (which would probably work if the subnet is the same as the destination prefix).

david-cermak, 2024-06-21:
@ammaree What is the problem you are facing?
I'd suggest checking the IDF sta+ap example: https://github.com/espressif/esp-idf/tree/master/examples/wifi/softap_sta

KaeLL, 2024-06-21:
@david-cermak IDF really ought to provide APIs for wrapping Wi-Fi init/deinit and tackle this ordering nonsense. There is literally no reason why there shouldn't be such an option because there's also no reason for bothering users with common use cases with this. I'd say take a page out of Arduino Core's book on convenience, but correctness alone should drive this point home. Of the top of my head, maybe have something like esp-mqtt init routine, with nested structs for grouping configurations, and the possibility of initializing multiple modes at once (ap, sta, ap+sta, etc).
There's also the issue of idempotency of `esp_event_create_loop_default`, that affects most other component initialization APIs as well, which again, makes the case for #8745.

eriksl, 2024-06-21:
If you don't mind me interfere, I am not quite crazy about Arduino and also I don't think everything should always be fully "automatic". It's really not bad to learn something once in a while. Just as long as everything is documented well and in this case, well, some improvement is possible. But it seems Espressif isn't unwilling there, so I fine with it. I am used to programming ESP8266 and this already is a world apart from that. Don't get me started...

I've seen a few "convenience" functions in the API docs, so maybe, if we ask nicely, some of these can be added for wifi initialisation too.

KaeLL, 2024-06-21:
I too am not a fan of the Arduino Core, but it felt right to use it as an example because this particular detail is one it gets right over IDF.
It's also not about being fully auto, but rather optionally manual, with convenient APIs for common use cases that would also handle the technicality of the order of function calls, and potentially ease the burden for users to upgrade IDF versions by managing eventual changes to WiFi init/deinit by itself.
As for documenting the behavior, in this case, documentation doesn't provide as much value IMO.

eriksl, 2024-08-01:
Original issue solved.

closed 2024-08-01 by eriksl

ammaree, 2024-10-15:
> @ammaree What is the problem you are facing? I'd suggest checking the IDF sta+ap example: https://github.com/espressif/esp-idf/tree/master/examples/wifi/softap_sta

@david-cermak 

I think we have solved this using info from this thread and a serious effort to simplify our WiFi manager. A lot of the complication arises from us having about 40 APs spread throughout a 3 story building with all APs offering 2 different SSIDs. Our Wifi manager stores details for up to 4 SSID/PSWD combinations and loops through these to make a connection. So our challenge was to consistently and reliably reconnect to an AP (either SSID) whenever the connection was dropped.

So far we have been running reliably on a couple of test devices for the last 72 hours, so far no errors. Will continue to monitor and confirm the outcome.

david-cermak, 2024-10-16:
> I think we have solved this using info from this thread and a serious effort to simplify our WiFi manager. 

FWIW, the IDF documentation has been updated based on the information from this thread:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_netif.html#initialization