espressif/esp-idf #14016: Exquisite Bug in WiFi LWIP layer being IRRESPONSIVE release/v5.2 ESP32 commit f81b598d1024175db6d5e75e4acdba9131c617d6

filzek, 2024-06-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Description:

We have discovered a critical bug in the LWIP/WIFI layer on the ESP32 in SDK commit f81b598d1024175db6d5e75e4acdba9131c617d6 (HEAD -> release/v5.2, origin/release/v5.2) on ESP32. This issue may be present in other versions as well.

Symptoms:

The WiFi on the ESP32 device is connected to the WiFi router, but the LWIP layer stops responding to any requests or sending data.
WebSocket connections simply time out.
Ping requests from external sources stop receiving responses.
When the WiFi router is turned off and then back on, the device reconnects to WiFi (the handler is called successfully), but remains unresponsive at the IP level on LWIP.
Workaround:

The issue can be temporarily resolved by destroying and rebuilding the network interface for the Wi-Fi station and LWIP layer using the following code:

```
// Destroy the network interface for the Wi-Fi station and Netif if already created, then rebuild it.
    if (sta_netif != NULL) {
        // Unregister event handlers
        ret = esp_event_handler_unregister(IP_EVENT, IP_EVENT_STA_GOT_IP, &event_handler);
        if (ret != ESP_OK) {
            printf("Error in %s at line %d: Failed to unregister IP_EVENT handler: %s\n", __FILE__, __LINE__, esp_err_to_name(ret));
        }
        ret = esp_event_handler_unregister(WIFI_EVENT, ESP_EVENT_ANY_ID, &event_handler);
        if (ret != ESP_OK) {
            printf("Error in %s at line %d: Failed to unregister WIFI_EVENT handler: %s\n", __FILE__, __LINE__, esp_err_to_name(ret));
        }

        // Stop Wi-Fi
        ret = esp_wifi_stop();
        if (ret != ESP_OK) {
            printf("Error in %s at line %d: Failed to stop Wi-Fi: %s\n", __FILE__, __LINE__, esp_err_to_name(ret));
        }

        // Deinitialize the Wi-Fi driver
        ret = esp_wifi_deinit();
        if (ret != ESP_OK) {
            printf("Error in %s at line %d: Failed to deinitialize Wi-Fi: %s\n", __FILE__, __LINE__, esp_err_to_name(ret));
        }

        // Clear Driver and Handlers of the Wi-Fi
        ret = esp_wifi_clear_default_wifi_driver_and_handlers(sta_netif);
        if (ret != ESP_OK) {
            printf("Error in %s at line %d: Failed to clear Wi-Fi driver and handlers: %s\n", __FILE__, __LINE__, esp_err_to_name(ret));
        }

        // Destroy the Wi-Fi station network interface
        esp_netif_destroy(sta_netif);
        sta_netif = NULL;

        // Delete the default event loop
        ret = esp_event_loop_delete_default();
        if (ret != ESP_OK) {
            printf("Error in %s at line %d: Failed to delete default event loop: %s\n", __FILE__, __LINE__, esp_err_to_name(ret));
        }

        // Destroy the event group
        if (s_wifi_event_group != NULL) {
            vEventGroupDelete(s_wifi_event_group);
            s_wifi_event_group = NULL;
        }

        // Deinitialize the network interface layer
        ret = esp_netif_deinit();
        if (ret != ESP_OK) {
            printf("Error in %s at line %d: Failed to deinitialize network interface: %s\n", __FILE__, __LINE__, esp_err_to_name(ret));
        }

        reconnect_wifi = 1;
    }
```

By destroying and reconstructing the LWIP and WIFI layer, the system returns to normal operation.

Unresolved Issues:

There is still an underlying issue causing this behavior, which has yet to be identified. Further investigation is needed to pinpoint the root cause of the problem.

david-cermak, 2024-07-10:
> The WiFi on the ESP32 device is connected to the WiFi router, but the LWIP layer stops responding to any requests or sending data.

Could you please share more details about this state? How can we replicate this condition? Could you please create a simple example that could be used to reproduce this issue? And maybe share more info about your local environment (e.g. type/make of your WiFi router...)

AxelLin, 2024-08-02:
@filzek Any feedback for commit https://github.com/espressif/esp-idf/issues/14016#issuecomment-2219832946 ?
BTW, from your report, you did destroy and reconstruct the LWIP and WIFI layer to make the system returns to normal operation. How do you detect if this issue is happened? (Or you just always destroy the network interface and create it again?)

filzek, 2024-08-09:
@AxelLin we cannot create a simple example to reproduce, but the wifi become irresponsive any packet will not flow by the TCP stact, disable and enable the WiFi will not solve the problem, to solve the problem its is really necessary to destroy and recreate it all.

We detect it by doing 4 kind of survey a UDP multicast request, a simple ping to the router that was previously answered, a ping to the cloud ping server, a websocket connectivity action and a http external request, those actions measure if the TCP/IP layer stack are working or not, and by the WiFi side we use the tools avaliable in the SDK, so, if there is no answer in a trivial check it so go to destroy the network after a defined constant check retries.

So far that was the unique way to make devices responsive again.

AxelLin, 2024-08-27:
@filzek 
Just wondering if this issue happens on older IDF versions?
If it's a new issue in certain IDF version, maybe you can try git bisect to find the first bad commit.

filzek, 2024-09-23:
Completele solved at commit c8c6fd141377b9a48b29c8ebb51fb7efadde0566 (HEAD -> release/v5.2, origin/release/v5.2)

closed 2024-09-23 by filzek