espressif/esp-idf #14017: ADC Continuous Mode Fails with Mixed Attenuation Values SDK 5.2.2 [BUG] [URGENT]

filzek, 2024-06-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

**Issue: ADC Continuous Mode Fails with Mixed Attenuation Values**

**Description**
I am experiencing a failure when configuring the ADC Continuous Mode using the ESP-IDF. The issue occurs when using two or more different attenuation values in the adc_digi_pattern_config_t pattern array to feed the adc_continuous_config_t configuration. Specifically, the call to ESP_ERROR_CHECK(adc_continuous_config(handle, &dig_cfg)); fails with an error.

**Steps to Reproduce**
Define a pattern array with multiple ADC channels, each having different attenuation values.
Configure the ADC continuous mode using this pattern array.
Observe that ESP_ERROR_CHECK(adc_continuous_config(handle, &dig_cfg)); fails.

Example Code

`static void continuous_adc_init(adc_continuous_handle_t *out_handle)
{
    int EXAMPLE_READ_LEN = 256;
    adc_continuous_handle_t handle = NULL;

    adc_continuous_handle_cfg_t adc_config = {
        .max_store_buf_size = 1024,
        .conv_frame_size = EXAMPLE_READ_LEN,
    };
    ESP_ERROR_CHECK(adc_continuous_new_handle(&adc_config, &handle));

    adc_continuous_config_t dig_cfg = {
        .sample_freq_hz = 20 * 1000,
        .conv_mode = ADC_CONV_SINGLE_UNIT_1,
        .format = ADC_DIGI_OUTPUT_FORMAT_TYPE1,
    };

    adc_digi_pattern_config_t adc_pattern[16] = {0};
    // Example pattern with different attenuation values
    adc_pattern[0].atten = ADC_ATTEN_DB_0;
    adc_pattern[0].channel = ADC_CHANNEL_3;
    adc_pattern[0].unit = ADC_UNIT_1;
    adc_pattern[0].bit_width = SOC_ADC_DIGI_MAX_BITWIDTH;

    adc_pattern[1].atten = ADC_ATTEN_DB_2_5;
    adc_pattern[1].channel = ADC_CHANNEL_6;
    adc_pattern[1].unit = ADC_UNIT_1;
    adc_pattern[1].bit_width = SOC_ADC_DIGI_MAX_BITWIDTH;

    dig_cfg.pattern_num = 2;
    dig_cfg.adc_pattern = adc_pattern;

    ESP_ERROR_CHECK(adc_continuous_config(handle, &dig_cfg));
    *out_handle = handle;
}
`

**Expected Behavior**
The adc_continuous_config function should successfully configure the ADC continuous mode even when using different attenuation values in the pattern array.

**Actual Behavior**
The ESP_ERROR_CHECK(adc_continuous_config(handle, &dig_cfg)); call fails, resulting in an ESP_ERR_INVALID_ARG error.

**Environment**
ESP-IDF Version: v5.2
Development Kit: [e.g., ESP32-DevKitC]
Module or chip used: [e.g., ESP32-WROOM-32]
Compiler version: [e.g., xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0]
Operating System: [e.g., Windows/Linux/macOS]
Power Supply: [e.g., USB]

**Additional Context**
This issue severely limits the use of the ADC continuous mode with channels that require different attenuation settings. It would be helpful to understand if this is a known limitation or if there is a workaround to allow mixed attenuation values in the configuration.

**LOG**

ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x400dddbd
0x400dddbd: continuous_adc_init at   main/include/timer_adc_read.c:106 (discriminator 1)

file: "./main/include/timer_adc_read.c" line 106
func: continuous_adc_init
expression: adc_continuous_config(handle, &dig_cfg)

abort() was called at PC 0x40092eb3 on core 1
0x40092eb3: _esp_error_check_failed at C:/Espressif/frameworks/esp-idf-v5.2/components/esp_system/esp_err.c:50

Backtrace: 0x40081b2a:0x3ffcc420 0x40092ebd:0x3ffcc440 0x40099f85:0x3ffcc460 0x40092eb3:0x3ffcc4d0 0x400dddbd:0x3ffcc500 0x40101e33:0x3ffcc590 0x40093b16:0x3ffccf10
0x40081b2a: panic_abort at C:/Espressif/frameworks/esp-idf-v5.2/components/esp_system/panic.c:466
0x40092ebd: esp_system_abort at C:/Espressif/frameworks/esp-idf-v5.2/components/esp_system/port/esp_system_chip.c:93
0x40099f85: abort at C:/Espressif/frameworks/esp-idf-v5.2/components/newlib/abort.c:38
0x40092eb3: _esp_error_check_failed at C:/Espressif/frameworks/esp-idf-v5.2/components/esp_system/esp_err.c:50

So, the Attenuation is not working at all for multiples continous reading at different channel settings.

suda-morris, 2024-06-19:
https://github.com/espressif/esp-idf/blob/master/components/esp_adc/adc_continuous.c#L490-L494

For the same ADC unit, the channels should use the same attenuation config.

filzek, 2024-06-19:
> https://github.com/espressif/esp-idf/blob/master/components/esp_adc/adc_continuous.c#L490-L494
> 
> For the same ADC unit, the channels should use the same attenuation config.

Make no sense, if we need to read it each channel wise, the continous reading doenst work at all, only for a fixed setup unit, so the document shall make this clear.

filzek, 2024-06-19:
The driver is not very well designed. The options to set are awkward and do not align with the intended functionality of the requested fields. It seems either the ADC was broken and not properly fixed, or the idea was abandoned before the task was complete.

If the adc_digi_pattern_config_t has a specific field for attenuation for each item added, this makes no sense at all again!

closed 2024-06-19 by filzek