espressif/esp-idf #14018: controller_vhci_ble_adv example on ESP32-C6 linking errors

mringwal, 2024-06-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-3065-g272b4091f1

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

The example should compile for ESP32-C6

### What is the actual behavior?

It fails to compile due to missing symbols: 
- undefined reference to `esp_vhci_host_send_packet'
- undefined reference to `esp_vhci_host_register_callback'
- others

### Steps to reproduce.

- Go to /examples/bluetooth/hci/controller_vhci_ble_adv
- idf.py set-target esp32-c6
- idf.py build

### Build or installation Logs.

```plain
➜  controller_vhci_ble_adv git:(master) ✗ idf.py build
Executing action: all (aliases: build)
Running ninja in directory /Users/mringwal/Projects/esp32/esp-idf/examples/bluetooth/hci/controller_vhci_ble_adv/build
Executing "ninja all"...
[1/1] cd /Users/mringwal/Projects/esp32/esp-idf/examples/bluetooth/hci/controller_vh...p-idf/examples/bluetooth/hci/controller_vhci_ble_adv/build/bootloader/bootloader.bin
Bootloader binary size 0x5400 bytes. 0x2c00 bytes (34%) free.
[4/6] Linking CXX executable ble_adv.elf
FAILED: ble_adv.elf
: && /Users/mringwal/Projects/esp32/idf-tools/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-g++ -march=rv32imac_zicsr_zifencei -nostartfiles -march=rv32imac_zicsr_zifencei --specs=nosys.specs    -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32C6=0 -Wl,--Map=/Users/mringwal/Projects/esp32/esp-idf/examples/bluetooth/hci/controller_vhci_ble_adv/build/ble_adv.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32c6.peripherals.ld -T esp32c6.rom.ld -T esp32c6.rom.api.ld -T esp32c6.rom.rvfp.ld -T esp32c6.rom.newlib.ld -T esp32c6.rom.version.ld -T esp32c6.rom.phy.ld -T esp32c6.rom.coexist.ld -T esp32c6.rom.net80211.ld -T esp32c6.rom.pp.ld -T esp32c6.rom.wdt.ld -T esp32c6.rom.newlib-normal.ld -T esp32c6.rom.heap.ld -T memory.ld -T sections.ld CMakeFiles/ble_adv.elf.dir/project_elf_src_esp32c6.c.obj -o ble_adv.elf -L/Users/mringwal/Projects/esp32/esp-idf/components/soc/esp32c6/ld   -L/Users/mringwal/Projects/esp32/esp-idf/components/esp_rom/esp32c6/ld   -L/Users/mringwal/Projects/esp32/esp-idf/examples/bluetooth/hci/controller_vhci_ble_adv/build/esp-idf/esp_system/ld   -L/Users/mringwal/Projects/esp32/esp-idf/components/esp_phy/lib/esp32c6   -L/Users/mringwal/Projects/esp32/esp-idf/components/esp_coex/lib/esp32c6   -L/Users/mringwal/Projects/esp32/esp-idf/components/esp_wifi/lib/esp32c6 esp-idf/riscv/libriscv.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_bootloader_format/libesp_bootloader_format.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/app_trace/libapp_trace.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_coex/libesp_coex.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/bt/libbt.a  esp-idf/btstack/libbtstack.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/console/libconsole.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/fatfs/libfatfs.a  esp-idf/ieee802154/libieee802154.a  esp-idf/json/libjson.a  esp-idf/mqtt/libmqtt.a  esp-idf/nvs_sec_provider/libnvs_sec_provider.a  esp-idf/spiffs/libspiffs.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/hci_common_component/libhci_common_component.a  esp-idf/main/libmain.a  esp-idf/app_trace/libapp_trace.a  esp-idf/app_trace/libapp_trace.a  esp-idf/btstack/libbtstack.a  esp-idf/cmock/libcmock.a  esp-idf/unity/libunity.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/fatfs/libfatfs.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/ieee802154/libieee802154.a  esp-idf/mqtt/libmqtt.a  esp-idf/nvs_sec_provider/libnvs_sec_provider.a  -u nvs_sec_provider_include_impl  esp-idf/spiffs/libspiffs.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/protocomm/libprotocomm.a  esp-idf/bt/libbt.a  /Users/mringwal/Projects/esp32/esp-idf/components/bt/controller/lib_esp32c6/esp32c6-bt-lib/libble_app.a  esp-idf/console/libconsole.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/json/libjson.a  esp-idf/hci_common_component/libhci_common_component.a  esp-idf/riscv/libriscv.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_bootloader_format/libesp_bootloader_format.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_coex/libesp_coex.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/mbedtls/mbedtls/library/libmbedtls.a  esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a  esp-idf/mbedtls/mbedtls/library/libmbedx509.a  /Users/mringwal/Projects/esp32/esp-idf/components/esp_coex/lib/esp32c6/libcoexist.a  /Users/mringwal/Projects/esp32/esp-idf/components/esp_wifi/lib/esp32c6/libcore.a  /Users/mringwal/Projects/esp32/esp-idf/components/esp_wifi/lib/esp32c6/libespnow.a  /Users/mringwal/Projects/esp32/esp-idf/components/esp_wifi/lib/esp32c6/libmesh.a  /Users/mringwal/Projects/esp32/esp-idf/components/esp_wifi/lib/esp32c6/libnet80211.a  /Users/mringwal/Projects/esp32/esp-idf/components/esp_wifi/lib/esp32c6/libpp.a  /Users/mringwal/Projects/esp32/esp-idf/components/esp_wifi/lib/esp32c6/libsmartconfig.a  /Users/mringwal/Projects/esp32/esp-idf/components/esp_wifi/lib/esp32c6/libwapi.a  esp-idf/riscv/libriscv.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/d [... cut ...]
This is the same issue as https://github.com/espressif/esp-idf/issues/12194
However, the original poster had to patch esp-idf & his code locally to work around.

The main issue here is that the VHCI Interface gets disabled without CONFIG_BT_BLUEDROID_ENABLED.
The VHCI interface is very important to allow for alternative Bluetooth stacks, e.g. our BTstack, to successfully run on all ESP32 variants that contain a Bluetooth Controller.

Please enabled the VHCI interface for the C6 as it was done for ESP32, ESP32-C3, ESP32-S3,...

Thanks!

zhaoweiliang2021, 2024-06-20:
@mringwal  hi, This example(/examples/bluetooth/hci/controller_vhci_ble_adv) does not currently support ESP32C6 chip. There are currently plans to support it. You need to wait for a while.

mringwal, 2024-06-20:
@zhaoweiliang2021 that would be great! Any idea on a time frame?

ricardoquesada, 2024-07-24:
Friendly ping. Any ETA for this fix ? ty!

zhaoweiliang2021, 2024-07-25:
@mringwal  @ricardoquesada This code is already supported and awaiting further verification and code synchronization. Thanks.

weiyuhannnn, 2024-09-06:
Hello, @mringwal @ricardoquesada  The support has been merged with the following commits. We recommend using the latest release branches for your projects. Thx!

- master: abfca50561dd1c3a3bc86fd251558d2ece0dc99d
- v5.3: 0efdd4ac0132981ab28901143983a4fb8ad1c088
- v5.2: d0da89f18a6169b2cf9b61c05db833bb04b85131
- v5.1: d82f9162dfe98d6aeec8a1cdcae08693f0738161
- v5.0: 8d86ea02a2725b57a8756daf6e6bc6d6221adc62

closed 2024-09-06 by weiyuhannnn

ricardoquesada, 2024-09-06:
@weiyuhannnn yay, many thanks!