espressif/esp-idf #14019: I'm working on an ESP32 HTTPS server using a simple example code. How do I solve this error?

vishnuprasadvp12345, 2024-06-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

esp-idf/components/esp_https_server/include/esp_https_server.h:170:1: error: 'httpd_ssl_config_t' {aka 'httpd_ssl_config'} has no non-static data member named 'cert_select_cb'
170 | }
| ^
esp-matter/examples/apmatterhub/main/Https.cpp:192:33: note: in expansion of macro 'HTTPD_SSL_CONFIG_DEFAULT'
192 | httpd_ssl_config_t config = HTTPD_SSL_CONFIG_DEFAULT();
| ^~~~~~~~~~~~~~~~~~~~~~~~

ninja: build stopped: subcommand failed.
[https_error.txt](https://github.com/user-attachments/files/15898606/https_error.txt)

hrushikesh430, 2024-06-20:
Hello @vishnuprasadvp12345,
which ESP-IDF version are you using?

Alvin1Zhang, 2024-06-26:
Thanks for reporting, will close due to short of feedback and details, feel free to reopen.

closed 2024-06-26 by Alvin1Zhang