espressif/esp-idf #14021: ESP32-C5 support status

igrr, 2024-06-19:
# ESP32-C5 support status in ESP-IDF

ESP-IDF v5.4, currently in development, will include initial support for the mass production version of the ESP32-C5 chip.

In the list below, supported features are marked with a checked checkbox, unsupported features have an unchecked checkbox. An internal issue reference (such as "IDF-1234") is listed at the end of the feature description to help us keep this list up to date:

- ⏳ Unsupported feature (IDF-1234)
- ✅ Supported feature

This list will be updated as the support for ESP32-C5 is being developed.

## Reporting issues about ESP32-C5

If you have an issue to report about any of the ESP32-C5 features, please create a separate issue. This issue is only intended for announcements of ESP32-C5 support status.

---
- Low Power System
  - ✅ Low Power Core (RISC-V)
  - ⏳ Low-power I2C Master (IDF-8634)
  - ⏳ Low-power Timer (IDF-8636)
  - ⏳ Low-power UART (IDF-8633)
  - ⏳ Low-power Watchdog Timer (IDF-8635)
- Peripherals
  - ADC
    - ⏳ ADC Calibration (IDF-8702)
    - ⏳ ADC continuous mode driver (IDF-8701)
      - ADC continuous mode digital monitor
    - ⏳ ADC oneshot mode driver (IDF-8703)
  - ⏳ Bit Scrambler (IDF-8711)
  - DMA (Direct Memory Access)
    - ✅ GDMA
  - ⏳ Dedicated GPIO driver (IDF-8725)
  - Ethernet driver
    - ⏳ SPI Ethernet driver (IDF-8697)
  - ⏳ Event Task Matrix (IDF-8693)
  - ✅ GPIO driver
    - ⏳ GPIO ETM (IDF-8720)
    - ⏳ Glitch Filter (IDF-8718)
    - ⏳ RTC IO (LP IO) driver (IDF-8719)
  - ⏳ GPIO sigma delta driver (IDF-8687)
  - GPSPI
    - ✅ SPI Master driver
    - ✅ SPI Slave HD driver
    - ✅ SPI Slave driver
  - ✅ GPTimer driver
  - I2C
    - ✅ I2C master driver
      - ⏳ Support Low Power I2C instance (IDF-8695)
    - ✅ I2C slave driver
  - I2S
    - ✅ I2S driver
      - I2S PDM tx mode
      - I2S STD mode
      - I2S TDM mode
    - ✅ I2S legacy driver
  - ⏳ LCD driver (IDF-8708)
    - I2C LCD driver
    - SPI LCD driver
  - ✅ LEDC driver
    - Gamma Curve Generation
  - ✅ MCPWM driver
  - ✅ PCNT driver
  - PSRAM
    - ⏳ .bss PSRAM (IDF-8689)
    - ⏳ PSRAM Device Driver (IDF-8690)
    - ⏳ XIP PSRAM (IDF-8688)
  - Parallel IO
    - ⏳ Parallel IO RX Driver (IDF-8686)
    - ⏳ Parallel IO TX Driver (IDF-8685)
  - ✅ RMT driver
  - SD/SDIO/MMC driver
    - ⏳ SDSPI Host driver (IDF-8704)
  - ⏳ SPI Flash driver (IDF-8715)
  - Systimer driver
    - ✅ Esptimer implementation over systimer
    - ✅ OS tick port over systimer
  - ⏳ TWAI driver (IDF-8691)
  - ⏳ Temperature Sensor driver (IDF-8727)
  - UART
    - ✅ Support Low Power UART instance
    - ✅ UART FIFO mode driver
    - ⏳ UHCI (UART DMA mode) driver (IDF-8724)
  - ⏳ USB Serial JTAG (IDF-8721)
- Power management
  - ⏳ Auto Light Sleep (IDF-8641)
  - ⏳ Deep Sleep (IDF-8638)
  - ⏳ Dynamic Frequency Switch (IDF-8639)
  - ✅ Light Sleep
- Security Features
  - ✅ AES accelerator
  - ✅ Digital Signature driver
  - ✅ ECC accelerator
  - ✅ ECDSA driver
  - ✅ Flash encryption
  - ✅ HMAC driver
  - ⏳ HUK Generator (IDF-8617)
  - ⏳ Key Manager (IDF-8621)
  - ⏳ RNG (IDF-8626)
  - ✅ RSA hardware accelerator driver
  - ✅ SHA accelerator
  - ✅ Secure boot
  - Trusted Execution Environnment
    - ⏳ Access Permission Management (IDF-8615)
    - ⏳ TEE Controller (IDF-8614)
- System Features
  - ✅ Bootloader Support
  - ⏳ Brownout Detector (IDF-8647)
  - Cache
    - ⏳ Cache Driver (IDF-8646)
  - ⏳ Clock Glitch Detector (IDF-8666)
  - ⏳ Console (IDF-8651)
  - ✅ Cxx Component
  - ✅ ESP Event
  - ✅ ESP ROM
  - ✅ ESP Ringbuffer
  - ✅ ESP Timer
  - ✅ Efuse controller driver
  - ✅ FreeRTOS
  - ⏳ MMU (IDF-8658)
  - MSPI
    - ⏳ MSPI tuning (Flash & PSRAM DDR and/or over 80 MHz) (IDF-8649)
    - ⏳ SPI Flash auto suspend (IDF-8648)
  - ✅ POSIX Threads
  - Panic Handling Features
    - ⏳ Core Dump (IDF-8661)
    - ✅ Debug Watchpoint
    - ⏳ GDB Stub (IDF-8665)
    - ✅ Panic Handler
    - ⏳ Stack Guard (IDF-8662)
  - ⏳ Ram App (IDF-8644)
  - ✅ Watch Dog Timers
  - ⏳ app trace (IDF-8657)
  - ⏳ newlib (IDF-8675)
- Wi-Fi
  - ⏳ 802.11bgn, 802.11an, HT40 (WIFI-6407)
  - ⏳ 802.11ax: 20MHz-only non-AP mode (WIFI-6408)
  - ⏳ 802.11ac: 20MHz-only non-AP mode (WIFI-6579)
  - ⏳ Station, SoftAP, sniffer mode (WIFI-6409)
  - ⏳ UL/DL OFDMA (26/52/106/242 tone) (WIFI-6410)
  - ⏳ DL MU-MIMO, DL MU-MIMO within OFDMA (WIFI-6412)
  - ⏳ HE ER (HE Extended Range), DCM and ESP LR (ESP Long Range) (WIFI-6413)
  - ⏳ BSS Color (WIFI-6414)
  - ⏳ CSI (Channel State Information) (WIFI-6415)
  - ⏳ ESP-NOW (WIFI-6416)
  - ⏳ ESP-WiFi-Mesh (WIFI-6417)
  - ⏳ ESP-Touch v1/v2 (WIFI-6418)
  - ⏳ Wi-Fi Security (WIFI-6419)
  - ⏳ DPP (Device Provisioning Protocol) (WIFI-6420)
  - ⏳ Network-assisted roaming using 802.11k (Radio Measurements), 802.11v (BTM) specification and 802.11r (Fast BSS Transition) (WIFI-6421)
  - ⏳ Individual TWT (WIFI-6422)
  - ⏳ Broadcast TWT (WIFI-6423)
  - ⏳ Advanced DTIM Sleep (WIFI-6424)
  - ⏳ Power Save: modem sleep, light sleep (WIFI-6425)
  - ⏳ FTM (Fine Time Measurement)  (WIFI-6426)
  - ⏳ 5G Channel and Country Code (WIFI-6427)
  - ⏳ DFS (Dynamic Frequency Selection)  (WIFI-6429)
  - ⏳ AP Choose (2G/5G)  (WIFI-6430)
  - ⏳ Fast scan, fast connect  (WIFI-6578)
- Coexistence
  - ⏳ Coexistence of Wi-Fi and Bluetooth LE (IDF-9138)
  - ⏳ Coexistence of Wi-Fi and 802.15.4 (IDF-9139)
  - ⏳ Coexistence of Bluetooth LE and 802.15.4 (IDF-9140)
  - ⏳ External coexistence (IDF-9141)
- 802.15.4
  - ⏳ 802.15.4 Mac (TZ-959)
  - Thread
    - ⏳ Thread FTD/MTD/SSED (TZ-956)
    - ⏳ Thread RCP (TZ-956)
    - ⏳ Thread Border Router (TZ-957)
    - ⏳ Thread Light/Deep Sleep Device (TZ-958)
  - Zigbee
    - ⏳ Zigbee 3.0 Home Automation (TZ-954)
    - ⏳ Zigbee RCP (TZ-954)
    - ⏳ Zigbee Gateway (TZ-954)
    - ⏳ Zigbee Light/Deep Sleep Device (TZ-955)

Alvin1Zhang, 2024-06-25:
Update on June 25th, 2024: support of Low-power Timer, Parallel IO RX Driver, Parallel IO TX Driver and USB Serial JTAG have been added.

Alvin1Zhang, 2024-07-03:
Update on July 3rd, 2024: support of Low-power Watchdog Timer, LCD driver, Cache Driver and MMU have been added.

Alvin1Zhang, 2024-07-29:
Update on July 29th, 2024: support of below features have been added. 

* Bluetooth 5 Controller (Bluetooth 5.3 Certified)
* ESP-Bluedroid Host
* ESP-NimBLE Host
* HCI
* Low-power UART
* ADC continuous mode driver
* ADC oneshot mode driver
* Dedicated GPIO driver
* Event Task Matrix
* Glitch Filter
* RTC IO (LP IO) driver
* GPIO sigma delta driver
* PSRAM Device Driver
* RMT driver
* Trusted Execution Environnment
* 802.11bgn, 802.11an, HT40
* 802.11ax: 20MHz-only non-AP mode
* Network-assisted roaming using 802.11k (Radio Measurements), 802.11v (BTM) specification and 802.11r (Fast BSS Transition)
* UL/DL OFDMA (26/52/106/242 tone)
* DL MU-MIMO, DL MU-MIMO within OFDMA
* HE ER (HE Extended Range), DCM and ESP LR (ESP Long Range)
* BSS Color
* Station, SoftAP, sniffer mode
* Wi-Fi Security
* Individual TWT
* Broadcast TWT
* Coexistence of Wi-Fi and 802.15.4
* CSI (Channel State Information)

Alvin1Zhang, 2024-08-30:
Update on August 30th, 2024: Support of ESP-BLE-MESH, Low-power I2C Master, SPI Ethernet driver, I2C LCD driver, SPI LCD driver, .bss/.noinit PSRAM, SDSPI Host driver, Temperature Sensor driver, RNG, Auto Light Sleep, Deep Sleep, Dynamic Frequency Switch, Brownout Detector, Console, newlib have been added.

jack0c, 2024-09-04:
Update on Sep 4th, 2024: support of below features have been added.

802.15.4
  - ✅ 802.15.4 Mac
Coexistence
  - ✅ Coexistence of Bluetooth and 802.15.4
  - ✅ Coexistence of Wi-Fi and Bluetooth
  - ✅  External coexistence leader mode
- Power management
  - ✅  Light Sleep
- Thread
  - ✅ Thread Border Router
  - ✅ Thread FTD/MTD
  - ✅ Thread RCP
Wi-Fi
  - ✅ ESP-NOW
  - ✅ ESP-WiFi-Mesh

xuxiao111, 2024-09-14:
For using Sample Firmware Version (**ESP-IDF v5.4-dev-624-g3d167a46ff-dirty**):
Recommended Usage Steps:

1. Connecting to a 2.4G AP:

- 1.1. Switch band to 2.4G: band 1

- 1.2. Scan for 2.4G AP: scan

- 1.3. Connect to the AP: sta <ssid> <password>

- 1.4. Perform an iperf test:
      iperf -s (as TCP server)
      iperf -s -u (as UDP server)
      iperf -c xxx:xxx:xxx:xxx (as TCP client)
      iperf -c xxx:xxx:xxx:xxx -u (as UDP client)

2. Connecting to a 5G AP:

- 2.1. Switch band to 5G: band 2

- 2.2. Scan for 5G AP (5G channels are more than 2G and include DFS channels, so the scan time is longer than for 2G): scan

- 2.3. Connect to the AP: sta <ssid> <password>

- 2.4. Perform an iperf test (If no data is sent or received within 10 seconds, iperf will be aborted, and the socket will be closed and shut down.):

      iperf -s (as TCP server)
      iperf -s -u (as UDP server)
      iperf -c xxx:xxx:xxx:xxx (as TCP client)
      iperf -c xxx:xxx:xxx:xxx -u (as UDP client)

**Know issues:**
    1. The serial port baud rate is **138240**; using 115200 will result in garbled output.
    2. After the device starts, it defaults to the 5G band, contrary to what is described in the log as 2G. You can switch between the 2G and 5G bands using the band command: band 1 switches to 2G, and band 2 switches to 5G.
    3. On the 5G band, Wi-Fi scanning cannot detect APs on channel 36, and connections to these APs cannot be established. For this firmware version, it is not recommended to use the device in softAP mode.
    4. When the softAP is operating on the 5G band, it defaults to channel 36 instead of channel 112, as described in the log.
These issues have been resolved in commit **https://github.com/espressif/esp-idf/commit/34813ecd6ce44bc6c42274ded2f9043434384645**. If you encounter the above issues, sorry for not providing a timely update on the aforementioned issues.And please update to this version and use the iperf example for testing. (Note that in the new version, some commands differ from those in the sample firmware, so please be aware when using them.)
And also, thanks @ESP32DE to for reporting these issues.

Alvin1Zhang, 2024-10-10:
Update on October 10th, 2024: support of features below has been added.

* BLE Light Sleep
* Support Low Power I2C instance
* XIP PSRAM
* SPI Flash auto suspend
* Core Dump
* GDB Stub
* Brownout Detector
* Ram App
* app trace
* Advanced DTIM Sleep
* ESP-Touch v1/v2
* Fast scan, fast connect
* Power Save: modem sleep, light sleep

Alvin1Zhang, 2024-11-05:
Update on November 5th, 2024: support of features below has been added.

* OpenOCD
* AP Choose (2G/5G) 
* DPP (Device Provisioning Protocol)

alberk8, 2024-11-11:
I have a question regarding the C5 Wifi band switching.   Does it automatically connect to the 5G or 2.4G of the AP (the AP has dual band enable on the same SSID) and switch to best band when available especially when roaming ?.    Thank you.

xuxiao111, 2024-11-11:
Hi @alberk8 , when the AP uses the same SSID and password for both 2.4G and 5G, the STA (station) will automatically choose the best band to connect to. Considering that the 5G signal attenuates more than 2.4G at the same distance, we have added a configuration option, [rssi_5g_adjustment](https://github.com/espressif/esp-idf/blob/ce6085349f8d5a95fc857e28e2d73d73dd3629b5/components/esp_wifi/include/esp_wifi_types_generic.h#L350), to help decide whether to connect to the 5G or 2.4G band.

martinius96, 2024-12-18:
and how to get that sample? :P

Espressif-yangxi, 2024-12-27:
> and how to get that sample? :P
For sample request, please [contact us](https://www.espressif.com/zh-hans/contact-us/sales-questions) for support. thanks.

martinius96, 2024-12-27:
I already have last week.
Sample is now in transit :-) But thanks

dmartauz, 2025-01-04:
Can Thread Border Router be operated solely on C5 without any additional SoC? Coexistence of 802.15.4 and WiFi is already implemented...

chshu, 2025-01-06:
@dmartauz  There is only one RF path in ESP32-C5, which means Wi-Fi and Thread can't receive simultaneously, it has a significant impact on performance. Hence the two SoCs solution is recommended: [ESP Thread BR](https://docs.espressif.com/projects/esp-thread-br/en/latest/hardware_platforms.html#wi-fi-based-thread-border-router)

chshu, 2025-01-07:
@dmartauz 

>  however it introduces complexity to OTA remote firmware update of RCP

The ESP Thread Border Router SDK supports the [RCP update](https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/rcp_update.html#rcp-update-mechanism) feature. It automatically flashes the RCP firmware to the RCP SoC during the first boot, making the process seamless for the application. A single OTA update to the host SoC is sufficient.

As a Thread Border Router, the 802.15.4 interface must continuously receive data. However, since there is only one RF path, whenever Wi-Fi uses the radio for transmission or receiving, the 802.15.4 interface cannot receive, potentially leading to packet loss. The higher the traffic on Wi-Fi, the greater the packet loss rate on 802.15.4.

The [ot_br](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_br) example can run on a single Wi-Fi + 802.15.4 SoC (ESP32-C6, ESP32-C5). The BR functionality works, but it is not recommended for production use due to the high packet loss rate and poor performance.

Please file an issue here: https://github.com/espressif/esp-thread-br/issues, if you have further questions.

h-milz, 2025-01-09:
Does "802.11ax: 20MHz-only non-AP mode" mean the C5 cannot work as softAP with 11ax? I received a pair of C5 sample devkits this week and am trying to get them to talk 11ax in a private 5G  network, one as ap and one as sta, and they always negotiate 11an. Is there a way to enforce 11ax in such a setup? TIA!

QingzhaoYin, 2025-01-10:
@h-milz C5 can work as softAP with 11ax. If do not set wifi protocol when config softap, the softap will start with 11N protocol. You can use API esp_wifi_set_protocol() or esp_wifi_set_protocols() in esp_wifi.h to set the softap protocol to 11AX.
If you are using iperf example of wifi, you can use command "wifi_protocol" to set the protocol.

h-milz, 2025-01-10:
@QingzhaoYin  thank you for picking up. In this case I may be missing something. I need to enforce an 11ax connection because the project is supposed to work in a congested environment, which is what 11ax was made for. As far as I can see, there is no way to enforce 11ax. I checked with the iperf example that you mentioned: 

```
iperf> wifi_mode ap
I (1488536) WIFI: mode: ap
W (1488537) wifi:(itwt)itwt_stop_process!
I (1488537) wifi:mode : softAP (60:55:f9:fb:cf:41)
I (1488539) wifi:Total power save buffer number: 24
I (1488549) wifi:Init max length of beacon: 752/752
I (1488550) wifi:Init max length of beacon: 752/752
I (1488550) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
I (1488550) WIFI: DONE.SET_WIFI_MODE,OK.
iperf> ap_set BLA ************
I (1495539) wifi:Total power save buffer number: 24
I (1495540) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
I (1495541) WIFI: DONE.SET_AP_CONFIG,OK.
iperf> wifi_band 5g
I (1503504) WIFI: DONE.SET_WIFI_BAND_MODE,OK.
iperf> wifi_protocol ax
I (1515651) WIFI: DONE.SET_WIFI_PROTOCOL,FAIL.258,ESP_ERR_INVALID_ARG
iperf> wifi_protocol --5g=ax
I (1520442) WIFI: DONE.SET_WIFI_PROTOCOL,FAIL.258,ESP_ERR_INVALID_ARG
iperf> wifi_protocol
E (1525456) WIFI: Can not convert bitmap 0x0 to protocol string.
I (1525456) WIFI: (sta)GET_WIFI_PROTO:unknown, GET_WIFI_PROTO_5G:a/n/ac/ax

```

As soon as you try to

`esp_wifi_set_protocol(WIFI_IF_AP, WIFI_PROTOCOL_11AX);`

you get the error `ESP_ERR_INVALID_ARG`. 

And `esp_wifi_set_protocols()` appears to also set the whole protocol family for 5G, and there seems no way to explicitly select only one of these. `cmd_do_wifi_protocol()` also invokes only one of these two library functions. 

The documentation says "2. When WiFi band mode is 5G only, support 802.11a or 802.11an or 802.11anac or 802.11anacax". 

Apparently, you need to set the whole family A, N, AC and AX, which is the default for 5G. And in this case the Wifi stack seems to scan the lower bits first, and choose the lowest one that succeeds, in this case N. It won't even try to select AX because N succeeds. (I'd really like to try and look this up in the source code of `libnet80211.a` but sadly, there's only the binary library.) This is different with my Wifi router which negotiates AX at least every other connection attempt. 

I figure this is a case for a feature request.

weoiss1998, 2025-01-10:
Can I use ESP-NOW with 5Ghz?

kylefmohr, 2025-01-10:
> Can I use ESP-NOW with 5Ghz?

@weoiss1998 If you have one of the ESP32-C5 samples, then yes. I was able to see and connect to my 5GHz networks.

Jddl, 2025-01-12:
Could you provide an approximate mass production schedule for ESP32-C5? I have a product that urgently requires a 5G WiFi chip.

QingzhaoYin, 2025-01-13:
@weoiss1998 Yes,  ESP-NOW is supported on both 2Ghz band and 5Ghz

Espressif-yangxi, 2025-01-13:
> Could you provide an approximate mass production schedule for ESP32-C5? I have a product that urgently requires a 5G WiFi chip.

@Jddl Please feel free to  [contact us](https://www.espressif.com/zh-hans/contact-us/sales-questions). Our customer support team will provide you with the latest schedule and resource updates. Thank you!

DogAwesome21, 2025-01-13:
@Espressif-yangxi I tried emailing sales@espressif.com but was told to go to the forums because I was emailing through a Gmail account and not a business account even though I am not a business and need it for a personal project as I am making software for the esp32 series that gives flipper zeros wifi (flipperHTTP) and need to make support for 5ghz. I was told to go to the forums, but there was no information there, and I was told to contact sales@espresif.com. How can I get a sample? Please help me. If anyone has any extras I would be willing to buy one off of you.

Espressif-yangxi, 2025-01-14:
> @Espressif-yangxi I tried emailing [sales@espressif.com](mailto:sales@espressif.com) but was told to go to the forums because I was emailing through a Gmail account and not a business account even though I am not a business and need it for a personal project as I am making software for the esp32 series that gives flipper zeros wifi (flipperHTTP) and need to make support for 5ghz. I was told to go to the forums, but there was no information there, and I was told to contact [sales@espresif.com](mailto:sales@espresif.com). How can I get a sample? Please help me. If anyone has any extras I would be willing to buy one off of you.

@DogAwesome21  Sorry for any confusion caused by our previous responses. Here’s the clarification to help resolve any misunderstanding:  
If you have technical questions, such as discussions about IDF support, we kindly invite you to visit the forum. 
For sample requests, access to unpublished documents, updates on new product timeline, mass production support, or any other inquiries,  please don’t hesitate to contact our customer support team([contact us](https://www.espressif.com/zh-hans/contact-us/sales-questions))—we’re ready to help!

h-milz, 2025-01-20:
Hi guys, I have a question concerning 5G Channel and Country Code. I assume this will be ready in IDF 5.5, but may I ask if there will be something similar to 2.4 GHz international safe mode, allowing only channels that are safe in most if not all countries according to [this list](https://en.wikipedia.org/wiki/List_of_WLAN_channels#5_GHz_(802.11a/h/n/ac/ax/be))? My project should be usable worldwide, and I would like to avoid having to add any controls for a country setting. 

Thank you!

QingzhaoYin, 2025-01-20:
@h-milz In the near future, we will support country codes related to 5G channels and there will be something similar to the 2.4 GHz international safe mode (country code "01"). For now, you can configure the supported 5G channels through the field `wifi_5g_channel_mask` of `wifi_country_t`. Below is a reference example:

[image]

```
#if CONFIG_SOC_WIFI_SUPPORT_5G
uint32_t wifi_cmd_channel_list2bitmsk(wifi_band_t band, uint8_t *channel_list, uint8_t len)
{
    uint32_t channel_bitmap = 0;
    if (!channel_list) {
        return 0;
    }
    for (int i = 0; i < len; i++) {
        uint8_t channel = channel_list[i];
        if (band == WIFI_BAND_2G && channel > 14) {
            continue;
        } else if (band == WIFI_BAND_5G && channel < 36) {
            continue;
        }
        channel_bitmap |= CHANNEL_TO_BIT(channel);
    }
    return channel_bitmap;
}
#endif
 
wifi_country_t country = {
    .cc = "01",
    .schan = 1,
    .nchan = 11,
    .policy = WIFI_COUNTRY_POLICY_MANUAL,
};

#if CONFIG_SOC_WIFI_SUPPORT_5G
uint8_t channel_list[] = {36, 40, 44, 48, 52, 56, 60, 64};
country.wifi_5g_channel_mask = wifi_cmd_channel_list2bitmsk(WIFI_BAND_5G, channel_list, 8);
#endif
esp_wifi_set_country(&country);
```

chegewara, 2025-02-28:
Hi team,
what is the ADC status, because my client want to release product and i need to use NTC component which is using ADC?

OP stands that ADC calibration is still not implemented.

Thanks

Alvin1Zhang, 2025-02-28:
> Hi team, what is the ADC status, because my client want to release product and i need to use NTC component which is using ADC?
> 
> OP stands that ADC calibration is still not implemented.
> 
> Thanks

Thanks for the notes and sorry for the slow updates, will update the thread soon today.

Alvin1Zhang, 2025-02-28:
Update on February 28th, 2025: support of below features have been added.

* External coexistence follower mode
* ADC Calibration
* Bit Scrambler
* Analog Comparator
* GPIO ETM
* Stack Guard

jumpjack, 2025-04-16:
Support for USB HID emulation would be very appreciated.

robopp55, 2025-04-16:
Hi,
I have a couple of questions regarding future C5 software:

1. Will SoftAP mode be limited to 802.11an when configured to 5Ghz, or will 802.11ac/ax be added in the future?
2. When operating in SoftAP mode, are there plans to add HT40 MCS7 data rates?

Thank you!

QingzhaoYin, 2025-04-17:
@robopp55  Hi~, In fact, 802.11an/ac/ax have been supporttd in C5 software on latest master branch of IDF. When operating in SoftAP mode, 40MHZ bandwidth is only supported in 802.11n/an mode， this means that HT40 MCS7 data rates are only supported in 802.11n/an mode.

robopp55, 2025-04-17:
Hi @QingzhaoYin So SoftAP mode is supported when configured for 802.11ax and a BW of 20MHz?

QingzhaoYin, 2025-04-17:
@robopp55 SoftAP mode can be configured  for 802.11ax or 802.11ac or 802.11an/n, but 40M bandwidth on supported in 802.11an/n mode.

robopp55, 2025-04-17:
Thank you @QingzhaoYin. Will the C5 support 802.11ax 40Mhz in the future, or is this a hardware limitation?

QingzhaoYin, 2025-04-18:
@robopp55 802.11ax 40Mhz cannot be supported on ESP32C5 for a hardware limitation.