espressif/esp-idf #14022: Classic Bluetooth A2DP SOURC distribution network audio crash

ygl176, 2024-06-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.2.2-dirty

### Espressif SoC revision.

ESP32-D0WD

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-LyraT-Mini V1.2

### Power Supply used.

USB

### What is the expected behavior?

Coexistence among Wi-Fi, A2DP, HFP and Bluetooth LE;
Classic Bluetooth A2DP SOURCE distributes audio steams.

### What is the actual behavior?

Program crashes after a few hours of operation。

### Steps to reproduce.

1. connect wifi
2. gatts init
3. a2dp source init

### More Information.

[log.txt](https://github.com/user-attachments/files/15901778/log.txt)
[pipeline_a2dp_source_stream.zip](https://github.com/user-attachments/files/15901790/pipeline_a2dp_source_stream.zip)

Espressif-liuuuu, 2024-06-25:
Hi @ygl176 , the crash is due to the internal exception log being printed in critical area.   
To fix the issue is simple, but we still wonder why the internal exception is triggered, could you pls try this wifi lib 
[esp32.zip](https://github.com/user-attachments/files/15969104/esp32.zip) and test if any log starting with`[issue]` could be reproduced? thats the exception debug log we want, thanks!    

Pls put the esp32 folder in $IDF_PATH/components/esp_wifi/lib, then fullclean the project and rebuild.

ygl176, 2024-06-26:
Thanks for your reply,This is a log of three runs.
[http_log.txt](https://github.com/user-attachments/files/15983395/http_log.txt)

Espressif-liuuuu, 2024-06-27:
Hi @ygl176 , thanks for your report.  
Its shown that there is an important wifi timer callback hasnt been executed for about 27s, leading to too much time gap and trigger the exception log. After that, it seems that wifi connection broken and so did http_stream.  
Could you pls help check the esp_timer using in your application and check if any operation in timer callback may taking too much time?

ygl176, 2024-07-01:
I'll try to check, thanks.

ygl176, 2024-07-02:
I found a new crash log.
[bt_crash_log.txt](https://github.com/user-attachments/files/16060577/bt_crash_log.txt)
[sdkconfig.zip](https://github.com/user-attachments/files/16060585/sdkconfig.zip)

ygl176, 2024-07-02:
Sometimes, the bt source stops, but ESP32 does not print any logs.And I turned off the BT slave device and also didn't print any logs.
[bt_stop_not_log.txt](https://github.com/user-attachments/files/16060642/bt_stop_not_log.txt)

ygl176, 2024-07-02:
Timer print every 5s.
[timer_log.txt](https://github.com/user-attachments/files/16060766/timer_log.txt)

ygl176, 2024-07-02:
BT stop,but not crash log.
[bt_not_crash_log.txt](https://github.com/user-attachments/files/16061464/bt_not_crash_log.txt)

ESP-YTGerd, 2024-07-02:
> BT stop,but not crash log. [bt_not_crash_log.txt](https://github.com/user-attachments/files/16061464/bt_not_crash_log.txt)

Hi, @ygl176  The issue `ASSERT_PARAM(1024 0)` was fixed with commit: https://github.com/espressif/esp-idf/commit/369c297e23b83bae44662ab28511b2d7a7683bd9 and MR was merged in branch release/v5.2.

ygl176, 2024-07-04:
Hi,  Wifi crash problem persists. 
I've found that if http is unstable, it's prone to crashes.
Normal should print :
>W (1371188)HTTP_STREAM: No more data,errno:0, total_bytes:12064800, rlen = 0.

ygl176, 2024-07-10:
Hi,I haven't used esp32_timer, ’aaTx‘ may be generated by adf.
Thanks for your reply.

ygl176, 2024-07-10:
> components\bt\host\bluedroid\btc\profile\std\a2dp\btc_a2dp_source.c:    a2dp_source_local_param.btc_aa_src_cb.media_alarm = osi_alarm_new("aaTx", btc_a2dp_source_alarm_cb, NULL, BTC_MEDIA_TIME_TICK_MS);

Hi, I've found ‘aaTx' and I'll try again. Thanks.

ESP-YTGerd, 2024-07-10:
Hi, @ygl176
The `aaTx` timer is a periodic timer used by the A2DP source to trigger the retrieval of audio data from the application layer, which is then encoded and sent.

It appears that the `aaTx` timer callback execution is taking too long. A possible reason is that the workqueue of the btc_task is full, causing the esp_timer callback to be blocked when attempting to send a message to the btc_task.

So can you please help test again and provide the following data?
1): Once the callback function `alarm_cb_handler(struct alarm_t *alarm)` is called, please get the depth of the btc_task workqueue. This depth can be obtained by calling `uxQueueMessagesWaiting(wq->queue)` within the `osi_thread_work_queue_put` function.

2): Each time the function `btc_a2dp_source_alarm_cb` is called, please provide the percentage of total runtime for each task in the system. This data can be obtained by calling the function `vTaskGetRunTimeStats(pcWriteBuffer)`, which will write the data into `pcWriteBuffer`. To use this function, you need to enable `portCONFIGURE_TIMER_FOR_RUN_TIME_STATS` in the configuration menu before compiling the project by running `idf.py menuconfig`. [GetRunTimeStats](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos_idf.html?highlight=configgenerate_run_time_stats)
[image]

ygl176, 2024-07-11:
@ESP-YTGerd 
Hi, do you mean that prints TaskGetRunTimeStats every 30ms. 'aaTx' cycle is 30ms, there are too many log messages.

ESP-YTGerd, 2024-07-11:
> @ESP-YTGerd Hi, do you mean that prints TaskGetRunTimeStats every 30ms? `aaTx` cycle is 30ms, there are too many log messages.

Hi, @ygl176 Based on the log you provided, when the aaTx timer triggers the callback execution and the time is consistently 29 seconds for several instances, you can call this function multiple times and check the depth of the work queue at that moment.

ygl176, 2024-07-11:
Hi, @ESP-YTGerd  Thanks for your reply, this is a log that is printed every 3s.
[aatx_log.txt](https://github.com/user-attachments/files/16174096/aatx_log.txt)

ESP-YTGerd, 2024-07-11:
> Hi, @ESP-YTGerd Thanks for your reply, this is a log that is printed every 3s. [aatx_log.txt](https://github.com/user-attachments/files/16174096/aatx_log.txt)
Are these logs only printed when the callback execution time reaches 29 seconds? If not, can we get the corresponding execution times for them?  Thanks~

ygl176, 2024-07-19:
[aatx_log.txt](https://github.com/user-attachments/files/16310490/aatx_log.txt)
Hi, @ESP-YTGerd only print when the callback execution time reaches 29 seconds.

ygl176, 2024-07-25:
@ESP-YTGerd ,Hi,Can this log help you solve the problem? If there is any progress, can you notify me in a timely manner?
Thanks.

ESP-YTGerd, 2024-08-02:
Hi, @ygl176 
Sorry for the late reply, I reviewed the source code you provided and found that in `a2dp_stream_init`, the data retrieval callback `bt_a2d_source_data_cb` is registered. This callback reads data from the ring buffer in a blocking manner because the `audio_element_set_input_timeout` function in `audio_element_init` sets the data read wait time to `portMAX_DELAY`.

Seems the call chain is as follows:

esp_timer (triggered every 30ms) -> `alarm_cb_handler` -> `btc_a2dp_source_alarm_cb` -> `bt_a2d_source_data_cb` -> `rb_read`. 
In `rb_read`, if no data is read, the timer callback blocks until data is read from the ring buffer.
So could you please test the following modification? In the `audio_element_init function`, change the second parameter of `audio_element_set_input_timeout` from `portMAX_DELAY` to `0`. Then, check if the aaTx callback execution time (cb_exex_time) is still relatively long during the test.

ygl176, 2024-08-07:
Hi, @ESP-YTGerd ,
I called `audio_element_set_input_timeout(bt_stream_writer, 0)` when I created the bt_stream_writer. Although the `aatx` is back to normal, the audio playback is very stuttering.Therefore, I removed the audio_element_input from the interrupt, and now the function is working properly.

ygl176, 2024-08-07:
But now it still occasionally prints E0ISR. Would this cause any other problems?

closed 2024-08-14 by ygl176