espressif/esp-idf #14025: Bug after 1st stage bootloader

TheElectronicMusicProject, 2024-06-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v2.1-dirty

### Espressif SoC revision.

ESP32-D0WD (revision v1.0)

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-WROOM-32D

### Power Supply used.

External 3.3V

### What is the expected behavior?

We are using an **ESP32-WROOM-32D** on a custom board, programmed with a firmware written with **ESP-IDF 2.0** (no custom bootloader).  

In a certain number of boards, the ESP32 works while in others it doesn't work. The ESP32 in these boards is the same one, with the same codes on the package.

### What is the actual behavior?

In a certain number of boards, the ESP32 works while in others it doesn't work. The ESP32 in these boards is the same one, with the same codes on the package.

### Steps to reproduce.

The peculiarity of our problem is due to the fact the ESP32 acts in a very strange way: the first stage bootloader starts correctly, but after that we cannot see what's after the 2nd stage bootloader because the baud-rate **changes from 115200 bps to something like 175400 bps**.

### Debug Logs.

```plain
This is the output we get when starting one module which doesn't work:

14:41:57.871> ets Jun  8 2016 00:22:57
14:41:57.871>
14:41:57.871> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
14:41:57.871> configsip: 0, SPIWP:0xee
14:41:57.871> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
14:41:57.871> mode:DIO, clock div:2
14:41:57.871> load:0x3fff0008,len:8
14:41:57.871> load:0x3fff0010,len:4416
14:41:57.871> load:0x40078000,len:11072
14:41:57.871> load:0x40080000,len:252
14:41:57.871> entry 0x40080034
14:41:57.871> EÂ¢â€ºÃpN@Ã‰y.!Â¤Ã (Â¢PWk)â€¡Â­z5QhÃ´Ã¶Ã©yÂ®yÂ¨v{%)eÂ¢Â»Ã­)(Pn@Ã©y.!Ãº~zYÂ¨ÃºYh5â€¹/9U%)eÂ¢Â»Ã­)(Po@Ã©y.!DrÃŠzY(ÃšjpiÃ¨|Ã—+z5YmvpkÃ¾)eÂ¢Â»Ã­)(Pl@Ã©y.!	jÃ¶Â§(!!59	aeÂ¢eaÃ¾5Â«jAÃ¨5%y.!	j	Q*!!!5	eÃŠâ€°Ãª eÂ¢Â»Ã­)(`n@Ã©y.!	j
I-Aea*59ÃŠâ€°Ãª eÂ¢Â»Ã­)(ho@Ã©y.!Ã†â€¹zâ€¡Â«AfpYmÃ¾)eÂ¢Â»Ã­)(H,!%y.!ÂºÂ¨IÃ©|!!!!!!Â¦Ã²Â§(!!!!â€¡vÂ£Ã´â€ºt!	Dvâ€¡eÂ¢eaÃ¾5Â«jAh5"!Ãº~-=eZ)Ã„R!!!11â€¹,111!11E,â€°Ãª eÂ¢Â»Ã­)(HÂ®!%y.!11igqÂ¨P!!!!)Pâ€¡nP!!!!1111-1!11â€š1â€°Ãª eÂ¢Â»Ã­)(HÃ¯!%y.!5-$Ã´A,!!!!!â€¡Â©~Ã›ÃpP!!1!111â€š111111â€°Ãª eÂ¢Â»Ã­)(@l!%y.!Dt-lpâ€¡ny+9Ã¼W%)eÂ¢Â»Ã­)(@Â¬!%y.!â€ mpiÃ«Pif)â€žÃ¼ÃÂ·.~z5YmvpkÃ¾)eÂ¢Â»Ã­)(@m!%y.!yÂ¨zYhpPRÂ«ZÂ§â€¹AË†ÃQKV1â€šYÃÃ¡EÂ¢â€ºÃxM%YÂ»Ã–Yâ€¹P1DTÃ›,P1â€š1â€š!â€¡ÂªÃ°1RÂ·)115â€¦Ã¶1Ã%$P@Ã¨19)IjÃ¾)eÂ¢Â»Ã­)(xN%YÂ»Ã–Yâ€¹Ã1DTÃ»P1â€š%Å¡Pâ€¡Å ÃTÃ›Ã˜gPâ€š%Ë†yË†ÃŠâ€°ÃŠ<0>eÂ¢Â»Ã­)(X.!%y.!â€ºÃ¶YÂ«PÃ«(rÂ§ÃPâ€š-GË†Râ€¡ÃTE5Â¥Ã–1ÃElP@-1u()IjÃ¾)eÂ¢Â»Ã­)(X,!%y.!â€ºÃ¶YÂ«ÃÃ«(rÂ§Ã­1P1Â¢1eË†Râ€¡ÃTÅ 5Â¥Ã–P)IJÃž)!EÂ¢â€ºÃp%YÂ»Ã–Yâ€¹PRâ€¡ÃPâ€š=DTÃ»PÃ›Ã˜gPU@ÂÃ£â€°ÃŠ<0>EÂ¢Â»ÃpÂ%YÂ»Ã–Yâ€¹ÃRâ€¡ÃPâ€¹=DTÃ»RGâ€šÃ›Ã˜gPU@Â)â€žÃŠâ€°ÃŠ<0>EÂ¢â€ºÃp%Yâ€ºÃ¶YÂ«PU(rÂ§Ã­1P1Â«Ã¯1=DtÃ›,P111!Ã›Ã¸g,Ã1Æ’!@m5â€¹(â€°Ãª eÂ¢Â»Ã­)(PÂ¬!%y.!â€ºÃ¶YÂ«ÃU(rÂ§Ã­1P1â€¹l1=DtÃ›,T1Gl15â€¦Ã¶1Râ€ºp!@Â¯9e()â€žÃŠâ€°Ãª eÂ¢Â»Ã­)(Xl!%Ã¶Ã®â€¡Â­XÃ»ÃÅ v9ÃœyÃ9aÃ¾5Â«jAÃ¨5Ã›hÃ°Ã·Â­râ€¹Xâ€¡Â­zYhpÃÅ v@ÂªQmPÃ¾q.!â€¹,T11Ã¤1yÃ9aÃ¾5Â«jA(!%Ã¶Ã®â€¡Â­Xp%v9ÃœyÃ9aÃ¾5Â«jAÃ¨9(Y(Ã®|yÃ©ÃŽâ€žrâ€¹Â inIjaÂ¨Y+5DÃÃ”ZIâ€°YË†^â€¹JÃœÃžIH\YIÃ”Yâ€°ÃŠ<0>EÂ¢â€ºÃx!ÃÃ¦QKÃŽÅ¡ÃÃ‹HTâ€“â‚¬Ã„)dP`HIA`Ë†Ã€ÃŽYÃAÃžâ€¹JAÃˆ!ÃÃ¦QKÃŽÅ¡ÃÃ‹HTâ€“%eË†YBTâ€š%%Zâ€¹Ã†Â¥)!14:41:58.091> EÂ¢â€ºÃ`L!ÃÃ¦QKÃŽÅ¡ÃÃ‹HTâ€“EË†YPÃ‚Ã˜Ã€`Â¨ÃžÃ )%)eÂ¢Â»Ã­)(`.!ÃÃ¦QKÃŽÅ¡ÃÃ‹HTâ€“â€šâ€šY"BhLÃ˜Ã€`Ë†Ãž!Ã€))!EÂ¢â€ºÃ`O!ÃÃ¦QKÃŽÅ¡ÃÃ‹HTÃ˜)dPh%%Zâ€¹<0>Ã†Â¥)!EÂ¢â€ºÃ`%Ã–Ã®â€¡ÂX1Ã»ÃÅ V5â€¡ÂÃÃ—ÃÃºÃ”w)!EÂ¢â€ºÃhM%Ã–Ã®â€¡ÂX5â€¡ÂZYHRYÃŠÃœÃ—ÃˆÃŠAÃ†()!EÂ¢â€ºÃ`%Ã–Ã®â€¡ÂX5â€¡ÂZYHRYÃŠÃ¼Ã—Ã¨qh`f(%)eÂ¢Â»Ã­)(@,!-ÃºaÃ¨Ã 9%ÃqÃ¬Ã´Â¶-1ÃY.vÂ§+!9
vmza(9FÃ¼Ã‡-1PÃ¦Ã±
```

In the board that works correctly, we see:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:4416
load:0x40078000,len:11072
load:0x40080000,len:252
entry 0x40080034
I (45) boot: ESP-IDF v2.1-dirty 2nd stage bootloader
I (45) boot: compile time 12:07:56
I (45) boot: Enabling RNG early entropy source...
I (60) boot: SPI Speed      : 40MHz
I (72) boot: SPI Mode       : DIO
I (85) boot: SPI Flash Size : 4MB
```
```

### More Information.

In the past, with the same boards and the same firmware we've never seen such a problem in so many modules!

What we've tried so far with a working board (A) and a not working board (B):

- erase the flash in both of them (A/B) and flash the same firmware. We read the flash content through the esptool and the contents are identical
- read the baudrate with an oscilloscope on board A. When it prints garbage, it's working with a baud of something like 175400 bps
- using the x-ray we didn't see any unconnected component inside the case of the ESP32, in both A and B

We don't understand why the second stage bootloader is not starting correcly.  

Do you have any suggestion?

o-marshmallow, 2024-06-20:
Hello @TheElectronicMusicProject ,

IDF v2.1 is a very old version that is already in end-of-life state, is it a project that you initiated a long time ago or is it a new project?

You say that the baudrate changes after the first stage bootloader, so it means that if you switch the baudrate of your monitor, the logs are properly shown? Or is the second stage bootloader failing at some point and the change of baudrate is just a side effect?

If the chips are exactly the same, and so is the binary flashed, there is no apparent reason for this change in baudrate.

TheElectronicMusicProject, 2024-06-20:
Hi @o-marshmallow ,
> IDF v2.1 is a very old version that is already in end-of-life state, is it a project that you initiated a long time ago or is it a new project?
Yes, it is an old project and I'm checking if newer ESP32 modules are compatible with this old version.

> You say that the baudrate changes after the first stage bootloader, so it means that if you switch the baudrate of your monitor, the logs are properly shown? Or is the second stage bootloader failing at some point and the change of baudrate is just a side effect?
Yesterday I checked it through an oscilloscope which is able to detect custom baudrates. Setting 175.4 kbps I saw garbage and the oscilloscope report an unstable baudrate (between about 160kbps and 185kbps), hence it isn't able to translate all the messages.
> If the chips are exactly the same, and so is the binary flashed, there is no apparent reason for this change in baudrate.
Exactly! But I need to find the reason and I continue investigating. I'll let you know if I discover something significant.  

If you have further suggestion please let me know

o-marshmallow, 2024-06-21:
Hello @TheElectronicMusicProject ,

Thanks for all these details! The earlier revisions of ESP32 modules had a crystal clock (XTAL) of 26MHz. This was later dropped and the newer ones have a 40MHz clock. I suspect your IDF v2.x based app and bootloader are compiled with 26MHz XTAL frequency in the `menuconfig`.

Could you try changing the `ESP32_XTAL_FREQ` configuration to 40MHz in your v2.x IDF bootloader/application build and flash it again?

Also, you may want to disable `ESP32_XTAL_FREQ_AUTO` configuration

TheElectronicMusicProject, 2024-06-21:
Hi @o-marshmallow and thank you for your technical support.

> Could you try changing the `ESP32_XTAL_FREQ` configuration to 40MHz in your v2.x IDF bootloader/application build and flash it again?
> Also, you may want to disable `ESP32_XTAL_FREQ_AUTO` configuration

We solved the issue!
As you said, there is a bug with the XTAL frequency.

When analyzing the bootloader, I found the following NOTE from the [official documentation](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/kconfig.html#config-xtal-freq-sel):
>Note: The XTAL_FREQ_AUTO option allows the ESP target to automatically estimating XTAL clock's operating frequency. However, this feature is only supported on the ESP32. The ESP32 uses the internal 8MHZ as a reference when estimating. Due to the internal oscillator's frequency being temperature dependent, usage of the XTAL_FREQ_AUTO is not recommended in applications that operate in high ambient temperatures or use high-temperature qualified chips and modules.

Fixing the XTAL frequency, the `sdkconfig` became from:
```
# CONFIG_ESP32_XTAL_FREQ_40 is not set
# CONFIG_ESP32_XTAL_FREQ_26 is not set
CONFIG_ESP32_XTAL_FREQ_AUTO=y
CONFIG_ESP32_XTAL_FREQ=0
```

to:

```
CONFIG_ESP32_XTAL_FREQ_40=y
CONFIG_ESP32_XTAL_FREQ_26=
CONFIG_ESP32_XTAL_FREQ_AUTO=
CONFIG_ESP32_XTAL_FREQ=40
```

and now it works without any kind of block.

I think we can now close this ticket. Thank you again!

closed 2024-06-21 by TheElectronicMusicProject

o-marshmallow, 2024-06-24:
Glad to hear it worked, thanks for the confirmation!