espressif/esp-idf #14028: timing exceedingly inaccurate to the point of unusable.

ulao, 2024-06-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I choose to use the esp-idf because I very much like to ovoid overhead and framework so arduino is out for me.  On the stm32 plat form I could by pass HAL to do what I needed. In the end I found HAL was rather helpful and regretted not using it. So with esp32 I want to use a frame work but also have control. I do a lot of specific timing. VERY specific. In some of my designed 40nS is to much so AVR is not working at 16 MHz.  

There are many things I do not get but first understand I based my first project from the usb host. I will in the end be using it, but I currently blocked the code. 

My goals is to do something like
 xTaskCreatePinnedToCore (my_code,	"",	4096, (void *)1, 1, NULL, 1);
 xTaskCreatePinnedToCore (usb_host_app_main,	"",	4096, (void *)1, 1, NULL, 0);

So for the understanding, the config is set up for USB host, as I'm sure that will be relevant when reading my questions. My issue may all be related  to understanding.  I'm using the command pallet in platform.io to do my config and I just now found out about SDKconfig.h. I'm not sure where my setting are but changing sdkconfig is pointless as it over rights its self each build. 

My concerns

1) if I make this code
 portENABLE_INTERRUPTS();
GPIO.out_w1tc = DEBUG_PIN; 
GPIO.out_w1ts = DEBUG_PIN; 
portDISABLE_INTERRUPTS();

my scope shows
[image]

I know we are not talking about 1 clock cycle here like AVR, but I'm expecting more like 20 ns here. @ 240 MHz  I should get 4ns per clock. this means it takes 8 clocks per each line.  Is there any way to shorten that?

I tried change some of the config and nothing see to change this any. 

 compiler opt didnt help
[image]

lots of speed setting 
 
[image]

https://community.platformio.org/uploads/default/original/3X/8/d/8dcc93ab88751e6e0f0b863faa14e59498865cfb.png

tried running on other CPU's and play with cpu settings
[image]

I need to get my debug down to a very small reaction time. 

2) Delays, just flat out do not work. 

GPIO.out_w1tc = DEBUG_PIN; portENABLE_INTERRUPTS();
vTaskDelay(1 / portTICK_PERIOD_MS);
GPIO.out_w1ts = DEBUG_PIN; portDISABLE_INTERRUPTS();

[image]

I fully get that accuracy is not a goal here but come-on ? 

And to make it worse. what is going on here?
vTaskDelay(2 / 1000 ); = 1.5.3~ ms
vTaskDelay(2 / 100 ); = 1.5.3~ ms
vTaskDelay(2 / 10 ); = 1.5.3~ ms
vTaskDelay(2 / 4 ); = 1.5.3~ ms
vTaskDelay(2 / 3 ); = 1.5.3~ ms
vTaskDelay(2 / 2 ); = 4.74~ ms
vTaskDelay(2 / 1 ); = 11.53~ ms

should portTICK_PERIOD_MS define out to 100?
[image]

ulao, 2024-06-20:
UPDATE I just leaded the IO buss is 80MHz. that explains it a bit but can I get better here? Is the a DMA way?

tore-espressif, 2024-06-20:
Hello @ulao , here are some pointer that might help you

> I just now found out about SDKconfig.h. I'm not sure where my setting are but changing sdkconfig is pointless as it over rights its self each build.

Yes, this file is generated based on your PlatformIO settings, do not edit it manually.

> vTaskDelay(2 / 10 ); That will make 0 ? does that mean 10 is the lowest I can send?

Yes, FreeRTOS delays work with the FreeRTOS scheduler timer which is 10ms resolution by default. These delays are used for delays in range of seconds and higher. This delay allows other tasks to run. For blocking delays you can use [esp_rom_delay_us()](https://docs.espressif.com/projects/esp-idf/en/v5.2.2/esp32/api-reference/system/internal-unstable.html#_CPPv416esp_rom_delay_us8uint32_t), which busy-blocks with micorsecond resolution

If you need precise (microsecond range) timing you can use [esp_timer](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/esp_timer.html) component. This is useful if you want to have i.e. precise periodic interrupts

Here are some tips about assembly code that can speed up the GPIO level change: https://esp32.com/viewtopic.php?t=27963

However, if you don't want to manually manipulate the pin, you can use dedicated hardware with DMA that generates a waveform. You can have a look at [RMT](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/rmt.html) peripheral. It's very flexible but it might be a little more difficult to set up

o-marshmallow, 2024-06-20:
As a side note, if you are using an ESP32-S3 or any RISC-V based chip, there is dedicated GPIO support, which lets you toggle some configurable GPIOs as fast as the CPU frequency. There is even on example on generating a UART signal using it: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/dedicated_gpio

ulao, 2024-06-20:
I think the uart example uses the bundles trick explain in the link above, I saw this link before and could not make it work. The uart example is even more complicated so that maybe not help. Yes I do use the s3, and this is what I need. Can you confirm you are referring to the bundled trick explained in this link? 
https://esp32.com/viewtopic.php?t=27963
if so I will concentration on that.

songruo, 2024-12-18:
@ulao Sorry for the late reply.

> Can you confirm you are referring to the bundled trick explained in this link?
> https://esp32.com/viewtopic.php?t=27963

Yes, please try using the dedicated GPIO feature.

> what do I use to edit the command "ESP-IDF: SDK Configutations Editor (MenuConfig)"

You can refer to https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/kconfig/project-configuration-guide.html#project-configuration-guide.

closed 2024-12-18 by Alvin1Zhang