espressif/esp-idf #14030: I2C NACK causes library to always return ESP_ERR_INVALID_STATE, solved by printf

BartolHrg, 2024-06-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2-169-g8e82062d41

### Espressif SoC revision.

wokwi ESP32

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

wokwi

### Power Supply used.

USB

### What is the expected behavior?

I2C functions (like `i2c_master_transmit`) should succeed after a few attempts.
I have a custom chip in Wokwi simulator that randomly returns NACK (1 out of 5 times: `rand()%5==0`)

### What is the actual behavior?

They always return ESP_ERR_INVALID_STATE.
After the custom Wokwi slave chip fails once, it then fails every time after that.
Even if I call the function 100 times after that.
But, **if I add printf inside i2c_master.c, it works**. <- this is important
(note: therefore it is probably not because of Wokwi or custom chip, because they are not dependent on i2c_master.c)

Also, function on_connect on custom chip is never called again, only on_disconnect

### Steps to reproduce.

1. Call i2c function (e.g. i2c_master_transmit)
2. The custom chip fails (NACK) (random 1 out of 5 times)
3. call it a few more times
4. return code after that is always ESP_ERR_INVALID_STATE

[gitbug.zip](https://github.com/user-attachments/files/15911108/gitbug.zip)

### Debug Logs.

```plain
not much to log
for `printf("%d: 0x%x", __LINE__, ret_error)`

66: 0
67: 0
69: 0
66: 0
67: 0
69: 0
57: 103
58: 103
59: 103

chip has following logs:
Hello from custom ALU chip!
on connect
1
I2C Process connected
PAR1 := 0.000000
PAR1 := 0.000000
PAR1 := 0.000000
PAR1 := 3.000000
I2C Process disconected!!!
on connect
2
I2C Process connected
Operation := 5
I2C Process disconected!!!
on connect
1
I2C Process connected
Sending rez [9.000000] [0][0]
Sending rez [9.000000] [1][0]
Sending rez [9.000000] [2][16]
Sending rez [9.000000] [3][65]
I2C Process disconected!!!
on connect
1
I2C Process connected
PAR1 := 3.000000
PAR1 := 3.000000
PAR1 := 4.000000
PAR1 := 4.000000
I2C Process disconected!!!
on connect
4
I2C Process connected
Operation := 5
I2C Process disconected!!!
on connect
1
I2C Process connected
Sending rez [16.000000] [0][0]
Sending rez [16.000000] [1][0]
Sending rez [16.000000] [2][128]
Sending rez [16.000000] [3][65]
I2C Process disconected!!!
on connect
0
Device not ready. Please repeat commad
I2C Process disconected!!!
I2C Process disconected!!!
I2C Process disconected!!!
I2C Process disconected!!!
I2C Process disconected!!!
I2C Process disconected!!!
I2C Process disconected!!!
I2C Process disconected!!!
I2C Process disconected!!!
I2C Process disconected!!!
I2C Process disconected!!!
I2C Process disconected!!!
I2C Process disconected!!!
I2C Process disconected!!!
I2C Process disconected!!!
I2C Process disconected!!!
I2C Process disconected!!!
```

### More Information.

Important note: when adding `printf("staa = %x   ", i2c_master->status);` into i2c_master.c, it works.
I suspect it may have something to do with `atomic_store` operation
it may be that the compiler does out-of-order execution and the status is not assigned a value until after `s_i2c_send_commands` is called

mythbuster5, 2024-06-20:
where you call the printf? And where you did the print?

BartolHrg, 2024-06-20:
I included modified i2c_master.c in zip file
though, printfs are commented
I put them inside `s_i2c_transaction_start` function around `s_i2c_send_commands` call

For printf inside my project, they are after 5 calls to `i2c_master_transmit`/`i2c_master_receive` (last one is logged) (I put 5 since NACK is random, so after 5 times, it should be ACK)

Optimization level is debug -Og (in vscode menuconfig) (I didn't touch it).

AxelLin, 2024-08-27:
> where you call the printf? And what is the optimization level?

@mythbuster5 
Any progress for this issue?

AxelLin, 2024-10-13:
> I think I am also experiencing this behaviour, with a real I2C device. I have not yet traced the I2C bus to verify there is actually a NACK occurring (I will do more detailed testing soon), but I sometimes get the following debug messages for the first I2C transaction:
> 
> ```
> E (1674) i2c.master: I2C hardware NACK detected
> E (1694) i2c.master: I2C transaction unexpected nack detected
> E (1704) i2c.master: s_i2c_synchronous_transaction(870): I2C transaction failed
> E (1714) i2c.master: i2c_master_transmit(1072): I2C transaction failed
> ```
> 
> **After getting this NACK, every subsequent I2C transaction fails with an ESP_ERR_INVALID_STATE error:**

@mythbuster5 Do you have fix for this?

RogerDavisWork, 2024-12-06:
I am having the exact same problem.  I am using the P4 eval kit and it occurs sometimes when I enable the touch screen.
     BSP_NULL_CHECK(disp_indev = bsp_display_indev_init(disp), NULL);
which is this code in the esp32_p4_function_ev_board.c file:
static lv_indev_t *bsp_display_indev_init(lv_display_t *disp)
{
    esp_lcd_touch_handle_t tp;
    BSP_ERROR_CHECK_RETURN_NULL(bsp_touch_new(NULL, &tp));
    assert(tp);

    /* Add touch input (for selected screen) */
    const lvgl_port_touch_cfg_t touch_cfg = {
        .disp = disp,
        .handle = tp,
    };

    return lvgl_port_add_touch(&touch_cfg);
}

When it happens it soft reboots and does it over and over again forever.
When I hard power cycle the board, it clears it about 50% of the time and then the board works perfectly.

GCTechnology, 2025-02-17:
Any possible solutions? Same issue on https://github.com/espressif/esp-adf/issues/1334.

vandy, 2025-03-11:
I have something similar but also the code crashes.

I'm using two devices each on its own I2C port (`I2C_NUM_0` and `I2C_NUM_1` which use different pins).
```
i2c_device_config_t dev_config = {
    .dev_addr_length = I2C_ADDR_BIT_LEN_7,
    .device_address = DEVICE_ADDR,
    .scl_speed_hz = 100000,
};
```
I'm using function to read data from devices:
```
static int readResult(struct device *device) {
    uint8_t buffer[6];
    int error = i2c_master_receive(device->dev_handle, buffer, sizeof buffer, TIMEOUT_MS);
    if (error) {
        ESP_LOGE("APP", "Problems to read data, error: %d", error);
        return error;
    }
    // ...
}
```
Which is called in the loop by a task. I deliberately set low timeout (10 ms), after several successful readings there is always one with the error:
```
2025-03-11 16:42:11 E (30768) i2c.master: s_i2c_synchronous_transaction(920): I2C transaction failed
2025-03-11 16:42:11 E (30768) i2c.master: i2c_master_receive(1236): I2C transaction failed
2025-03-11 16:42:11 E (30778) APP: Problems to read data, error: 259

// and after that code crashes (function readResult returns, some logic in caller is executed)

2025-03-11 16:42:12 Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.
2025-03-11 16:42:12
2025-03-11 16:42:12 Core  0 register dump:
2025-03-11 16:42:12 MEPC    : 0x40804b6c  RA      : 0x40804b1e  SP      : 0x4080fb10  GP      : 0x4080dac4
--- 2025-03-11 16:42:12 0x40804b6c: i2c_ll_read_rxfifo at D:/dev/github.com/esp-idf/components/hal/esp32h2/include/hal/i2c_ll.h:642
2025-03-11 16:42:12  (inlined by) i2c_isr_receive_handler at D:/dev/github.com/esp-idf/components/esp_driver_i2c/i2c_master.c:672
2025-03-11 16:42:12 0x40804b1e: i2c_isr_receive_handler at D:/dev/github.com/esp-idf/components/esp_driver_i2c/i2c_master.c:671

2025-03-11 16:42:12 TP      : 0x40812af0  T0      : 0x0000001f  T1      : 0x2000103c  T2      : 0x00000002
2025-03-11 16:42:12 S0/FP   : 0x40819690  S1      : 0x408196c4  A0      : 0x4080d404  A1      : 0x60005000
2025-03-11 16:42:12 A2      : 0x00000000  A3      : 0x000000a4  A4      : 0x00000000  A5      : 0x00000000
2025-03-11 16:42:12 A6      : 0x60005000  A7      : 0x00000001  S2      : 0x40819964  S3      : 0x00000001
2025-03-11 16:42:12 S4      : 0x00000000  S5      : 0x40819974  S6      : 0x00000000  S7      : 0x00000000
2025-03-11 16:42:12 S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
2025-03-11 16:42:12 T3      : 0x4080d404  T4      : 0x00000900  T5      : 0x00000000  T6      : 0x00000000
2025-03-11 16:42:12 MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
--- 2025-03-11 16:42:12 0x40800001: _vector_table at D:/dev/github.com/esp-idf/components/riscv/vectors_intc.S:54

2025-03-11 16:42:12 MHARTID : 0x00000000
2025-03-11 16:42:12
2025-03-11 16:42:12 Stack memory:
2025-03-11 16:42:12 4080fb10: 0x00000000 0x40819974 0x000000a4 0x00000001 0x40819964 0x00000080 0x40819690 0x40804c42
--- 2025-03-11 16:42:12 0x40804c42: i2c_master_isr_handler_default at D:/dev/github.com/esp-idf/components/esp_driver_i2c/i2c_master.c:710

2025-03-11 16:42:12 4080fb30: 0x00000004 0x40818c14 0x00000001 0x4080347e 0x00000000 0x00000000 0x00000001 0x00000000
--- 2025-03-11 16:42:12 0x4080347e: esp_vApplicationTickHook at D:/dev/github.com/esp-idf/components/esp_system/freertos_hooks.c:36

2025-03-11 16:42:12 4080fb50: 0x00001881 0x8000000b 0x40819ac8 0x40800e42 0x00000000 0x80000008 0x40819690 0x4080c36c
--- 2025-03-11 16:42:12 0x40800e42: shared_intr_isr at D:/dev/github.com/esp-idf/components/esp_hw_support/intr_alloc.c:452
2025-03-11 16:42:12 0x4080c36c: _global_interrupt_handler at D:/dev/github.com/esp-idf/components/riscv/interrupt.c:69

2025-03-11 16:42:12 4080fb70: 0x00000000 0x00000000 0x00000000 0x4080026e 0x6000b000 0x40805f36 0x40805f42 0x00000000
--- 2025-03-11 16:42:13 0x4080026e: _interrupt_handler at D:/dev/github.com/esp-idf/components/riscv/vectors.S:369
2025-03-11 16:42:13 0x40805f36: systimer_ticks_to_us at D:/dev/github.com/esp-idf/components/esp_hw_support/port/esp32h2/systimer.c:15
2025-03-11 16:42:13 0x40805f42: systimer_us_to_ticks at D:/dev/github.com/esp-idf/components/esp_hw_support/port/esp32h2/systimer.c:20

2025-03-11 16:42:13 4080fb90: 0x4080fb8c 0x00000000 0x00000000 0x00000000 0x4080fba4 0xffffffff 0x4080fba4 0x4080fba4
2025-03-11 16:42:13 4080fbb0: 0x00000000 0x4080fbb8 0xffffffff 0x4080fbb8 0x4080fbb8 0x00000001 0x00000001 0x00000000
2025-03-11 16:42:13 4080fbd0: 0x0001ffff 0x00000000 0x00000000 0x00000000 0x00000000 0x4080fbe0 0x00000000 0x00000000
2025-03-11 16:42:13 4080fbf0: 0x00000000 0x4080fbf8 0xffffffff 0x4080fbf8 0x4080fbf8 0x00000000 0x4080fc0c 0xffffffff
[... 7 lines cut ...]
2025-03-11 16:42:13 4080fc90: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
2025-03-11 16:42:13 4080fcb0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
2025-03-11 16:42:13 4080fcd0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
2025-03-11 16:42:13 4080fcf0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
2025-03-11 16:42:13 4080fd10: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
2025-03-11 16:42:13 4080fd30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[... cut ...]
It looks like i2c_master (`i2c_master_receive`) code returns with error, my function returns with error, buffer is destroyed and somehow interrupt is trying to read data to provided buffer.

But if I put `ESP_LOGE(...)` inside [s_i2c_transaction_start](https://github.com/espressif/esp-idf/blob/94cfe394fe94af5e53da8398af5b886016e7fc11/components/esp_driver_i2c/i2c_master.c#L634) when there is a wrong state, crash doesn't happen. It seems like log takes a lot of time which allows interrupt to happen and not corrupt memory. My `i2c_master->status` is always `7` (when error happens) which is timeout. But I don't see any error logs about timeouts which exist in `s_i2c_send_commands`.

AxelLin, 2025-05-06:
@mythbuster5 How is the status of this issue?

mythbuster5, 2025-05-12:
I think for the original issue, it can be solved after this commit https://github.com/espressif/esp-idf/commit/459b75f81a121dc83beb103a10aee8216c657fce

andrew-sz, 2025-06-25:
> IMHO there are two issues:
> 
> * unexpected NAK's
> * IDF driver code verbosely reporting this fact
> 
> Ad 2) sometimes a NAK is unavoidable (e.g. when a slave isn't ready to address, that's it way to say: retry later). So this is not always an ERROR. I don't think the IDF driver should report it on the console, an "error" status return is sufficient. Ad 1) I've seen this phenomenon a few times too. I2C reporting a NAK while no NAK was sent (NAK = failure to pull SDA low (in time)). I think I2C module samples the SDA level exactly when the SCL level has returned to 1. Some slaves may be a bit slow to pull SDA in some cases. For those devices it would be beneficial to have the possibility to skew SDA sampling a few nanoseconds from SCL clock rise.

What is an unexpected NAK?  Should this always be handled in user code?  i.e. device not ready, or something?

eriksl, 2025-06-25:
Exactly. IDF code should not concern nor report about NAK's, as the IDF can't know what's "expected" (or not). It should simply return a **valid** or **invalid** status code.

I read in the development branch an alternative interface is available where you can have much more control over the transaction. I hope it will get into main soon.

andrew-sz, 2025-06-25:
> Exactly. IDF code should not concern nor report about NAK's, as the IDF can't know what's "expected" (or not). It should simply return a **valid** or **invalid** status code.
>
> I read in the [...]

I hope so also!  Until then I will catch exception and handle in my code.

ryancog, 2025-09-06:
I'm seeing similar behavior with crashes after a failed i2c master transaction. I see ESP_ERR_INVALID_STATE returned from s_i2c_synchronous_transaction and shortly after the chip crashes.

This is on esp32s3 on esp-idf v5.4.2. The crash occurs in the ISR handler i2c_master_isr_handler_default, specifically within i2c_ll_read_rxfifo, as seen in @vandy's output above.

ryancog, 2025-09-10:
Thank you @eriksl. For now I'm unable to build v5.5.1 (#17567), but I'll test it out once I'm able to and update here.

sandynomad, 2026-01-05:
Updating my last comment - it seems the ESP32s3 board I was using failed mid-test hence the results. While trying to diagnose this issue further the vendor demo software now also fails. Testing my original code with a QMI8658 on a different ESP32s3 board works fine.