espressif/esp-idf #14035: Persistent Boot Failure on ESP32-S3

Gspohu, 2024-06-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

### Board

ESP32-S3-WROOM-1U-N4R2

### Device Description

Plain module

### Version

IDF v5.1.4

### IDE Name

PlatformIO

### Operating System

Linux Ubuntu 22.04

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

Hello,

I am experiencing a consistent boot failure issue with my ESP32-S3, regardless of the complexity or simplicity of the firmware I attempt to upload. This issue persists across both my pretty complex custom firmware and basic example code, such as the WiFi connection example. The behavior suggests a fundamental problem related to the boot process.

### Issue Description
The device fails to boot properly. The error message that is consistently output is as follows:

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377989
...
E (681) mmap: esp_mmu_unmap(595): munmap target pointer is outside external memory regions
assert failed: spi_flash_munmap flash_mmap.c:241 (0 && "invalid handle, or handle already unmapped")
Backtrace: 0x40377d1e:0x3fcf4610 ...
ELF file SHA256: fd4bb4dfa7adf1ca
```

My point of view is that it's linked to the model chip I have.

I'm currently stuck on this model of chip, but it's not the first time I've had to unravel memory-related boot problems with this model of chip ESP32-S3-WROOM-1U-N4R2.

### Additional Context
- This issue began after I updated my ESP32-S3 with the latest release available on the GitHub repository, following a period of two months without updates since the device was functioning as needed without them.
- The problem persists even when go to the last commit on the master branch.

This seems to be a regression or a newly introduced bug associated with the latest updates. Any guidance or suggestions on how to mitigate this issue would be greatly appreciated. Thank you for your attention to this matter.
[... 22 lines cut ...]

I took the time to make a more methodical study of the problem, to understand when it first appeared.

So I went back to the version where I was before the problem appeared.
Here's a table summarising my research/tests:
| Date       | Commit                                                                                                                                                                                                                                                                                                                                                 | Build ? | Start ?                    | WiFi ? | WPS ? | Note                             | Note                                                                                                                                                                                        |
|------------|--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------|---------|----------------------------|--------|-------|----------------------------------|---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------|
| 29/04/2024 | platform = https://github.com/platformio/platform-espressif32.git<br>platform_packages =<br>  framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#7e7c01aadf81af799ab296574827d16ddfe8621f<br>  framework-arduinoespressif32-libs@https://github.com/espressif/esp32-arduino-libs.git#b0850286f4be64cb08eeedcf1594cbd5b59ea7f1 | Yes     | Yes                        | Yes    | Yes   |                                  |                                                                                                                                                                                             |
| 02/05/2024 | platform = https://github.com/platformio/platform-espressif32.git<br>platform_packages =<br>  framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#5a023568a640f7b55f40381a28389d05df3af9c5<br>  framework-arduinoespressif32-libs@https://github.com/espressif/esp32-arduino-libs.git#9250b6970c27d2863935fad42fd8d491b4731ff1 | Yes     | Yes                        | Yes    | Yes   |                                  |                                                                                                                                                                                             |
| 07/05/2024 | platform = https://github.com/platformio/platform-espressif32.git<br>platform_packages =<br>  framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#789b1a84258e1b50f40d651be8a3f39226c98753<br>  framework-arduinoespressif32-libs@https://github.com/espressif/esp32-arduino-libs.git#831d0d92c7c998493dbc379798b8efe4c617b300 | Yes     | Yes                        | Yes    | Yes   |                                  |                                                                                                                                                                                             |
| 08/05/2024 | platform = https://github.com/platformio/platform-espressif32.git<br>platform_packages =<br>  framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#e883a2ccfac32b64b50926a3c8a498745a9546de<br>  framework-arduinoespressif32-libs@https://github.com/espressif/esp32-arduino-libs.git#07d6415e1df493c23a3091761be59ece49527313 | Yes     | Yes                        | Yes    | Yes   |                                  |                                                                                                                                                                                             |
| 13/05/2024 | platform = https://github.com/platformio/platform-espressif32.git<br>platform_packages =<br>  framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#0c4b35e07ea696a94de859151c00acbef12eda7f<br>  framework-arduinoespressif32-libs@https://github.com/espressif/esp32-arduino-libs.git#951ade74d7886e1ce931ea46614c4ac47ae3a6c0 | Yes     | Yes                        | Yes    | Yes   |                                  |                                                                                                                                                                                             |
| 22/05/2024 | platform = https://github.com/platformio/platform-espressif32.git<br>platform_packages =<br>  framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#d164df89de350aa945ccb0cb4261f56306bfdb3c<br>  framework-arduinoespressif32-libs@https://github.com/espressif/esp32-arduino-libs.git#8206d4ee579b83a955ba22dfa45588c836522c0b | Yes     | Yes                        | Yes    | Yes   |                                  |                                                                                                                                                                                             |
| 29/05/2024 | platform = https://github.com/platformio/platform-espressif32.git<br>platform_packages =<br>  framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#79b835059f6a2f39b74dde92144909a52617ee5f<br>  framework-arduinoespressif32-libs@https://github.com/espressif/esp32-arduino-libs.git#db132bfa666a5b1705ec810d9addbb8aaede2b33 | Yes     | Yes                        | Yes    | Yes   |                                  |                                                                                                                                                                                             |
[... cut ...]                      |        |       |                                  |                                                                                                                                                                                             |

I come from this issue : https://github.com/espressif/arduino-esp32/issues/9874 , we think that the issue came from the  precompile lib, so I'm here. Did anyone know if there are recent change about PSRAM for the version  IDF v5.1.4 ?

HamzaHajeir, 2024-06-26:
I'm having the same issue!

TL;DR: The issue goes to ESP-IDF after some commit.
- Tested with Arduino 3.0.0 (May 27 2024) - Passed
- Tested with Arduino 3.0.1 (Jun 05 2024) - Passed
- Tested with [custom build](https://github.com/HamzaHajeir/esp32-arduino-lib-builder/releases/tag/11) (Jun 09 2024) - Passed
- Tested with [custom build](https://github.com/HamzaHajeir/esp32-arduino-lib-builder/releases/tag/17) (Jun 13 2024) - Failed
- Tested with Arduino 3.0.2 (Jun 24 2024) - Failed
- Tested with [custom build](https://github.com/HamzaHajeir/esp32-arduino-lib-builder/releases/tag/18) (Jun 25 2024) - Failed

It seems an ESP-IDF commit(s) between (Jun 09) and (Jun 13).

By enabling the correct configuration of PSRAM of I have N8R2, so:
```ini
build_flags = -DBOARD_HAS_PSRAM
	-mfix-esp32-psram-cache-issue
board_build.arduino.memory_type = qio_qspi
```
The errors goes like:
```
E (693) mmap: esp_mmu_unmap(595): munmap target pointer is outside external memory regions

assert failed: spi_flash_munmap flash_mmap.c:241 (0 && "invalid handle, or handle already unmapped") 

Backtrace: 0x40377caa:0x3fcf77b0 0x4037d7d1:0x3fcf77d0 0x403836bd:0x3fcf77f0 0x4202dda7:0x3fcf7920 0x4202d7dd:0x3fcf7940 0x4202d8e1:0x3fcf7960 0x4202dc7b:0x3fcf7990 0x420244b8:0x3fcf79f0 0x42026af5:0x3fcf7a30 0x420d95bd:0x3fcf7a60 0x40380462:0x3fcf7aa0
  #0  0x40377caa in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:466
  #1  0x4037d7d1 in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c:84
  #2  0x403836bd in __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:81
  #3  0x4202dda7 in spi_flash_munmap at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_mmap.c:241
  #4  0x4202d7dd in esp_partition_munmap at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_partition/partition_target.c:173
  #5  0x4202d8e1 in read_otadata at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/app_update/esp_ota_ops.c:97
  #6  0x4202dc7b in esp_ota_get_state_partition at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/app_update/esp_ota_ops.c:836
      (inlined by) esp_ota_get_state_partition at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/app_update/esp_ota_ops.c:824
  #7  0x420244b8 in initArduino at C:/Users/Hamza/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-misc.c:262
  #8  0x42026af5 in app_main at C:/Users/Hamza/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:104
  #9  0x420d95bd in main_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/app_startup.c:208 (discriminator 15)
  #10 0x40380462 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: 97d9d44772f1ba59

Rebooting...
```

cbacott, 2024-07-02:
Just adding a "me too" on this, I just got one of these devices for the first time and spent far too long following the docs to try to get the demo code to actually work. I was finally able to get the demo code to function after downgrading to version 3.0.1, as it defaulted to installing 3.0.2.

While I cannot add anything further in the way of error messages, as mine was identical to the above two, I am simply confirming that this error is present for me as well with PSRAM enabled on version 3.0.2.

GeorgeIoak, 2024-07-07:
> ```ini
> build_flags = -DBOARD_HAS_PSRAM
> 	-mfix-esp32-psram-cache-issue
> board_build.arduino.memory_type = qio_qspi
> ```

I could be wrong but I thought I had read that this was for ESP32 devices, not ESP32-S3 and this was fixed with die rev 3(?) so not needed anymore with new devices. I don't think it will change your results but I thought I would add that point

HamzaHajeir, 2024-08-11:
> I could be wrong but I thought I had read that this was for ESP32 devices, not ESP32-S3 and this was fixed with die rev 3(?) so not needed anymore with new devices. I don't think it will change your results but I thought I would add that point

No problem, and as you said it does not change the result, but a good hit.

Looking forward for someone to look here and to take it seriously.

HamzaHajeir, 2024-09-15:
I've just tested with a custom Arduino lib builder that corresponds to v3.0.4 and it now works. 

It can be a good hint to retest with latest Arduino and recheck.

closed 2025-08-01 by Alvin1Zhang