espressif/esp-idf #14038: Issue in esp_driver_parlio/src /parlio_rx.c parlio_rx_unit_receive and data type for valid_sig_line_id

MichaelLNeil, 2024-06-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

Release/v5.3

### Espressif SoC revision.

ESP32-c6

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

WaveShare ESP32-C6-Zero-M

### Power Supply used.

USB

### What is the expected behavior?

Using PARIO RX to receive an 8-bit wide data path with a Pulse delimiter and valid_sig_line, free running external clock.

### What is the actual behavior?

When calling parlio_rx_unit_receive with a valid_sig_line_id of 2 and a data_width of 8, you get an ESP_RETURN_ON_FALSE assert "the valid_sig_line_id of this delimiter is conflict with rx unit data width".

### Steps to reproduce.

    // Allocate the parlio rx instance
    parlio_rx_unit_config_t parlio_rx_cfg = {
        .trans_queue_depth = 8,     //  ESP_PROBE_DEFAULT_Q_DEPTH
        .max_recv_size = 8 * 4092 , //  ESP_PROBE_DEFAULT_MAX_RECV_SIZE
        .data_width = 8, //  sample_width
        .clk_src = PARLIO_CLK_SRC_EXTERNAL,
        .ext_clk_freq_hz = 14318180,        // 14.31818 Mhz
        .exp_clk_freq_hz = 14318180,        // Set expected clock frequency (i.e., sample rate)
        .clk_in_gpio_num = GPIO_NUM_0,      // 14M 
        .clk_out_gpio_num = GPIO_NUM_NC,    // Use the internal clock for sampling and does not need to output
        .valid_gpio_num = GPIO_NUM_2,       // WNDW
        .flags = {
            .free_clk = true,               //  Free running clock
            .clk_gate_en = false,           //  Not supported on C6
            .io_loop_back = false,          //  No loop back
            .io_no_init = false,            //  Init the GPIOs
        }
    };

    //  Setup data_gpio_nums
    parlio_rx_cfg.data_gpio_nums[0] = GPIO_NUM_1;       
    parlio_rx_cfg.data_gpio_nums[1] = GPIO_NUM_2;       //  Pulse Trigger
    parlio_rx_cfg.data_gpio_nums[2] = GPIO_NUM_3;       //  
    parlio_rx_cfg.data_gpio_nums[3] = GPIO_NUM_4;       //  
    parlio_rx_cfg.data_gpio_nums[4] = GPIO_NUM_5;       //  
    parlio_rx_cfg.data_gpio_nums[5] = GPIO_NUM_18;       //  
    parlio_rx_cfg.data_gpio_nums[6] = GPIO_NUM_19;       // 

    parlio_rx_cfg.data_gpio_nums[7] = GPIO_NUM_NC;

    parlio_rx_cfg.data_gpio_nums[8] = GPIO_NUM_NC;
    parlio_rx_cfg.data_gpio_nums[9] = GPIO_NUM_NC;
    parlio_rx_cfg.data_gpio_nums[10] = GPIO_NUM_NC;
    parlio_rx_cfg.data_gpio_nums[11] = GPIO_NUM_NC;
    parlio_rx_cfg.data_gpio_nums[12] = GPIO_NUM_NC;
    parlio_rx_cfg.data_gpio_nums[13] = GPIO_NUM_NC;
    parlio_rx_cfg.data_gpio_nums[14] = GPIO_NUM_NC;
    parlio_rx_cfg.data_gpio_nums[15] = GPIO_NUM_NC;

    // memcpy(parlio_rx_cfg.data_gpio_nums, config->probe_gpio, PARLIO_RX_UNIT_MAX_DATA_WIDTH * sizeof(gpio_num_t));
    ESP_ERROR_CHECK(parlio_new_rx_unit(&parlio_rx_cfg, &s_ephi->rx_unit)); // , err, TAG, "Failed to allocate the parlio rx unit");

    /*
    // Allocate the software delimiter
    parlio_rx_soft_delimiter_config_t sft_deli_cfg = {
        .sample_edge = PARLIO_SAMPLE_EDGE_NEG,          //  Falling edge of the clock
        .bit_pack_order = PARLIO_BIT_PACK_ORDER_LSB,
        .eof_data_len = 0xFFFF > ((8 * 4092)) ? (8 * 4092) : 0xFFFF,
        .timeout_ticks = 0,
    };

    ESP_ERROR_CHECK(parlio_new_rx_soft_delimiter(&sft_deli_cfg, &s_ephi->deli)) ; // , err, TAG, "Failed to allocate the delimiter");
    */

   // Allocate the software delimiter
    parlio_rx_pulse_delimiter_config_t pulse_deli_cfg = {
        .valid_sig_line_id = GPIO_NUM_2,
        .sample_edge = PARLIO_SAMPLE_EDGE_NEG,          //  Falling edge of the clock
        .bit_pack_order = PARLIO_BIT_PACK_ORDER_LSB,
        .eof_data_len = 560,
        .timeout_ticks = 0,
        .flags.start_bit_included = true,
        .flags.end_bit_included = true,
        .flags.has_end_pulse = false,
        .flags.pulse_invert = true,                     //  WNDW is low when data is valid
    };

    ESP_ERROR_CHECK(parlio_new_rx_pulse_delimiter(&pulse_deli_cfg, &s_ephi->deli)) ; // , err, TAG, "Failed to allocate the delimiter");

    // Register the data receive callback
    parlio_rx_event_callbacks_t cbs = {
        .on_receive_done = s_data_on_recv_callback,
    };

    ESP_ERROR_CHECK(parlio_rx_unit_register_event_callbacks(s_ephi->rx_unit, &cbs, 0)); // 0 is the user data handle // , err, TAG, "Failed to register the receive callback");

    //  Enable
    ESP_ERROR_CHECK(parlio_rx_unit_enable(s_ephi->rx_unit, true)); // , TAG, "Failed to enable the parlio rx unit");
    // ESP_ERROR_CHECK(parlio_rx_soft_delimiter_start_stop(s_ephi->rx_unit, s_ephi->deli, true)); // , TAG, "Failed to start the soft delimiter");
    parlio_receive_config_t recv_cfg = {
        .delimiter = s_ephi->deli,
        .flags.partial_rx_en = true,      // Infinite receiving, use callback to get received data
    };

    ESP_ERROR_CHECK(parlio_rx_unit_receive(s_ephi->rx_unit, s_ephi->payload, (8*4092), &recv_cfg)); // , TAG, "Failed to receive data");

### More Information.

(Code in steps to reproduce, but it is pretty clear to see the issue described below)

In /components/esp_driver_parlio/src/parlio_rx.c parlio_rx_unit_receive (currently line 970)

        /* Check if the valid_sig_line_id is equal or greater than data width, otherwise valid_sig_line_id is conflict with data signal.
         * Specifically, level or pulse delimiter requires one data line as valid signal, so these two delimiters can't support PARLIO_RX_UNIT_MAX_DATA_WIDTH */
        ESP_RETURN_ON_FALSE(recv_cfg->delimiter->valid_sig_line_id >= rx_unit->cfg.data_width,
                            ESP_ERR_INVALID_ARG, TAG, "the valid_sig_line_id of this delimiter is conflict with rx unit data width");

The issue appears to be with the ESP_RETURN_ON_FALSE assert, I think it should be "recv_cfg->delimiter->valid_sig_line_id< rx_unit->cfg.data_width", With the less than instead of greater than or equal to.  

But this also brings up a type issue.  valid_sig_line_id is typed gpio_num_t, which is an enum of GPIO pins.  I believe this should be an int (numerical zero-based index) as it isn't the absolute GPIO number, but is an index into the data GPIO pins assigned to the RX unit.    See PARL_IO_RX_EXT_EN_SEL on page 1238 of the ESP32-C6 Technical Reference Manual (link below).  This is also discussed on page 1225 in section 37.5.4.2 Pulse Enable Mode.

https://www.espressif.com/sites/default/files/documentation/esp32-c6_technical_reference_manual_en.pdf

L-KAYA, 2024-06-24:
Hi @MichaelLNeil , thanks for the feed back!

According to your code, I think you might confuse `parlio_rx_pulse_delimiter_config_t::valid_sig_line_id` with `parlio_rx_unit_config_t:: valid_gpio_num`.

- `parlio_rx_unit_config_t:: valid_gpio_num` is the GPIO for the valid signal, you set it to `GPIO_NUM_2` which is correct.
- `parlio_rx_pulse_delimiter_config_t::valid_sig_line_id` is the data line ID which is supposed to be used as valid signal. As you know, Parlio on C6 supports up to 16 data lines, and your `data_width` is 8, then the first 8 data lines are configured as data bus, then you can pick any one of the other 8 lines as a valid signal. 

In short, the range of the `valid_sig_line_id` in your case is (8, 16], for details, please see the [comment](https://github.com/espressif/esp-idf/blob/master/components/esp_driver_parlio/include/driver/parlio_rx.h#L80) of this field.

MichaelLNeil, 2024-06-25:
Thank you for correcting me.  My mistake.  Sorry about that.

-Mike

L-KAYA, 2024-06-26:
Never mind~ If there's no other questions, you can close this issue, thanks!

closed 2024-06-26 by MichaelLNeil