espressif/esp-idf #14039: MIPI DSI HAL does not have timeouts

robotman2412, 2024-06-20:
### Is your feature request related to a problem?

I have software that tries to initialize a MIPI DSI display, but I'd like for it to still work if the display doesn't. If I physically disconnect the display, ESP-IDF will hang when trying to initialize it, rather than returning an error.

### Describe the solution you'd like.

Add support for setting a timeout value as part of the MIPI DSI and MIPI DSI DBI functions/API.

suda-morris, 2024-06-21:
I think DSI doesn't support hot-plug-in, what's the value in initializing the DSI LCD when there's no LCD connected?

robotman2412, 2024-06-21:
Because my code will run on 3000 devices and not all of them are going to have the display working first try.