espressif/esp-idf #14040: idf.py monitor --no-reset resets ESP32-S3 USB-JTAG

nebkat, 2024-06-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-dev-1031-g91fc98d24d

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

`idf.py monitor --no-reset` should not reset ESP32-S3 connected over USB-JTAG.

### What is the actual behavior?

Device gets reset with or without `--no-reset`.

### Steps to reproduce.

`idf.py monitor --no-reset` an ESP32-S3

### More Information.

Using `python -m esp_idf_monitor --no-reset` directly works as expected.

fhrbata, 2024-06-20:
Hello @nebkat ,

I believe this is expected behavior. Currently, `idf.py monitor` uses a different method for port auto-detection compared to `esp_idf_monitor`. The former attempts to reset the chip during detection, while the latter uses the last device listed by pyserial without resetting the chip. A warning should be shown: `WARNING: --no-reset is ignored. Please specify the port with the --port argument to use this option.` if `idf.py monitor` is executed with the `--no-reset` option and without specifying the port using `-p/--port`. Could you please try to specify the port explicitly?

Thank you

nebkat, 2024-06-20:
Thank you for the prompt reply - that indeed solves it for me.

A warning for other users would probably be good!

fhrbata, 2024-06-20:
Hello @nebkat 

> A warning for other users would probably be good!

I think the following warning should already appear in yellow.

```
WARNING: --no-reset is ignored. Please specify the port with the --port argument to use this option.
```

It might have been moved up with the application output.  Maybe try scrolling up in the terminal to see if it's there.

Thank you

fhrbata, 2024-06-20:
I realize that the warning can be easily missed. Since the `esp_idf_monitor` has its own port detection, albeit different, it might be logical to let it handle port detection when `--no-reset` is used in `idf.py monitor` without an explicitly specified port. Perhaps others might have better suggestions on how this can be improved.

nebkat, 2024-06-20:
Sorry I should be paying more attention... it is indeed there but I'm too used to ignoring the wall of text. Maybe it should actually throw an error instead of a warning until it is resolved? If the user intended not to reset they may be losing important device state to debug.

closed 2024-06-26 by espressif-bot

mjgci, 2024-10-09:
@nebkat 
@fhrbata 

I am seeing the issue is unresolved for me. Using the port option doesn't have any effect and the chipset is restarted with or without the expected options available.

idf.py monitor --no-reset --port COMx   --> Not Working
python -m esp_idf_monitor --no-reset    --> Works

Any suggestions will be helpful. Thanks.

fhrbata, 2024-10-09:
Hello @mjgci , seems to be working for me, even though I tried on linux. Maybe some additional information could help.
From esp-idf environment activated: 
1. What esp-idf-monitor version are you using `pip show esp-idf-monitor`?
2. What is the esp-idf version `idf.py --version`?
3. Also beginning of esp-idf-monitor output when started through `idf.py monitor ...`  may help.

I'm tagging @peterdragun in case he has some idea why this may not be working.

Thank you

mjgci, 2024-10-10:
Hi @fhrbata Thanks for your response.

> 1. What esp-idf-monitor version are you using `pip show esp-idf-monitor`?

```
Name: esp-idf-monitor
Version: 1.4.0
Summary: Serial monitor for esp-idf
Home-page: https://github.com/espressif/esp-idf-monitor
Author: Espressif Systems
Author-email:
License:
Location: C:\Espressif\python_env\idf5.2_py3.11_env\Lib\site-packages
Requires: esp-coredump, esp-idf-panic-decoder, pyserial
Required-by:
```

> 2\. What is the esp-idf version `idf.py --version`?

`ESP-IDF v5.2-dirty`

> 3\. Also beginning of esp-idf-monitor output when started through `idf.py monitor ...`  may help.

[image]

I have a requirement to use IDF v5.2, maybe the problem is fixed in later ones between 5.2 and 5.2.2 as per the original problem description.

fhrbata, 2024-10-12:
Hello @mjgci ,

thank you very much for the additional information. You are right, the chip undergoes reset, but not directly caused by esp-idf-monitor. If the reset is done by esp-idf-monitor, the output would contain messages from the first stage(ROM) bootloader and CPU cycle counter(numbers in parentheses) would start much lower. This means that something else is resetting the chip. It took me a while and I had to reboot to window to understand what's going on and I'm still not 100% sure I got it right :). This problem occurs also on master branch and seems to be limited to USB->UART and Windows. Since this is a different issue than the one originally reported, I filed a new one here https://github.com/espressif/esp-idf/issues/14714.

Thank you very much!

mjgci, 2024-10-17:
Hello @fhrbata ,

Thanks for your further information and help.
Yes, I can confirm this. The first few lines do not show up with esp-idf-monitor as the case with a power-on or a reset.

My device: ESP32-S3-DevKitC-1