espressif/esp-idf #14042: Bootloader after updating the software from OTA, keeps the OTA status to ESP_OTA_IMG_NEW instead of ESP_OTA_IMG_PENDING_VERIFY

AmritpalSinghSarao, 2024-06-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am using 5.1.2 version, and i am facing this issue. The bootloader should convert the status but its not doing it. I am using pre_encrypted_ota.c example. I have set the ENABLE ROLLBACK but its not working.

Alvin1Zhang, 2024-06-21:
Thanks for reporting.

AmritpalSinghSarao, 2024-06-21:
@Alvin1Zhang Is this is known issue?

mahavirj, 2024-06-24:
@AmritpalSinghSarao 

- What is your bootloader build configuration? Could you attach the `sdkconfig` file here.
- Could you please share the verbose log enabled console output here during OTA process?
- Did you check by putting some debug prints around following code:

https://github.com/espressif/esp-idf/blob/0479494e7abe5aef71393fba2e184b3a78ea488f/components/bootloader_support/src/bootloader_utility.c#L389-L395

AmritpalSinghSarao, 2024-06-24:
@mahavirj Thanks for your response.
I resolved the issue by increasing the bootloader size. The bootloader code was not being updated when enable rollback.
Shouldn’t the code not being compiled in case it has not enough memory for bootloader?

mahavirj, 2024-06-24:
@AmritpalSinghSarao 

> The bootloader code was not being updated when enable rollback.
Shouldn’t the code not being compiled in case it has not enough memory for bootloader?

Do you mean the bootloader build part here? If yes, can you please share build output? 
Please supply detailed information on the issue you are facing, otherwise it becomes difficult to comment.

AmritpalSinghSarao, 2024-06-24:
@mahavirj. the following test has been done with CONFIG_PARTITION_TABLE_OFFSET=0x8000, with this config its not working so I had to increase the bootloader to work. 
Following are details of build logs. 
-- Configuring done (2.2s)
-- Generating done (0.1s)
-- Build files have been written to: ************************/build/bootloader
[98/100] Linking C executable bootloader.elf==============================================================================
Bootloader built. Secure boot enabled, so bootloader not flashed automatically.
One-time flash command is:
        *******************************/bin/python  ***********************************-esp-idf-v5.1.2/components/esptool_py/esptool/esptool.py --chip esp32 --port=(PORT) --baud=(BAUD) --before=default_reset --after=no_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 8MB 0x1000 **************************build/bootloader/bootloader.bin
* IMPORTANT: After first boot, BOOTLOADER CANNOT BE RE-FLASHED on same device
[99/100] Generating binary image from built executableesptool.py v4.6.2
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Generated ****************************build/bootloader/bootloader.bin
[100/100] cd ************************************build/bootloader/bootloader.binBootloader binary size 0x6a10 bytes. 0x5f0 bytes (5%) free.
[878/945] Building C object esp-idf/espressif__esp-modbus/...idf_espressif__esp-modbus.dir/freemodbus/modbus/mb_m.c.objIn file included from **********************************managed_components/espressif__esp-modbus/freemodbus/port/port.h:53,
                 from ******************************managed_components/espressif__esp-modbus/freemodbus/modbus/mb_m.c:44:
**********************************managed_components/espressif__esp-modbus/freemodbus/modbus/mb_m.c: In function 'eMBMasterTCPInit':
***************************************managed_components/espressif__esp-modbus/freemodbus/modbus/include/mbconfig.h:194:49: warning: conversion from 'long unsigned int' to 'USHORT' {aka 'short unsigned int'} changes value from '100000' to '34464' [-Woverflow]
  194 | #define MB_MASTER_TIMEOUT_MS_RESPOND            ( CONFIG_FMB_MASTER_TIMEOUT_MS_RESPOND )
*************************************managed_components/espressif__esp-modbus/freemodbus/modbus/mb_m.c:181:37: note: in expansion of macro 'MB_MASTER_TIMEOUT_MS_RESPOND'
  181 |         if (xMBMasterPortTimersInit(MB_MASTER_TIMEOUT_MS_RESPOND * MB_TIMER_TICS_PER_MS) != TRUE)
      |                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
[943/945] Generating binary image from built executableesptool.py v4.6.2
Creating esp32 image...
Merged 27 ELF sections
Successfully created esp32 image.
Generated *******************************firmware-unsigned.bin
[944/945] Generating signed binary imageespsecure.py v4.6.2
Signed 1179568 bytes of data from ***************************build/firmware-unsigned.bin
Generated signed binary image **************************build/firmware.bin from ******************************************build/firmware-unsigned.bin
[945/945] cd********************************************build/es...e.bin *****************************build/firmware.biniot_c.bin binary size 0x11fff4 bytes. Smallest app partition is 0x160000 bytes. 0x4000c bytes (18%) free.

Project build complete. To flash, run this command:
***************************************************/bin/python ../../../../../*****************************************-esp-idf-v5.1.2/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after no_reset --chip esp32  write_flash --flash_mode dio --flash_size 8MB --flash_freq 40m 0x8000 build/partition_table/partition-table.bin 0x50000 build/ota_data_initial.bin 0x60000 build/firmware.bin
or run 'idf.py -p (PORT) flash'

mahavirj, 2024-06-25:
@AmritpalSinghSarao 

> Successfully created esp32 image.
Generated ****************************build/bootloader/bootloader.bin
[100/100] cd ************************************build/bootloader/bootloader.binBootloader binary size 0x6a10 bytes. 0x5f0 bytes (5%) free.

From the above logs, the build part for the bootloader image looks good. It should have the required code for rollback feature present (if enabled in the `sdkconfig`). I hope you are not using some different bootloader image here.

AmritpalSinghSarao, 2024-06-26:
@mahavirj 
I enabled the App rollback. What I am trying to say here is that, when we enable App rollback or some other config in bootloader, memory provided to bootloader might no be enough, but the build run successfully as seems that the code is not compiled correctly in that case. In this case, it becomes difficult to troubleshoot. I resolved that issue by increasing the size of bootloader even though the App rollback was enabled.

mahavirj, 2024-06-27:
@AmritpalSinghSarao 

> What I am trying to say here is that, when we enable App rollback or some other config in bootloader, memory provided to bootloader might no be enough, but the build run successfully as seems that the code is not compiled correctly in that case.

This is quite strange. Ideally if the code can not really fit into the bootloader reserved code segment then build would fail. Do you have any logs where the rollback feature enabled bootloader build was successful but still the functionality (e.g., some debug log) was not present?

mahavirj, 2024-07-05:
Closing the issue. Please create new issue sharing detailed steps if you are still facing any problem.

closed 2024-07-05 by mahavirj