espressif/esp-idf #14044: rtc_gpio_pulldown_en vs gpio_sleep_set_pull_mode

JVKran, 2024-06-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I was wondering what the difference between these two are. Are they interchangable? In what way do the snippets below behave differently?

``` cpp
esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_AUTO);
rtc_gpio_pulldown_en(GPIO_NUM_2);
rtc_gpio_pullup_dis(m_GPIO_NUM_2);
```

``` cpp
gpio_pull_mode_t pull_mode = GPIO_PULLDOWN_ONLY;
gpio_sleep_set_pull_mode(GPIO_NUM_2, pull_mode);
gpio_sleep_sel_dis(GPIO_NUM_2);
```

esp-lis, 2024-06-24:
@JVKran In the ESP32 series chips, IOs are categorized into Digital GPIO and RTC GPIO. The difference between the two is that Digital GPIOs are usually more numerous (which means they occupy a larger circuit area) and are located in the Digital Peripheral power domain (on some chips, this power domain can be configured to power down during SoC sleep to save power). RTC GPIOs are usually fewer in number (which means they occupy a smaller circuit area) and are located in the RTC Peripheral power domain (this power domain can be configured to power down during SoC sleep to save power).

For Digital GPIO input/output, the pull-up/pull-down functions have a set of control logic for both the SoC Active and Sleep states. This allows independent control of Digital GPIOs in different SoC states. For example, most of the time, Digital GPIOs only need to function for input/output or capturing external interrupts in the SoC Active state. During the SoC Sleep state, Digital GPIOs are not needed. In this case, you can configure the Digital GPIO mode in the Active state to input/output and set the corresponding pull-up/pull-down status. In the Sleep state, you can configure the mode to be disabled to prevent leakage current on the IO and save power.

JVKran, 2024-07-17:
Hi @esp-lis,

Thanks for the clarification. So with regards to the two pasted snippets, the difference is that the first one specifies the configuration in active and sleep state, while the last one only configures the state in sleep. 

To make the snippets functionally the same, I would only have to call gpio_config to also configure them in active state right? So these two are functionally the same, in that they both configure a pin to have a pulldown both in active and sleep state. However, the rtc_gpio only works on RTC-gpio's while the gpio api works on all GPIO's.

``` c
esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_AUTO);
rtc_gpio_pulldown_en(GPIO_NUM_2);
rtc_gpio_pullup_dis(m_GPIO_NUM_2);
```

``` c
gpio_config_t config;
config.pull_down_en = GPIO_PULLDOWN_ENABLE;
config.pull_up_en = GPIO_PULLUP_DISABLE;
gpio_config(&config);
gpio_pull_mode_t pull_mode = GPIO_PULLDOWN_ONLY;
gpio_sleep_set_pull_mode(GPIO_NUM_2, pull_mode);
gpio_sleep_sel_dis(GPIO_NUM_2);
```

closed 2024-11-14 by Sherry616