espressif/esp-idf #14045: Command to show dependency tree of components

stkw0, 2024-06-21:
### Is your feature request related to a problem?

We had a somewhat complex image. We were facing some bugs and wanted to reduce it so it can fit into RAM memory and boot it directly from RAM. Using "size", "size-files" and "size-components" we have seen which components contribute more to the final size. The component that affects more, is a dependency of some other components. Hence, having the possibility to show the dependency tree would be very useful.

### Describe the solution you'd like.

Having a new command that shows the dependency tree of components.

ESP-Marius, 2024-07-04:
I think what you are looking for is already possible.

See https://github.com/espressif/esp-idf/tree/master/tools/test_apps/system/g1_components#component-dependencies-graph-component_depsdot for an example where we do this.

stkw0, 2024-07-04:
Oh, thank you very much! I missed that.

closed 2024-07-04 by stkw0