espressif/esp-idf #14047: CMAKE_TOOLCHAIN_FILE 'toolchain-esp32' does not match currently selected IDF_TARGET 'esp32c6'. To change the target, clear the build directory and sdkconfig file, and build the project again.

RJSDevel, 2024-06-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v5.2.2 and esp-idf-v5.3

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Normaly build for C6

### What is the actual behavior?

For H2 build nornally, but this error for c6

### Steps to reproduce.

1. Create any zigbee project 
2. Select target c6

### Build or installation Logs.

```plain
cmake -G Ninja -DCMAKE_TOOLCHAIN_FILE=D:\Espressif\frameworks\esp-idf-v5.2.2\tools\cmake\toolchain-esp32.cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCCACHE_ENABLE=1 -DIDF_TARGET=esp32c6 C:\Users\Developer\espressif\YNXT-CG32WBZ
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.42.0.windows.2") 
CMake Error at D:/Espressif/frameworks/esp-idf-v5.2.2/tools/cmake/targets.cmake:158 (message):
   CMAKE_TOOLCHAIN_FILE 'toolchain-esp32' does not match currently selected IDF_TARGET 'esp32c6'. To change the target, clear the build directory and sdkconfig file, and build the project again.
-- Configuring incomplete, errors occurred!
Call Stack (most recent call first):
  D:/Espressif/frameworks/esp-idf-v5.2.2/tools/cmake/project.cmake:493 (__target_set_toolchain)
  CMakeLists.txt:6 (project)
```

### More Information.

WIN11
[image]

RJSDevel, 2024-06-21:
Executing C:\Users\Developer\AppData\Local\Programs\Python\Python311\python.exe D:\Espressif\frameworks\esp-idf-v5.2.2\tools\idf_tools.py list
* xtensa-esp-elf-gdb: GDB for Xtensa
  - 14.2_20240403 (recommended, installed)
* riscv32-esp-elf-gdb: GDB for RISC-V
  - 14.2_20240403 (recommended, installed)
* xtensa-esp-elf: Toolchain for 32-bit Xtensa based on GCC
  - esp-13.2.0_20230928 (recommended, installed)
* esp-clang: Toolchain for all Espressif chips based on clang (optional)
  - 16.0.1-fe4f10a809 (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - esp-13.2.0_20230928 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.35_20220830 (recommended, installed)
* cmake: CMake build system
  - 3.24.0 (recommended, installed)
  - 3.16.3 (supported)
* openocd-esp32: OpenOCD for ESP32
  - v0.12.0-esp32-20240318 (recommended, installed)
* ninja: Ninja build system
  - 1.11.1 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.3 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 4.8 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.11 (recommended, installed)
* esp-rom-elfs: ESP ROM ELFs
  - 20230320 (recommended, installed)
* qemu-xtensa: QEMU for Xtensa (optional)
  - esp_develop_8.2.0_20240122 (recommended, installed)
* qemu-riscv32: QEMU for RISC-V (optional)
  - esp_develop_8.2.0_20240122 (recommended, installed)
OK

CDT Build environment variables
PATH: D:\Espressif\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin;D:\Espressif\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin;D:\Espressif\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin;D:\Espressif\tools\esp-clang\16.0.1-fe4f10a809\esp-clang\bin;D:\Espressif\tools\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\bin;D:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;D:\Espressif\tools\cmake\3.24.0\bin;D:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin;D:\Espressif\tools\ninja\1.11.1\;D:\Espressif\tools\idf-exe\1.0.3\;D:\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;D:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;D:\Espressif\tools\qemu-xtensa\esp_develop_8.2.0_20240122\qemu\bin;D:\Espressif\tools\qemu-riscv32\esp_develop_8.2.0_20240122\qemu\bin;D:\Espressif\python_env\idf5.2_py3.11_env\Scripts;D:\Espressif\frameworks\esp-idf-v5.2.2\tools;C:\Program Files\Common Files\Oracle\Java\javapath;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\Git\cmd;C:\Program Files\nodejs\;C:\ProgramData\chocolatey\bin;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;D:\TelinkSDK\bin;C:\Users\Developer\AppData\Local\Programs\Python\Python311\Scripts\;C:\Users\Developer\AppData\Local\Programs\Python\Python311\;C:\Users\Developer\AppData\Local\Microsoft\WindowsApps;C:\Users\Developer\AppData\Roaming\npm;C:\Program Files\JetBrains\WebStorm 2023.2.4\bin;;C:\Users\Developer\AppData\Local\Programs\Microsoft VS Code\bin;D:\Program Files\SDCC\bin;C:\Program Files\JetBrains\PyCharm Community Edition 2023.2.5\bin;;C:\Users\Developer\AppData\Local\Android\Sdk\platform-tools;C:\Program Files\JetBrains\IntelliJ IDEA Community Edition 2023.3\bin;;C:\cygwin64\OpenOCD-20231002-0.12.0\bin;D:\TelinkSDK\bin;
USERDOMAIN_ROAMINGPROFILE: DESKTOP
PROCESSOR_LEVEL: 6
SESSIONNAME: Console
ALLUSERSPROFILE: C:\ProgramData
PROCESSOR_ARCHITECTURE: AMD64
ESP_IDF_VERSION: 5.2
PSModulePath: C:\Program Files\WindowsPowerShell\Modules;C:\Windows\system32\WindowsPowerShell\v1.0\Modules
SystemDrive: C:
OPENOCD_SCRIPTS: D:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\share\openocd\scripts
IDF_DEACTIVATE_FILE_PATH: C:\Users\DEVELO~1\AppData\Local\Temp\tmp6zzehc7sidf_1560
IDF_COMPONENT_STORAGE_URL: file:///D:/Espressif/registry;default
IDF_CCACHE_ENABLE: 1
USERNAME: Developer
ProgramFiles(x86): C:\Program Files (x86)
MOSQUITTO_DIR: C:\Program Files\mosquitto
FPS_BROWSER_USER_PROFILE_STRING: Default
PATHEXT: .COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC
DriverData: C:\Windows\System32\Drivers\DriverData
IntelliJ IDEA Community Edition: C:\Program Files\JetBrains\IntelliJ IDEA Community Edition 2023.3\bin
PyCharm Community Edition: C:\Program Files\JetBrains\PyCharm Community Edition 2023.2.5\bin
ProgramData: C:\ProgramData
IDF_COMPONENT_MANAGER: 1
ProgramW6432: C:\Program Files
SRFPROG: C:\Program Files (x86)\Texas Instruments\SmartRF Tools\Flash Programmer\bin
HOMEPATH: \Users\Developer
PROCESSOR_IDENTIFIER: Intel64 Family 6 Model 191 Stepping 2, GenuineIntel
EFC_7684: 1
PYTHON_EXE_PATH: C:\Users\Developer\AppData\Local\Programs\Python\Python311\python.exe
ProgramFiles: C:\Program Files
PUBLIC: C:\Users\Public
windir: C:\Windows
=::: ::\
ZES_ENABLE_SYSMAN: 1
LOCALAPPDATA: C:\Users\Developer\AppData\Local
[... 11 lines cut ...]
OS: Windows_NT
COMPUTERNAME: DESKTOP
IDF_PATH: D:\Espressif\frameworks\esp-idf-v5.2.2
PROCESSOR_REVISION: bf02
CommonProgramW6432: C:\Program Files\Common Files
ComSpec: C:\Windows\system32\cmd.exe
SRFPROG2: C:\Program Files (x86)\Texas Instruments\SmartRF Tools\Flash Programmer 2\bin
IDF_MAINTAINER: 1
TEMP: C:\Users\DEVELO~1\AppData\Local\Temp
SystemRoot: C:\Windows
HOMEDRIVE: C:
[... cut ...]
TMP: C:\Users\DEVELO~1\AppData\Local\Temp
CommonProgramFiles(x86): C:\Program Files (x86)\Common Files
NUMBER_OF_PROCESSORS: 16
IDF_PYTHON_ENV_PATH: D:\Espressif\python_env\idf5.2_py3.11_env
ESP_ROM_ELF_DIR: D:\Espressif\tools\esp-rom-elfs\20230320\

Operating System: windows 11
Java Runtime Version: 21.0.1+12-LTS-29
Eclipse Version: 4.25.0.v20220831-1800
Eclipse CDT Version: 10.7.1.202208160035
IDF Eclipse Plugin Version: 2.9.1.202304060814
ESP-IDF v5.2.2-dirty
Python set for IDF_PYTHON_ENV: Python 3.11.0

RJSDevel, 2024-06-23:
[image]
Fixed with deleting check block

fhrbata, 2024-06-24:
Hello @RJSDevel,

I believe the check is functioning correctly in this case. The CMake is being called with the `esp32c6` target but using the toolchain for `esp32`. These are incompatible because the target is RISC-V and the toolchain is for Xtensa.
The question is why CMake is being called this way from the Eclipse plugin and whether the plugin should check for this, or if this behavior is expected when the user sets incompatible options. Perhaps someone from the Eclipse plugin team
 can provide more insight.

Thank you.

o-marshmallow, 2024-06-25:
Hello @RJSDevel ,

Sorry for the delay of response. you have an inconsistency here, as @fhrbata said, you provide the `esp32` toolchain file (`toolchain-esp32.cmake`) while you are trying to compile for the ESP32-C6. These two have completely different architectures.

If you are trying to build from the command line, as you mentioned in the description of your issue, you simply need to perform the following commands:
```
cd <your_idf_project>
idf.py set-target esp32c6
idf.py build
```

Make sure you follow all the steps here to install the toolchain properly: https://docs.espressif.com/projects/esp-idf/en/v5.2.2/esp32c6/get-started/windows-setup.html

RJSDevel, 2024-06-29:
> Hello @RJSDevel ,
>
> Sorry for the delay of response. you have an inconsistency here, as @fhrbata said, you provide the `esp32` toolchain file (`toolchain-esp32.cmake`) while you are trying to compile [...]

Hi, I tried to build from Espressif-IDE, I think the IDF Plugin has a trubble again

alirana01, 2024-07-10:
@RJSDevel Can you please try to use project full clean and see if the issue still persists. You can also try to update to latest Espressif-IDE if you want it comes with cdt-lsp editor and if that doesn't suit you try to go with a previous version higher than 2.10 as I believe this was fixed without the requirement of using project full clean in later versions.

alirana01, 2024-08-29:
Closing as no further response from user on this and newer versions should resolve this.

closed 2025-03-07 by Alvin1Zhang

mstroven-hiviz, 2025-05-22:
So, there's no fix for this?

kolipakakondal, 2025-05-23:
Hi @mstroven-hiviz Please clean the build directory and try to build the project again by selecting the correct target, see if that helps?