espressif/esp-idf #14049: WiFi Connect should have an AP Connect function

LucasSaintrain, 2024-06-21:
### Is your feature request related to a problem?

In "esp-idf/examples/common_components/protocol_examples_common/" we have some files that help easily connect to interfaces like Ethernet or WiFi.

In "wifi_connect.c" we have the example_wifi_connect() function that in turn calls example_wifi_sta_do_connect() to setup a WiFi STA connection.

The problem is, STA is the only option for example connect. I think it would be very practical and useful to implement a way of easily connecting in AP mode using example_connect().

### Describe the solution you'd like.

The functions example_wifi_ap_do_connect(...) and example_wifi_ap_do_disconnect(...) should be implemented in "wifi_connect.c".

### Describe alternatives you've considered.

I can implement the AP mode myself as usual, but it would be better to just use the function provided in common component.

Xiehanxin, 2024-06-25:
hi  @LucasSaintrain, I think I didn't understand your requirement correctly, because an AP does not go through a connection process,  it is the STA that connects to it. Could you please explain what you want the function "AP connecting" to do?

LucasSaintrain, 2024-06-25:
Hello @Xiehanxin,
You're correct, I missed the point that the AP isn't actually connecting to anyone.

What I meant to ask is for a utility function that works just like the STA connect, but for creating an AP.
This means a simple function call that uses a SSID and Password from some global #defines and uses it to configure the access point.

MaxwellAlan, 2024-09-10:
Hi @LucasSaintrain 

Maybe you can refer to [examples/wifi/getting_started/softAP](https://github.com/espressif/esp-idf/blob/59e18382702b6986be3d3f55e9ac7763c1397cf7/examples/wifi/getting_started/softAP/main/softap_example_main.c#L47)

closed 2025-02-19 by MaxwellAlan