espressif/esp-idf #14050: exclusive use of 1 cpu.

ulao, 2024-06-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Are there any examples or functions that get me closer to exclusively using a cpu? I want all my peripherals and tasks to go on cpu0 and to leave cpu1 as free as possible.  If there are some tasks and interrupts that must run, I wonder if they can be block, disabled, for a period of time. And how to do that

o-marshmallow, 2024-06-24:
Hello @ulao ,

By default, when you run ESP-IDF, the only tasks that will run on App CPU (CPU1) is the Idle Task and the IPC Tasks. No interrupt, apart from the tick and IPC interrupts, will be bound to the Core 1 since no task on that core will use any peripheral. For tasks related to Wifi or Bluetooth, it is possible to choose on which core they have to run thanks to the `menuconfig`.

At runtime, when you create tasks, you have to use `xTaskCreatePinnedToCore` to restrict the core your new Task has to run on.

If you want any of your tasks on any to never be pre-empted by another task, make sure its priority is the highest among all your tasks.

More info about FreeRTOS in ESP-IDF are available in the official documentation: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos.html
and https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/freertos_idf.html

ESP-Marius, 2024-07-12:
Marking this one as resolved, feel free to reopen if there are still any outstanding questions regarding this topic.

closed 2024-07-12 by ESP-Marius

ulao, 2024-10-04:
I'm still waiting on help

ulao, 2024-10-04:
I close the IDE and reran it and it updated a few things but the container update (few gig ) didnt work.

[9194 ms] Exit code 1
[9199 ms] Command failed: /home/ulao/.vscode-remote-containers/bin/dc96b837cf6bb4af9cd736aa3af08cf8279f7685/node /home/ulao/.vscode-remote-containers/dist/dev-containers-cli-0.388.0/dist/spec-node/devContainersSpecCLI.js up --container-session-data-folder /tmp/devcontainers-35be6c34-4fc4-4ec7-9d47-956a240fe1b01728052403842 --workspace-folder /mnt/c/Users/Administrator/Desktop/usb2llapi/usb_host_lib --workspace-mount-consistency cached --gpu-availability detect --id-label devcontainer.local_folder=c:\Users\Administrator\Desktop\usb2llapi\usb_host_lib --id-label devcontainer.config_file=/mnt/c/Users/Administrator/Desktop/usb2llapi/usb_host_lib/.devcontainer/devcontainer.json --log-level debug --log-format json --config /mnt/c/Users/Administrator/Desktop/usb2llapi/usb_host_lib/.devcontainer/devcontainer.json --default-user-env-probe loginInteractiveShell --mount type=volume,source=vscode,target=/vscode,external=true --mount type=bind,source=/run/user/1000/wayland-0,target=/tmp/vscode-wayland-647395a3-e842-4db6-aa6a-f72ef97e9184.sock --skip-post-create --update-remote-user-uid-default on --mount-workspace-git-root --include-configuration --include-merged-configuration
[9199 ms] Exit code 1

now the IDE is broken.