espressif/esp-idf #14051: ESP32H2 Change Console Uart device reboot

wensenl, 2024-06-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-ide-master/esp-idf-v5.3-beta2

### Espressif SoC revision.

ESP32H2

### Operating System used.

macOS

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP Thread Border Router/Zigbee Gateway

### Power Supply used.

USB

### What is the expected behavior?

I change the console log output to USB Serial/JTAG Controller, expected it can output success.
[image]

### What is the actual behavior?

device reboot always.

### Steps to reproduce.

normal build 

### Debug Logs.

```plain
esp32h2 usb output log:
SPIWP:0xee
mode:DIO, clock div:1
load:0x408460f0,len:0x12f4
load:0x4083cad0,len:0xe00
load:0x4083efd0,len:0x2998
entry 0x4083cada
ESP-ROM:esp32h2-20221101
Build:Nov  1 2022
rst:0xc (SW_CPU),boot:0xe (SPI_FAST_FLASH_BOOT)
Saved PC:0x400031b6
0x400031b6: __call_r_ble_util_buf_acl_tx_elt_get in ROM
```

wensenl, 2024-06-22:
the test demo is:
1、esp32s3 run esp-zigbee-sdk-main/examples/esp_zigbee_gateway
2、esp32h2 run esp-idf-master/examples/openthread/ot_rcp

gytxxsy, 2024-07-17:
`ot_rcp` does not support `USB Serial/JTAG Controller`. In fact, if you need to debug ot_rcp and output some logs from ot_rcp to the computer while running the Gateway, we do not recommend using the `ESP Thread Border Router/Zigbee Gateway` board. In this case, we recommend using an ESP32S3 and an ESP32-H2, connecting them manually, and modifying some configurations of `ot_rcp`. As shown below:
[image]
[image]

chshu, 2024-09-14:
Closing the issue as it's answered, @wensenl feel free to reopen if any follow up questions.

closed 2024-09-14 by chshu