espressif/esp-idf #14053: [NimBLE][ESP32s3] GAP disconnect; reason: 0x0228 hci, INSTANT_PASSED

chipweinberger, 2024-06-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.4

### Espressif SoC revision.

SP32-s3

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom pcb

### Power Supply used.

USB

### What is the expected behavior?

A stable BLE connection with my iOS device 17.4.1

### What is the actual behavior?

"GAP disconnect; reason: 0x0228 hci, INSTANT_PASSED" causes disconnection every few minutes.

### Steps to reproduce.

1. establish BLE connection with iOS device
2. wait

### Debug Logs.

```plain
[0;32mI (15836512) pd ble: GAP disconnect; reason: 0x0228 hci, INSTANT_PASSED      gapConnHandle: 2 [0m
 [0;32mI (15836522) pd ble: Connection Description: [0m
 [0;32mI (15836522) pd ble:   role: Slave [0m
 [0;32mI (15836522) pd ble:   gap connection handle: 2 [0m
 [0;32mI (15836522) pd ble:   our id:  Public 00:35:38:64:62:34 [0m
 [0;32mI (15836522) pd ble:   our ota: Public 00:35:38:64:62:34 [0m
 [0;32mI (15836522) pd ble:   peer id:  Public f8:4d:89:7c:1c:7c [0m
 [0;32mI (15836522) pd ble:   peer ota: Public f8:4d:89:7c:1c:7c [0m
 [0;32mI (15836532) pd ble:   connection interval:     12 (15.000ms) [0m
 [0;32mI (15836532) pd ble:   connection latency:      0 times [0m
 [0;32mI (15836532) pd ble:   supervision timeout:     36 (360ms) [0m
 [0;32mI (15836532) pd ble:   Secure - encrypted:      0 [0m
 [0;32mI (15836532) pd ble:   Secure - authenticated:  0 [0m
 [0;32mI (15836532) pd ble:   Secure - bonded:         0 [0m
 [0;32mI (15836532) pd ble:   Secure - keySize:        0 [0m
```

### More Information.

iOS logs:

"didDisconnectPeripheral: The connection has timed out unexpectedly."

chipweinberger, 2024-06-22:
it happens even when the devices are right next to each other.

chipweinberger, 2024-06-22:
previous discussion:

https://github.com/espressif/esp-idf/issues/3877

chipweinberger, 2024-07-02:
@rahult-github @SumeetSingh19 

Have you seen this issue before?

rahult-github, 2024-07-03:
Hi @chipweinberger , 

This error code (instant passed) implies : 

[image]

Can you possibly share the complete console side log to get more information of what operations are being performed when this behaviour is observed . Do you have any OTA log for this case ?

chipweinberger, 2024-07-03:
Yes I will share the next time it happens. As I recall, there was nothing notable in the logs. It happens even after being "idle" for awhile.

Is there anything I should do? my BLE tasks are already highest priority. LL is highest. then normal ble task 2nd highest.

chipweinberger, 2024-07-03:
Here are more logs leading up to it. This time I got `CONN_TERM_LOCAL`. These disconnections happen about every 1-5 minutes.

```
[0;32mI (82197936) NimBLE: GATT procedure initiated: indicate;  [0m
 [0;32mI (82197946) NimBLE: att_handle=37
 [0m
[In-Din]            82197.939 | Cb:0 Ch:0  | Note Off E6 vel:64
[Out-Din]
[Rec ]           35
  [0;32mI (82197966) pd mss: stone bounds 22 5372 [0m
 [0;32mI (82197966) NimBLE: GATT procedure initiated: indicate;  [0m
 [0;32mI (82197966) NimBLE: att_handle=37
 [0m
[In-Din]            82197.999 | Cb:0 Ch:0  | Note Off G4 vel:64
[Out-Din]
[Rec ]           60
 silence
 [0;32mI (82198016) pd mss: stone bounds 22 5376 [0m
 [0;32mI (82198026) NimBLE: GATT procedure initiated: indicate;  [0m
 [0;32mI (82198026) NimBLE: att_handle=37
 [0m
 [0;32mI (82199026) pd mss: all clients up to date. sleeping [0m
 [0;32mI (82201026) pd led ux: recording stoppped [0m
LED: solid r:0 g:0 b:255
[Rec ]       0     82197.999 | Cb:0 File Meta Event | (Meta) Manufacturer Specific, (260 bytes)jmxEof{
	"fileOffset":	5376,
	"totalMillis":	73715,
	"totalNotes":	337,
	"totalMsgs":	749,
	"stoneOffsetPrev":	22,
	"stoneIdxPrev":	1,
	"stoneUuidPrev":	"ff0d6635-2538-6231-22cb-ba265ab6569c",
	"unixtime":	1720040867,
	"localOffset":	-420,
	"tzNotSet":	true
}
[Rec ]       0     82197.999 | Cb:0 File Meta Event | (Meta) Manufacturer Specific, (25 bytes)jmxBlm{"b":294}
[Rec ]       0     82197.999 | Cb:0 File Meta Event | (Meta) End Of Track
 [0;32mI (82201056) pd midi recorder: flushed [0m
 [0;32mI (82201076) pd mss: stone bounds 22 5376 [0m
 [0;32mI (82201086) NimBLE: GATT procedure initiated: indicate;  [0m
 [0;32mI (82201086) NimBLE: att_handle=37
 [0m
 [0;32mI (82202086) pd mss: all clients up to date. sleeping [0m
 [0;32mI (82231086) NimBLE: GAP procedure initiated: terminate connection; conn_handle=3 hci_reason=19
 [0m
 [0;32mI (82231096) pd ble: GAP subscribe event: [0m
 [0;32mI (82231106) pd ble:    gap conn handle: 3 [0m
 [0;32mI (82231106) pd ble:    att handle:      8 (GATT Service Changed Chr) [0m
 [0;32mI (82231106) pd ble:    cur indicate:    0 [0m
 [0;32mI (82231106) pd ble:    cur notify:      0 [0m
 [0;32mI (82231106) pd ble:    prev inficate:   1 [0m
 [0;32mI (82231106) pd ble:    prev notify:     0 [0m
 [0;32mI (82231106) pd ble:    reason:          Termination [0m
 [0;32mI (82231116) pd ble: GAP subscribe event: [0m
 [0;32mI (82231116) pd ble:    gap conn handle: 3 [0m
 [0;32mI (82231116) pd ble:    att handle:      33 (Jmx BtHttp Chr) [0m
 [0;32mI (82231116) pd ble:    cur indicate:    0 [0m
 [0;32mI (82231116) pd ble:    cur notify:      0 [0m
 [0;32mI (82231116) pd ble:    prev inficate:   1 [0m
 [0;32mI (82231116) pd ble:    prev notify:     0 [0m
 [0;32mI (82231116) pd ble:    reason:          Termination [0m
 [0;32mI (82231126) pd ble: GAP subscribe event: [0m
 [0;32mI (82231126) pd ble:    gap conn handle: 3 [0m
 [0;32mI (82231126) pd ble:    att handle:      37 (Jmx Midi Stone Stream Chr) [0m
 [0;32mI (82231126) pd ble:    cur indicate:    0 [0m
 [0;32mI (82231126) pd ble:    cur notify:      0 [0m
 [0;32mI (82231126) pd ble:    prev inficate:   1 [0m
 [0;32mI (82231126) pd ble:    prev notify:     0 [0m
 [0;32mI (82231136) pd ble:    reason:          Termination [0m
 [0;32mI (82231136) pd mss: remove mss client (gapCannHandle 3) [0m
 [0;32mI (82231136) pd sfile: relinquish refcount 1 filename:/JAMC/2024/d58db40f-0a3b-d074-96c1-8d560d23ff70/Jmx-A00038-Jul-03-2024.mid [0m
 [0;32mI (82231136) pd ble: GAP disconnect; reason: 0x0216 hci, CONN_TERM_LOCAL     gapConnHandle: 3 [0m
 [0;32mI (82231136) pd ble: Connection Description: [0m
 [0;32mI (82231146) pd ble:   role: Slave [0m
 [0;32mI (82231146) pd ble:   gap connection handle: 3 [0m
 [0;32mI (82231146) pd ble:   our id:  Public 00:35:38:64:62:34 [0m
 [0;32mI (82231146) pd ble:   our ota: Public 00:35:38:64:62:34 [0m
 [0;32mI (82231146) pd ble:   peer id:  Public f8:4d:89:7c:1c:7c [0m
 [0;32mI (82231146) pd ble:   peer ota: Public f8:4d:89:7c:1c:7c [0m
 [0;32mI (82231146) pd ble:   connection interval:     12 (15.000ms) [0m
 [0;32mI (82231156) pd ble:   connection latency:      0 times [0m
 [0;32mI (82231156) pd ble:   supervision timeout:     72 (720ms) [0m
 [0;32mI (82231156) pd ble:   Secure - encrypted:      0 [0m
 [0;32mI (82231156) pd ble:   Secure - authenticated:  0 [0m
 [0;32mI (82231156) pd ble:   Secure - bonded:         0 [0m
 [0;32mI (82231156) pd ble:   Secure - keySize:        0 [0m
 [0;32mI (82231156) pd alerts: notify bt [0m
 [0;32mI (82231166) pd bthttp: on client disconnected. gapConnHandle: 0 [0m
 [0;32mI (82231166) pd ble: ***** start advertising ***** [0m
 [0;32mI (82231166) NimBLE: GAP procedure initiated: advertise;  [0m
 [0;32mI (82231176) NimBLE: disc_mode=2 [0m
 [0;32mI (82231176) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0 [0m
 [0;32mI (82231176) NimBLE: 
```

chipweinberger, 2024-07-03:
I think the INSTANT_PASSED problem may have been caused by having my ble tasks the same priority.

```
#define PD_PRIORITY_BLE_LL_TASK 24
#define PD_PRIORITY_BLE_TASK 24
```

Ive now set it to this.

#define PD_PRIORITY_BLE_LL_TASK 24
#define PD_PRIORITY_BLE_TASK 23

Still testing.

chipweinberger, 2024-07-05:
happened again. the logs aren't that useful. It just happens randomly after many minutes of being idle.

```
  [0;32mI (178848473) pd mss: stone bounds 22 6128 [0m
 [0;32mI (178848473) NimBLE: GATT procedure initiated: indicate;  [0m
 [0;32mI (178848473) NimBLE: att_handle=37
 [0m
[In-Din]            178848.516 | Cb:0 Ch:0  | Note Off F5 vel:64
[Out-Din]
[Rec ]           80
 silence
 [0;32mI (178848533) pd mss: stone bounds 22 6132 [0m
 [0;32mI (178848543) NimBLE: GATT procedure initiated: indicate;  [0m
 [0;32mI (178848543) NimBLE: att_handle=37
 [0m
[In-Din]            178848.656 | Cb:0 Ch:0  | Note On D#5 vel:68
[Out-Din]
[Rec ]          140
  [0;32mI (178848693) pd mss: stone bounds 22 6137 [0m
 [0;32mI (178848693) NimBLE: GATT procedure initiated: indicate;  [0m
 [0;32mI (178848693) NimBLE: att_handle=37
 [0m
[In-Din]            178848.711 | Cb:0 Ch:0  | Note Off D#5 vel:64
[Out-Din]
[Rec ]           55
 silence
 [0;32mI (178848733) pd mss: stone bounds 22 6141 [0m
 [0;32mI (178848733) NimBLE: GATT procedure initiated: indicate;  [0m
 [0;32mI (178848733) NimBLE: att_handle=37
 [0m
[In-Din]            178848.846 | Cb:0 Ch:0  | Note On C5 vel:79
[Out-Din]
[Rec ]          135
  [0;32mI (178848883) pd mss: stone bounds 22 6146 [0m
 [0;32mI (178848883) NimBLE: GATT procedure initiated: indicate;  [0m
 [0;32mI (178848883) NimBLE: att_handle=37
 [0m
[In-Din]            178848.931 | Cb:0 Ch:0  | Note Off C5 vel:64
[Out-Din]
[Rec ]           85
 silence
 [0;32mI (178848953) pd mss: stone bounds 22 6150 [0m
 [0;32mI (178848953) NimBLE: GATT procedure initiated: indicate;  [0m
 [0;32mI (178848953) NimBLE: att_handle=37
 [0m
 [0;32mI (178849963) pd mss: all clients up to date. sleeping [0m
 [0;32mI (178851963) pd led ux: recording stoppped [0m
LED: solid r:0 g:0 b:255
[Rec ]       0     178848.931 | Cb:0 File Meta Event | (Meta) Manufacturer Specific, (261 bytes)jmxEof{
	"fileOffset":	6150,
	"totalMillis":	143145,
	"totalNotes":	360,
	"totalMsgs":	755,
	"stoneOffsetPrev":	22,
	"stoneIdxPrev":	1,
	"stoneUuidPrev":	"ff77bd70-6de2-d6fe-4d7d-656b6851a522",
	"unixtime":	1720221166,
	"localOffset":	-420,
	"tzNotSet":	true
}
[Rec ]       0     178848.931 | Cb:0 File Meta Event | (Meta) Manufacturer Specific, (25 bytes)jmxBlm{"b":295}
[Rec ]       0     178848.931 | Cb:0 File Meta Event | (Meta) End Of Track
 [0;32mI (178851993) pd midi recorder: flushed [0m
 [0;32mI (178852013) pd mss: stone bounds 22 6150 [0m
 [0;32mI (178852013) NimBLE: GATT procedure initiated: indicate;  [0m
 [0;32mI (178852023) NimBLE: att_handle=37
 [0m
 [0;32mI (178853023) pd mss: all clients up to date. sleeping [0m
 [0;32mI (178909003) pd lifestats: incremented LifeNotes (played) to 78974 [0m
 [0;32mI (178909003) pd lifestats: incremented LifeMillis (played) to 17182484 [0m
write jLifeStats, (offset:16384, commits:183, length:63) {"kLifeStats":{"kNotesPlayed":78974,"kMillisPlayed":17182484}}
 [0;32mI (180997983) pd ble: GAP subscribe event: [0m
 [0;32mI (180997993) pd ble:    gap conn handle: 2 [0m
 [0;32mI (180997993) pd ble:    att handle:      8 (GATT Service Changed Chr) [0m
 [0;32mI (180997993) pd ble:    cur indicate:    0 [0m
 [0;32mI (180997993) pd ble:    cur notify:      0 [0m
 [0;32mI (180997993) pd ble:    prev inficate:   1 [0m
 [0;32mI (180997993) pd ble:    prev notify:     0 [0m
 [0;32mI (180997993) pd ble:    reason:          Termination [0m
 [0;32mI (180998003) pd ble: GAP subscribe event: [0m
 [0;32mI (180998003) pd ble:    gap conn handle: 2 [0m
 [0;32mI (180998003) pd ble:    att handle:      33 (Jmx BtHttp Chr) [0m
 [0;32mI (180998003) pd ble:    cur indicate:    0 [0m
 [0;32mI (180998003) pd ble:    cur notify:      0 [0m
 [0;32mI (180998003) pd ble:    prev inficate:   1 [0m
 [0;32mI (180998003) pd ble:    prev notify:     0 [0m
 [0;32mI (180998003) pd ble:    reason:          Termination [0m
 [0;32mI (180998013) pd ble: GAP subscribe event: [0m
 [0;32mI (180998013) pd ble:    gap conn handle: 2 [0m
 [0;32mI (180998013) pd ble:    att handle:      37 (Jmx Midi Stone Stream Chr) [0m
 [0;32mI (180998013) pd ble:    cur indicate:    0 [0m
 [0;32mI (180998013) pd ble:    cur notify:      0 [0m
 [0;32mI (180998013) pd ble:    prev inficate:   1 [0m
 [0;32mI (180998013) pd ble:    prev notify:     0 [0m
 [0;32mI (180998023) pd ble:    reason:          Termination [0m
 [0;32mI (180998023) pd mss: remove mss client (gapCannHandle 2) [0m
 [0;32mI (180998023) pd sfile: relinquish refcount 1 filename:/JAMC/2024/d58db40f-0a3b-d074-96c1-8d560d23ff70/Jmx-A00040-Jul-05-2024.mid [0m
 [0;32mI (180998023) pd ble: GAP disconnect; reason: 0x0228 hci, INSTANT_PASSED      gapConnHandle: 2 [0m
 [0;32mI (180998023) pd ble: Connection Description: [0m
 [0;32mI (180998023) pd ble:   role: Slave [0m
 [0;32mI (180998033) pd ble:   gap connection handle: 2 [0m
 [0;32mI (180998033) pd ble:   our id:  Public 00:35:38:64:62:34 [0m
 [0;32mI (180998033) pd ble:   our ota: Public 00:35:38:64:62:34 [0m
 [0;32mI (180998033) pd ble:   peer id:  Public f8:4d:89:7c:1c:7c [0m
 [0;32mI (180998033) pd ble:   peer ota: Public f8:4d:89:7c:1c:7c [0m
 [0;32mI (180998033) pd ble:   connection interval:     12 (15.000ms) [0m
 [0;32mI (180998043) pd ble:   connection latency:      0 times [0m
 [0;32mI (180998043) pd ble:   supervision timeout:     72 (720ms) [0m
 [0;32mI (180998043) pd ble:   Secure - encrypted:      0 [0m
 [0;32mI (180998043) pd ble:   Secure - authenticated:  0 [0m
 [0;32mI (180998043) pd ble:   Secure - bonded:         0 [0m
 [0;32mI (180998043) pd ble:   Secure - keySize:        0 [0m
 [0;32mI (180998043) pd led ux: bluetooth client disconnected [0m
LED: solid r:0 g:255 b:0
 [0;32mI (180998053) pd alerts: notify bt [0m
 [0;32mI (180998053) pd bthttp: on client disconnected. gapConnHandle: 0 [0m
```