espressif/esp-idf #14054: bundle code  seem to be bugged

ulao, 2024-06-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

OK this is my first bug report, took me a long to to make this work because of all the bugs, I may have figure this out a lot soon but I wan new to it. 

If this is the wrong play to report I'm sorry, it was not a runtime bug I think, and not an install bug. Its just a programming bug. 

1) dedic_gpio_new_bundle and dedic_gpio_del_bundle do not share the same types for core_id. I tried to set both t int but the struct want uint32_t. I think uint32_t is pointless, there are only 2 CPUs.... Anyways, the core_id is references as uint32_t  in most placed but other section of code do not like it set to uint32_t .  When I did get most set to in the compare in the delete function compared as core_id: 1 to 1070156032 so it fails and says wrong cpu

2)

using
const int bundleA_gpios[] = {0,1,2,3};

allows pins 1,2 and 3 to oscillate . 

using
const int bundleA_gpios[] = { 1,2,3};

I can not use pin 3, its like the array has to start at 0

3)

using test code like this
#define DEBUG __asm__ __volatile__ ("ee.set_bit_gpio_out %0\n nop\n nop \n ee.clr_bit_gpio_out %0" : : "I"(DEBUG_BIT) : );
while(1){DEBUG};
with pin 3 I get 24 ns gaps
with pin 1 and 2 I get 9ms

[image]

ulao, 2024-09-13:
Was curios, if its in progress but no assignees is it still being worked?

Kainarx, 2024-09-14:
Hi @ulao 
Sorry for reply late. About you question:
1. Yes, we delete the bundle by the handle.
2. Please note the `mask` when writing or reading the dedic_gpio_bundle. Or you may miss something when configing the GPIO. It works well in my test case.
3. We allow to use dedic_gpio driver and gpio driver together. It also works well in my test case. Can you tell which chip and which version of IDF you are using? This will help me locate the problem faster.

ulao, 2024-09-14:
Oh wow, I took this status to mean it was confirmed and being worked on. Its been so long now checking every day in hopes to see the bug resoled that I do not even remember much about  any of this.  Looking at my skconfug I see this

Espressif IoT Development Framework (ESP-IDF) 5.4.0 Project Configuration

I know I was using the latest chip (s3) 

What I remember about the bug was there bundles worked, just not the cleanup and reuse of them. I will need to swtich back and forth in the running code sample.

ulao, 2024-09-18:
I will, try just need to set up the environment again and give it a go.

ulao, 2024-09-19:
Ok got my head wrapped around this again, first thing I did was verified the issue remains and is does. 

then I did a poll and now I have a mess.

`[0/1] Re-running CMake...-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
-- Project sdkconfig file C:/Users/Administrator/Desktop/usb2llapi/usb_host_lib/sdkconfig
-- Compiler supported targets: xtensa-esp-elf
CMake Error at C:/Users/Administrator/Desktop/esp/esp-idf/tools/cmake/tool_version_check.cmake:36 (message):
  

  Tool doesn't match supported version from list ['esp-14.2.0_20240906']:
  C:/Users/Administrator/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe   

  Please try to run 'idf.py fullclean' to solve it.

Call Stack (most recent call first):
  C:/Users/Administrator/Desktop/esp/esp-idf/components/esp_common/project_include.cmake:10 (check_expected_tool_version)    
  C:/Users/Administrator/Desktop/esp/esp-idf/tools/cmake/build.cmake:437 (include)
  C:/Users/Administrator/Desktop/esp/esp-idf/tools/cmake/build.cmake:669 (__build_process_project_includes)
  C:/Users/Administrator/Desktop/esp/esp-idf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/Administrator/Desktop/usb2llapi/usb_host_lib/build/CMakeFiles/CMakeOutput.log".
ninja: error: rebuilding 'build.ninja': subcommand failed

FAILED: build.ninja
C:\Users\Administrator\.espressif\tools\cmake\3.24.0\bin\cmake.exe --regenerate-during-build -SC:\Users\Administrator\Desktop\usb2llapi\usb_host_lib -BC:\Users\Administrator\Desktop\usb2llapi\usb_host_lib\build

 *  The terminal process "C:\Users\Administrator\.espressif\tools\ninja\1.11.1\ninja.exe" terminated with exit code: 1. `
can you help? Or is this the wrong place.

Kainarx, 2024-09-20:
It seems that your toolchain is not up to date and you are missing the `psutilf` package. You could try execute ./install.sh and then  ./export.sh in your IDF path

ulao, 2024-09-20:
oh I needed to go to the right path, I ran them both, CMD opened up and closed.

Kainarx, 2024-09-24:
About the environment you can refer the [programming guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/get-started/windows-setup.html)
And you can try to use `dedic_gpio_bundle_read_in` fucntion instead of reading gpio register

ulao, 2024-09-28:
If there a better place to get help with this sort of thing, Everything worked so well on the old branch but I can not make it work on the latest code.

Kainarx, 2024-09-30:
```
Tool doesn't match supported version from list ['esp-14.2.0_20240906']:
C:/Users/Administrator/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe
```  
From your log, it looks like you are using windows, it should be that your local toolchain does not match what is required in master. Please update your tools as  [https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/get-started/windows-setup-update.html](https://docs.espressif.com/projects/esp-idf/en/latest/esp32p4/get-started/windows-setup-update.html)

ulao, 2024-09-30:
I got a message saying python is missing it directed me here
https://dl.espressif.com/dl/esp-idf/

I do not know what version of python I need, and if that installer helps, I do not know what version of the installer to use, there are 6 of them.

ulao, 2024-10-04:
look like I needed to run install from the esp cmd, not the regular cmd. After running the install it says to run export.bat, then it says good to compile esp-idf. but compiling gives me

after install.bat I get

All done! You can now run:
   export.bat

after explore.bat I get

Done! You can now compile ESP-IDF projects.

 If I close down the IDE and re run it, it tries to update the container and kicks out errors. but I can edit my files locally still and build.

The following Python requirements are not satisfied:
Error while checking requirement 'psutil'. Package was not found and is required by the application: No package metadata was found for psutil
To install the missing packages, please run "install.bat"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: C:\Users\Administrator\.espressif\python_env\idf5.4_py3.11_env
    Python interpreter used: C:\Users\Administrator\.espressif\python_env\idf5.4_py3.11_env\Scripts\python.exe

ulao, 2024-11-02:
Does no one understand how any of this works? Everything was working great until I updated.

ulao, 2024-11-04:
After fishing for outside help the problem was very simple, just delete the .espressif folder and reconfigure the plugin..

but I have to report that this issue is not resolved.

Again, its simple to recreate. 

I made some test code to oscillate pins

while(1)
{
PIN_PULLUP(1 << 2);__asm__ __volatile__ ("nop\n nop\n nop\n"); PIN_PULLDOWN(1<<2);__asm__ __volatile__ ("nop\n nop\n nop\n");
   PIN_PULLUP(1 << 3);__asm__ __volatile__ ("nop\n nop\n nop\n"); PIN_PULLDOWN(1<<3);__asm__ __volatile__ ("nop\n nop\n nop\n");
   PIN_PULLUP(1 << 4);__asm__ __volatile__ ("nop\n nop\n nop\n"); PIN_PULLDOWN(1<<4);__asm__ __volatile__ ("nop\n nop\n nop\n");
 _delay_us(10);   
};

Simply, the pins go up and down ever 10 us and do so very fast. This will not work without setting up bundles. 

to recreate steps steps. 

1) Run my test code above. - resolve nothing shows up on the wire. the pin 1,2, and 3 remain low.
2) Recompile with the following lines.

const int bundleA_gpois[]={0,1,2,3};
bundles[0] = createBundle( bundleA_gpios,sizeof(bundleA_gpios) / sizeof(bundleA_gpios[0]), true );
 
3) run code. You will now see the lines oscillating ever 10 us. ( use a fast scope )
4) add the delete

const int bundleA_gpois[]={0,1,2,3};
bundles[0] = createBundle( bundleA_gpios,sizeof(bundleA_gpios) / sizeof(bundleA_gpios[0]), true ); deleteBundle(bundles[0]);

5) run the code and its still working. A delete should not be working, it should act like step 1

Kainarx, 2024-11-05:
I know what you mean. [This commit](https://github.com/espressif/esp-idf/commit/85a0845dad24d8e37c38953e36598d5a66a279ad) fixed the problem. You can rebase to the latest master to have a try

ulao, 2024-11-05:
I did a pull at master and now there seems to be an issue with the master?
----------------------
Make Error at C:/Users/Administrator/Desktop/esp/esp-idf/tools/cmake/component.cmake:317 (message):
  Include directory
  'C:/Users/Administrator/Desktop/esp/esp-idf/components/heap/tlsf/include'
  is not a directory.
Call Stack (most recent call first):
  C:/Users/Administrator/Desktop/esp/esp-idf/tools/cmake/component.cmake:493 (__component_add_include_dirs)
  C:/Users/Administrator/Desktop/esp/esp-idf/components/heap/CMakeLists.txt:54 (idf_component_register)

--------------

ulao, 2024-11-05:
yes it now is working correctly, thx for the help We finally got there.

closed 2024-11-06 by suda-morris