espressif/esp-idf #14055: OpenThread Network Time Synchronization Fails to Build on the ESP Thread Border Router

SimeonAT, 2024-06-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-dev-1030-g0479494e7a

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I should be able to use the OpenThread Network Time Synchronization feature on an ESP Thread Border Router, whether it be programmatically through the [API](https://openthread.io/reference/group/api-network-time) or the [command line interface](https://github.com/openthread/openthread/tree/main/src/cli#networktime).

### What is the actual behavior?

I am using the [ESP Thread Border Router SDK.](https://github.com/espressif/esp-thread-br), using the `OPENTHREAD_TIME_SYNC` flag per the instructions given in [Issue 12154](https://github.com/espressif/esp-idf/issues/12154#issuecomment-1776611060). I enabled that flag in **both** the RCP and Border Router programs. 

<details>
<summary>
The `idf.py build` process is successful for the RCP. However, the `idf.py build` process fails for the Border Router, giving the following error shown in this collapsible block.</summary>

```bash
/Users/simeon/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/openthread/libopenthread.a(mle_router.cpp.obj):(.literal._ZN2ot3Mle9MleRouter18SendAddressSolicitENS_15ThreadStatusTlv6StatusE+0x0): undefined reference to `otPlatTimeGetXtalAccuracy'
/Users/simeon/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/openthread/libopenthread.a(mle_router.cpp.obj): in function `_ZN2ot3Tlv6AppendINS_15ThreadStatusTlvEEE7otErrorRNS_7MessageENT_13UintValueTypeE':
/Users/simeon/esp/esp-idf/components/openthread/openthread/src/core/common/tlvs.hpp:584:(.text._ZN2ot3Mle9MleRouter18SendAddressSolicitENS_15ThreadStatusTlv6StatusE+0x66): undefined reference to `otPlatTimeGetXtalAccuracy'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```
</details>

I were to build this on an ESP32-H2 or ESP32-C6 using the `ot_cli` [example program](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_cli), the build will happen successfully, and the devices will be able to use network time synchronization without any issues.

I describe how I attempted to solve this issue in the *More Information* section. However, I had to modify the ESP-IDF source code in order to do so. This leads me to believe that it may be the case that the OpenThread Network Time Synchronization for the ESP Border Router does not have support yet. 

**As a result, I would like to ask the following question:**

Is it possible to use OpenThread Network Time Synchronization on an ESP Thread Border Router? If so, what steps do I need to take in order to properly navigate this build error (rather than using the ad-hoc solution I described in the *More Information* section?

I am not asking for a feature or bug fix to be implemented. However, my higher level question is if such a feature is currently possible in ESP-IDF. If it is not possible, I can change the OpenThread app layer programs that I am writing given that fact. However, I don't want to assume that it isn't possible to do so without asking if it possible first, as it may be the case that I am using the ESP-IDF OpenThread Network Time Sync feature incorrectly. 

### Steps to reproduce.

I want to note that I was running a [fork](https://github.com/UCSC-ThreadAscon) of the ESP Thread Border Router SDK and a [modified version](https://github.com/UCSC-ThreadAscon/network-performance-ftd) of the `ot_cli` program that I am using for my own purposes. However, I believe the problem can be easily reproduced on using the [ESP Thread Border Router SDK](https://docs.espressif.com/projects/esp-thread-br/en/latest/hardware_platforms.html) and the `ot_rcp` [example program](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_rcp).

The specific hardware I am using is the [ESP Thread Border Router/Zigbee Gateway](https://docs.espressif.com/projects/esp-thread-br/en/latest/hardware_platforms.html).

1. Clone the [ESP Thread Border Router SDK](https://docs.espressif.com/projects/esp-thread-br/en/latest/hardware_platforms.html).
2. Navigate to the `ot_rcp` [example program](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_rcp). Use `idf.py menuconfig`,  enable the `OPENTHREAD_TIME_SYNC` flag, and then run `idf.py build`. 
3. Navigate to the [Basic Thread Border Router example program](https://github.com/espressif/esp-thread-br/tree/main/examples/basic_thread_border_router) in the ESP Thread Border Router SDK. Use `idf.py menuconfig`,  enable the `OPENTHREAD_TIME_SYNC`, and run `idf.py build`. 
4. During the build process that was started in Step 3, the error will be displayed.

no2chem, 2024-06-23:
Hi, I am running into some similar issues as you. I think the issue is because the espressif RCP does not process the MAC header IE field properly if present, which results in a RCP RESET_FAULT. 

I started documenting some of the issues in the openthread repository (I am using ot-br-posix)
https://github.com/openthread/openthread/issues/10426

SimeonAT, 2024-06-24:
Hi @no2chem,

Thank you so much for taking the time to look over my issue! After reading your [Github issue](https://github.com/openthread/openthread/issues/10426) and corresponding [pull request](https://github.com/openthread/openthread/pull/10427) in the OpenThread source code, it makes much more sense why the RCP is crashing with my changes.

But more importantly, it makes a lot more sense why I can't get time synchronization to work on my ESP Thread Border: there isn't support for it yet for ESP32 devices, and it was [only recently](https://github.com/openthread/openthread/issues/10426) that RCP support for time sync was added to OpenThread (which is essential for border routers if they want to time sync).

Is my understanding of the issue correct? If not, what parts am I misunderstanding? Any guidance you can give me will be greatly appreciated.

no2chem, 2024-06-24:
It seems that the RCP build isn't even passing the TIME_SYNC config options correctly, so there's no chance for it to work in the first place...

I opened an issue and will submit a PR soon:
https://github.com/espressif/esp-idf/issues/14059

SimeonAT, 2024-06-24:
Thank you again @no2chem! Please keep me posted your discoveries and when your pull requests get approved.

This may be an obvious question, but for the sake of learning, I want to ask. To the best of my knowledge, the way that a Thread Border Router works (in general, not just ESP border routers), is that packets get received from the RCP, then get passed to through the Spinel to the code running the network layers and above in the OpenThread stack.

Thus, the problem with time sync is that the RCP fails to pass in the time sync packets to the network layer and above. I don't believe the problem is with the RCP itself when it first receives a time sync packet, [since it's able to parse it when it first sees it](https://github.com/espressif/esp-idf/blob/master/components/openthread/src/port/esp_openthread_radio.c#L734). The problem must arise after the RCP parses the packet, and it sends it to the OpenThread network layer (and the code response for MLE) for further processing.

Is my understanding correct, or is there some parts I got wrong or haven't fully understood yet?

no2chem, 2024-06-24:
honestly, I'm not really a thread expert myself. 

AFAICT, the line you highlighted isn't even in the RCP until you apply my PR (you can test yourself by adding a #error directive in the middle of the function). 

The time sync stuff in openthread isn't really documented. From reading the code, sync messages don't get sent until a leader shows up with time sync support. If the leader has time sync support, the mle sends a broadcast with the current time sequence. This is the problematic message, if you have a cli then running `state leader` with time sync enabled will cause the crash.

If the message contains a TimeIE, then for some reason the RCP will puke on it. Sync messages contain a Time IE that gets set in `mac.cpp`. If the TimeIEOffset is 0 (and it will be for the leader sync message), the ESP RCP will crash, even with the PR I sent. 

I haven't really debugged it more, but I will once I get a debug UART on the h2 working.

no2chem, 2024-06-24:
well, got logging working on a different UART than the RCP UART.

Looks like theres an assertion that is failing now:

```
assert failed: 0x40809702
0x40809702: isr_handle_rx_abort at /esp-idf/components/ieee802154/driver/esp_ieee802154_dev.c:523
 (inlined by) ieee802154_isr at /esp-idf/components/ieee802154/driver/esp_ieee802154_dev.c:657
```

I'll look into it further tomorrow...

no2chem, 2024-06-24:
I've debugged more into this and the problem is an assert, but not the line above (for some reason the backtrace seems to be inaccurate). I'm trying to determine whether we actually ended up in the wrong state or not.

https://github.com/espressif/esp-idf/issues/14068

no2chem, 2024-06-27:
Hi @SimeonAT - it looks like I resolved the issue, it appears the 802154 driver has some issues on state transition changes when transmit_at and rx_on_idle were enabled at the same time.

If you apply #14060 and #14089 and turn the time sync option on, you should get a RCP that works with time sync. *however* I think you won't be able to get the ESP thread border router working with time sync because the ESP border router is provided as a closed source library so you can't change the compile options on it. I just have the ESP32-S3 run serial over ethernet (or wi-fi) and use `socat` to pipe to a ot-br-posix instance to give you full control over the BR (it looks like you're running some experiments?, so perhaps that would be the best to give you some room to play with the BR code).

SimeonAT, 2024-06-28:
Thank you so much for the help @no2chem! I really appreciate the help, and the detailed and thorough explanation you given me. I don't necessarily need to run it over the border router. If I can get it in to run in the border router - then great. However, given such problems, I won't run my experiments in the border router - I'll stick with running them on non-border router FTDs. What's important for me in this case is if I don't run it in the border router, I'll need a justification as to why, and the discussion we have in this Github issue will allow me to describe why I'm using the border router in my experiments when my work gets evaluated by my thesis committee.

closed 2024-06-28 by SimeonAT