espressif/esp-idf #14056: ESP32 host(gateway) report OPENTHREAD:[W] P-RadioSpinel-: Handle transmit done failed: ChannelAccessFailure(esp32h2 is rcp device)

wensenl, 2024-06-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v5.1.4

### Espressif SoC revision.

ESP32/ESP32H2

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Self developed motherboard（base ）

### Power Supply used.

External 5V

### What is the expected behavior?

Hope there is no error（OPENTHREAD:[W] P-RadioSpinel-: Handle transmit done failed: ChannelAccessFailure） reported.

### What is the actual behavior?

[image]

### Steps to reproduce.

1. gateway host is esp32，rcp is esp32h2，gateway is zigbee ESP_ZB_DEVICE_TYPE_COORDINATOR. rcp is esp-idf-v5.1.4/examples/openthread/ot_rcp
2. Zigbee device is esp32h2. Zigbee device is ESP_ZB_DEVICE_TYPE_ROUTER. Networking of 4 Zigbee devices.
3. When they are working, the gateway randomly prints the following error. error is:
W(822185) OPENTHREAD:[W] P-RadioSpinel-: Handle transmit done failed: ChannelAccessFailure
[comm_mqtt.c:134]^heartbeat response
W(832285) OPENTHREAD:[W] P-RadioSpinel-: Handle transmit done failed: ChannelAccessFailure
4. The further away the gateway and device are from each other, the more errors will be printed

### Debug Logs.

```plain
[comm_mqtt.c:154]^puback, packet id: 168

[comm_mqtt.c:134]^heartbeat response

W(1483125) OPENTHREAD:[W] P-RadioSpinel-: Handle transmit done failed: ChannelAccessFailure
W(1488485) OPENTHREAD:[W] P-RadioSpinel-: Handle transmit done failed: ChannelAccessFailure
connect error, errno: 113
success to establish tcp, fd=0
W (1497925) ESP_ZB_GATEWAY: Network(0x02f3) closed, devices joining not allowed.
W (1497945) ESP_ZB_GATEWAY: Network(0x02f3) is open for 180 seconds
W(1497975) OPENTHREAD:[W] P-RadioSpinel-: Handle transmit done failed: ChannelAccessFailure
W(1498485) OPENTHREAD:[W] P-RadioSpinel-: Handle transmit done failed: ChannelAccessFailure
W (1498505) ESP_ZB_GATEWAY: Network(0x02f3) is open for 180 seconds
[comm_mqtt.c:134]^heartbeat response

_core_sysdep_network_tcp_send, nwk select timeout
[comm_mqtt.c:89]^[1500.550][LK-0315] MQTT connect packet send timeout: 5000
```

gytxxsy, 2024-07-17:
A fix is missing in v5.1.4 and you can switch to release/v5.1 to resolve this issue. If you do not want to switch to the branch, you can manually add a line of code to fix the problem:
https://github.com/espressif/esp-idf/blob/release/v5.1/components/openthread/src/spinel/esp_radio_spinel.cpp#L272

[image]

chshu, 2024-08-02:
@wensenl Hope https://github.com/espressif/esp-idf/issues/14056#issuecomment-2232669482 resolved your issue, feel free to reopen if any follow up questions.

closed 2024-08-02 by chshu