espressif/esp-idf #14057: fatal error: esp_nimble_hci.h: No such file or directory

zzzh, 2024-06-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

no error

### What is the actual behavior?

fatal error: esp_nimble_hci.h: No such file or directory

### Steps to reproduce.

1. #include "esp_nimble_hci.h" in a c file.
2. idf.py set-target esp32c2
3. idf.py menuconfig, enable vluetooth and select NimBLE - BLE only
4. idf.py build

### Build or installation Logs.

```plain
../main/ble1.c:22:10: fatal error: esp_nimble_hci.h: No such file or directory
   22 | #include "esp_nimble_hci.h"
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
```

rahult-github, 2024-06-23:
Hi @zzzh ,

 > #include "esp_nimble_hci.h" in a c file. 
 
 This step is incorrect. May i know why you are explicitly including this file ?  The VHCI interface is not used in esp32c2 and hence this file won't be found during compilation for esp32c2.

rahult-github, 2024-07-24:
Closing this issue. Please feel free to reopen if issue still exists.

closed 2024-07-24 by rahult-github