espressif/esp-idf #14061: ESP-IDF v5.3-beta2-dirty 为什么不可以配置S3芯片的PSRAM

shenxiaozheng, 2024-06-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.3-beta2-dirty

### Espressif SoC revision.

ESP32S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-Korvo v4.0

### Power Supply used.

USB

### What is the expected behavior?

我在写代码的过程中，硬件完好的情况下，PSRAM无法通过sdkconfig进行配置。

### What is the actual behavior?

我在写代码的过程中，硬件完好的情况下，PSRAM无法通过sdkconfig进行配置。

### Steps to reproduce.

1. 写一个c程序
2. 程序中在psram上申请1k的内存
3. 编译通过之后烧板子
4. 直接报错
...

nopnop2002, 2024-07-01:
I was able to successfully mount it using ESP-IDF v5.3 and ESP32S3.

```
$ idf.py --version
ESP-IDF v5.3-beta2
```

```
I (177) esp_psram: Found 8MB PSRAM device
```

ginkgm, 2024-07-17:
Hi @shenxiaozheng ,

The Korvo has a Octal PSRAM. Could you select this config option and try again:         config SPIRAM_MODE_OCT?

If you still fail, please provide more detailed information, including sdkconfig, IDF sha, your testing code, failing log.

Thanks,
Michael

shenxiaozheng, 2024-07-17:
@nopnop2002 
Well, thank you very much. I have followed your instructions to resolve this issue.

closed 2024-07-17 by Alvin1Zhang