espressif/esp-idf #14065: strlcat 函数为什么会乱码？

ktech0, 2024-06-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2

### Espressif SoC revision.

ESP32s3-20210327

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

使用 strlcat 函数，在电脑上正常拼接字符串
[image]

### What is the actual behavior?

使用 strlcat 函数，在 esp32-s3 板子上拼接字符串乱码，长度也异常
[image]

### Steps to reproduce.

刷写以下代码
```c
char *a[3] = {"ABC", "XYZ", "DEF"};
char *b = (char *)malloc(10 * sizeof(char));
for (int i = 0; i < 3; i++) {
    strlcat(b, a[i], 10);
    printf("b length: %d\n", strlen(b));
}
printf("%s\n", b);
b[9] = '\0';
printf("%s\n", b);
```

### Debug Logs.

```plain
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x403759b0
0x403759b0: esp_restart_noos at /opt/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:159

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x178c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xcbc
load:0x403cc700,len:0x2d9c
entry 0x403c9914
I (26) boot: ESP-IDF v5.2.2 2nd stage bootloader
I (26) boot: compile time Jun 24 2024 20:25:16
I (26) boot: Multicore bootloader
I (29) boot: chip revision: v0.1
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 2MB
I (47) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00100000
I (86) boot: End of partition table
I (90) esp_image: segment 0: paddr=00010020 vaddr=3c0b0020 size=25cech (154860) map
I (126) esp_image: segment 1: paddr=00035d14 vaddr=3fc98d00 size=049a8h ( 18856) load
I (131) esp_image: segment 2: paddr=0003a6c4 vaddr=40374000 size=05954h ( 22868) load
I (137) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=a442ch (672812) map
I (261) esp_image: segment 4: paddr=000e4454 vaddr=40379954 size=0f334h ( 62260) load
I (284) boot: Loaded app from partition at offset 0x10000
I (284) boot: Disabling RNG early entropy source...
I (295) cpu_start: Multicore app
I (305) cpu_start: Pro cpu start user code
I (305) cpu_start: cpu freq: 160000000 Hz
I (305) cpu_start: Application information:
I (308) cpu_start: Project name:     sd_card
I (313) cpu_start: App version:      1
I (317) cpu_start: Compile time:     Jun 24 2024 20:25:10
I (324) cpu_start: ELF file SHA256:  1cb05209f...
I (329) cpu_start: ESP-IDF:          v5.2.2
I (334) cpu_start: Min chip rev:     v0.0
I (338) cpu_start: Max chip rev:     v0.99 
I (343) cpu_start: Chip rev:         v0.1
I (348) heap_init: Initializing. RAM available for dynamic allocation:
I (355) heap_init: At 3FCA1990 len 00047D80 (287 KiB): RAM
I (361) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (368) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (374) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (381) spi_flash: detected chip: gd
I (384) spi_flash: flash io: dio
W (388) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (402) sleep: Configure to isolate all GPIO pins in sleep state
I (408) sleep: Enable automatic switching of GPIO sleep configuration
I (416) main_task: Started on CPU0
I (426) main_task: Calling app_main()
b length: 15
b length: 15
b length: 15
��?��?li�?�-
��?��?l
I (426) SDMMC: mounting sdcard filesystem...
I (436) gpio: GPIO[3]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (446) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (456) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (506) SDMMC: sdcard mounted success.
Name: 00000
Type: SDHC/SDXC
Speed: 40.00 MHz (limit: 40.00 MHz)
Size: 14992MB
CSD: ver=2, sector_size=512, capacity=30703616 read_bl_len=9
SSR: bus_width=1
```

suda-morris, 2024-06-25:
Please replace `malloc` with `calloc`

ktech0, 2024-06-25:
> @ktech0 The destination buffer, `b` is **uninitialized** when you call `strlcat`, so you are concatenating a string that contains garbage with strings from `a`.
> 
> Either you follow what @suda-morris suggested, either you initialize your destination buffer with `b[0] = 0` (or `memset`):
> 
> ```
> char *b = (char *)malloc(10 * sizeof(char));
> b[0] = 0;
> ...
> ```

So that explains it. Thanks very much.

ktech0, 2024-06-25:
这就导致所有用到 strlcat 的地方，都必须用 calloc 初始化，比如 `httpd_req_get_url_query_str()`

closed 2024-06-25 by ktech0