espressif/esp-idf #14067: [SECURE_BOOT_V2] How to OTA device without WiFi via command line

ravinder-bb, 2024-06-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

We have flashed 1000 devices with secured bootloader and application binaries. However, the devices that were connected to Wi-Fi experienced bugs in the code. Can we perform an over-the-air (OTA) update of all devices using the command line?

**sdkconfig**
```
#
# Security features
#
CONFIG_SECURE_SIGNED_ON_BOOT=y
CONFIG_SECURE_SIGNED_ON_UPDATE=y
CONFIG_SECURE_SIGNED_APPS=y
CONFIG_SECURE_BOOT_V2_RSA_SUPPORTED=y
CONFIG_SECURE_BOOT_V2_PREFERRED=y
CONFIG_SECURE_BOOT_V2_RSA_ENABLED=y
CONFIG_SECURE_SIGNED_APPS_RSA_SCHEME=y
CONFIG_SECURE_BOOT=y
CONFIG_SECURE_BOOT_V2_ENABLED=y
CONFIG_SECURE_BOOT_BUILD_SIGNED_BINARIES=y
CONFIG_SECURE_BOOT_SIGNING_KEY="secure_boot_signing_key.pem"
# CONFIG_SECURE_BOOT_ENABLE_AGGRESSIVE_KEY_REVOKE is not set
# CONFIG_SECURE_BOOT_FLASH_BOOTLOADER_DEFAULT is not set
# CONFIG_SECURE_BOOT_INSECURE is not set
CONFIG_SECURE_FLASH_ENC_ENABLED=y
CONFIG_SECURE_FLASH_ENCRYPTION_AES128=y
# CONFIG_SECURE_FLASH_ENCRYPTION_AES256 is not set
# CONFIG_SECURE_FLASH_ENCRYPTION_MODE_DEVELOPMENT is not set
CONFIG_SECURE_FLASH_ENCRYPTION_MODE_RELEASE=y
CONFIG_SECURE_FLASH_HAS_WRITE_PROTECTION_CACHE=y
# CONFIG_SECURE_FLASH_ENCRYPT_ONLY_IMAGE_LEN_IN_APP_PART is not set
CONFIG_SECURE_FLASH_CHECK_ENC_EN_IN_APP=y
CONFIG_SECURE_ROM_DL_MODE_ENABLED=y
# CONFIG_SECURE_DISABLE_ROM_DL_MODE is not set
CONFIG_SECURE_ENABLE_SECURE_ROM_DL_MODE=y
# CONFIG_SECURE_INSECURE_ALLOW_DL_MODE is not set
# end of Security features
```

mahavirj, 2024-07-01:
@ravinder-bb 

- How was the flash encryption key generated for these devices? Was it externally programmed or it was internally generated while enabling the flash encryption for first time?
- `CONFIG_SECURE_ENABLE_SECURE_ROM_DL_MODE` this means that secure ROM DL mode has been enabled on these devices and hence loading the stub code is not possible anymore (please see [docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/security/security.html#uart-download-mode) here). If the flash encryption key was internally generated then encrypting the artifacts is not possible and hence serial flashing without stub also won't work.
- Can the currently executing firmware on the device reach to OTA update server?

ravinder-bb, 2024-07-06:
@mahavirj,

**esptool.py --chip esp32s2 --port /dev/ttyUSB1 get_security_info**

>esptool.py v3.0
Serial port /dev/ttyUSB1
Connecting....
Chip is ESP32-S2 in Secure Download Mode
WARNING: Stub loader is not supported in Secure Download Mode, setting --no-stub
Enabling default SPI flash mode...
Flags: 0x000006f5 (0b11011110101)
Flash_Crypt_Cnt: 0x1
Key_Purposes: (9, 4, 0, 0, 0, 0, 0)
Hard resetting via RTS pin...

**espefuse.py --chip esp32s2 --port /dev/ttyUSB1 -p /dev/ttyUSB1 summary**

>Connecting.....
Traceback (most recent call last):
  File "/home/shivay/esp/esp-idf/components/esptool_py/esptool/espefuse.py", line 131, in <module>
    _main()
  File "/home/shivay/esp/esp-idf/components/esptool_py/esptool/espefuse.py", line 124, in _main
    main()
  File "/home/shivay/esp/esp-idf/components/esptool_py/esptool/espefuse.py", line 104, in main
    efuses, efuse_operations = get_efuses(esp, just_print_help, debug_mode, args1.do_not_confirm)
  File "/home/shivay/esp/esp-idf/components/esptool_py/esptool/espefuse.py", line 69, in get_efuses
    return (efuse.EspEfuses(esp, skip_connect, debug_mode, do_not_confirm), efuse.operations)
  File "/home/shivay/esp/esp-idf/components/esptool_py/esptool/espressif/efuse/esp32s2/fields.py", line 81, in __init__
    self.blocks = [EfuseBlock(self, self.Blocks.get(block), skip_read=skip_connect) for block in self.Blocks.BLOCKS]
  File "/home/shivay/esp/esp-idf/components/esptool_py/esptool/espressif/efuse/esp32s2/fields.py", line 81, in <listcomp>
    self.blocks = [EfuseBlock(self, self.Blocks.get(block), skip_read=skip_connect) for block in self.Blocks.BLOCKS]
  File "/home/shivay/esp/esp-idf/components/esptool_py/esptool/espressif/efuse/esp32s2/fields.py", line 39, in __init__
    super(EfuseBlock, self).__init__(parent, param, skip_read=skip_read)
  File "/home/shivay/esp/esp-idf/components/esptool_py/esptool/espressif/efuse/base_fields.py", line 133, in __init__
    self.read()
  File "/home/shivay/esp/esp-idf/components/esptool_py/esptool/espressif/efuse/base_fields.py", line 179, in read
    words = self.get_words()
  File "/home/shivay/esp/esp-idf/components/esptool_py/esptool/espressif/efuse/base_fields.py", line 176, in get_words
    return [self.parent.read_reg(offs) for offs in get_offsets(self)]
  File "/home/shivay/esp/esp-idf/components/esptool_py/esptool/espressif/efuse/base_fields.py", line 176, in <listcomp>
    return [self.parent.read_reg(offs) for offs in get_offsets(self)]
  File "/home/shivay/esp/esp-idf/components/esptool_py/esptool/espressif/efuse/base_fields.py", line 337, in read_reg
    return self._esp.read_reg(addr)
  File "/home/shivay/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 562, in read_reg
    val, data = self.command(self.ESP_READ_REG, struct.pack('<I', addr))
  File "/home/shivay/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 398, in command
    raise UnsupportedCommandError(self, op)
esptool.UnsupportedCommandError: This command (0xa) is not supported in Secure Download Mode

ravinder-bb, 2024-07-08:
We have tried but got this error.

```
esptool.py v4.7.0
Serial port /dev/ttyUSB1
Connecting......
Chip is ESP32-S2 in Secure Download Mode
Changing baud rate to 460800
Changed.
Enabling default SPI flash mode...
Configuring flash size...
Flash will be erased from 0x00040000 to 0x00190fff...
Erasing flash...

A fatal error occurred: Failed to enter Flash download mode (result was 01640000: Unknown result)
```

ravinder-bb, 2024-07-09:
@mahavirj, Any other way is possiable just like **otatool.py** to update only OTA files?

ravinder-bb, 2024-07-09:
@mahavirj, Any suggestion ?

mahavirj, 2024-07-12:
@ravinder-bb Sorry, I don't have any other suggestions here. I gave you all possible information. If you don't have access to flash encryption key for these devices then unfortunately updating the firmware is not possible over serial in their current state.

closed 2024-07-15 by mahavirj