espressif/esp-idf #14069: No master clock (MCLK) GPIO for hfp_hf example

abel-belay, 2024-06-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm trying to run the hfp_hf example on a ESP32-DevKitC and a PMOD I2S2 codec. The codec requires a master clock (MCLK) input but the example does not assign the master clock to any pin and does not provide a way of doing so.

esp-qing, 2024-06-28:
Hi, Sorry for replying so late. Which audio data path are you using (PCM or HCI)?

jtwaleson, 2024-10-23:
Hi @esp-qing . I am now running into this same issue, after giving up on my software approach for the microphone, my hardware module requires a MCLK to work. This is for the PCM data path.

jtwaleson, 2024-10-23:
For a bit more context, the master clock (MCLK, PCM_MCLK) is a multiple of the bit clock (BCLK). In HFP_HF we can only specify:
- DOUT
- DIN
- CLK (BLCK)
- FSYNC (WS)

The Master Clock is missing. For the frequency that HFP_HF is using by default (8khz FSYNC, 256khz CLK) a 2Mhz MCLK can be used. By supplying an external 2Mhz clock signal to my device (PCM5102) I could get it to work. Regular I2S examples (https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/i2s.html) allow the MCLK to be mapped to a GPIO. Would be very nice if this could be made to work with HFP_HF too.

jtwaleson, 2024-10-23:
I just realized that as a very hacky workaround, the second i2s slot can be used to generate just the clock signal. In my very limited experience the clocks don't have to be perfectly. I will give this a go.

esp-qing, 2024-10-30:
Hi, @jtwaleson sorry, I just saw your comment.
The IIS configuration of esp-idf is not effective for the PCM of HFP. The PCM of HFP is equivalent to an independent IIS channel implemented separately. And there is no MCLK.
But I have tried using PCM5100 with hfp_hf via PCM, it works well.  Maybe there is not much difference between PCM5100 and PCM5102.  MCLK is not used. Only two clock (LCK and BCK) connecte to esp32. Note that XSMT and FMT need to be set to high.

jtwaleson, 2024-10-30:
Hi @esp-qing , sorry I made a mistake in my message, indeed PCM5102 worked fine, but my mic input ADC (based on WM8782) needed the MCLK. I think I needed a bit more sleep when I was typing ;)

FYI the completed project is here: https://blog.waleson.com/2024/10/bakelite-to-future-1950s-rotary-phone.html , thanks a lot for your help!

esp-qing, 2024-10-30:
@jtwaleson Wow, this is such an interesting and meaningful project! I also learned a lot from the production process in your blog. Thank you for sharing!

closed 2024-12-23 by Alvin1Zhang