espressif/esp-idf #14070: 运行例程softAP，手机荣耀50SE无法连接

yangjinyiyiyi, 2024-06-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

在ESP32S3和ESP32C3模组上分别运行例程softAP，IDF版本4.4.7和5.1.4均测试，手机荣耀50SE始终无法连接，同一时间换周边其他同事手机（苹果，小米，oppo等）都可以连接，荣耀50SE连接路由器都正常，荣耀50SE连接AP热点时log如下，麻烦帮分析定位下问题，谢谢
[image]

Xiehanxin, 2024-06-26:
你好，这个问题已经在该commit上修复:04ea82474047bd51a72bea4fd9811a69b58ffa02，请使用之后的commit 或者更新到v5.2 tag

leeebo, 2024-06-26:
fixed in 04ea82474047bd51a72bea4fd9811a69b58ffa02

Indastri, 2024-09-10:
Since this was fixed and there is no response from the user. Closing. Feel free to reopen or create new issue if the problem persists.

Cheers!