espressif/esp-idf #14071: new i2c driver can not run without an i2c address

achao1104, 2024-06-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

idf5.2.2

### Espressif SoC revision.

esp32s2

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

esp32s3

### Power Supply used.

USB

### What is the expected behavior?

There are some chipsets such as ch423s without an i2c address, however, you can use old i2c driver to control the reading and writing to those kinds of chipsets. But in new version of i2c driver, you have to config an address in the bus and config, thus, it failed to communicate with those chipsets.

The followings are codes in old driver, which runs ok with ch423s.

    err = i2c_master_start(handle);

    if (err != ESP_OK) {

        goto end;

    }
    err = i2c_master_write_byte(handle, ( unsigned char )( cmd>>8 ) , true);

    if (err != ESP_OK) {

        goto end;

    }

    err = i2c_master_write_byte(handle, ( unsigned char ) cmd , true);

    if (err != ESP_OK) {

        goto end;

    }

### What is the actual behavior?

failed wit communicate with ch423s

### Steps to reproduce.

as above

mythbuster5, 2024-06-25:
TBH, the new driver is designed strictly follow the standard i2c protocal - which means there must be an address. For product you provide ch423s. I went through its datasheet, it doesn't call its I2C at all. It only says this is `two wire serial interface` also it uses `scl/sda` name

Though it's possible to support this in new driver but it will not be treated as a high priority task fyi.

mythbuster5, 2024-11-06:
Won't do this support now.

closed 2024-11-06 by mythbuster5