espressif/esp-idf #14073: Wifi AP on ESP32S3 working in APSTA Mode when connected to a AP, but not when not connected

diplfranzhoepfinger, 2024-06-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

f

diplfranzhoepfinger, 2024-06-25:
can it be some thin in FLASH ? so that the Wifi API stores something into flash which makes this Connection run bad ?

diplfranzhoepfinger, 2024-06-25:
does this tell something: 

appears on the Server when not connected to the Internet AP 

```
I (52391) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (52391) wifi:station: 50:02:91:8d:db:ec join, AID=1, bgn, 40U
I (56456) wifi:station: 50:02:91:8d:db:ec leave, AID = 1, bss_flags is 33786978, bss:0x3c23346c
I (56456) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (116364) wifi:removing station <50:02:91:8d:db:ec> after unsuccessful auth/assoc, AID = 0
W (116364) wifi:rm mis
I (116365) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
```

hansw123, 2024-06-26:
hi @diplfranzhoepfinger 
I'm sorry I didn't get your point, could you be more specific about the connection between sta, apsta , your phone and router and describe the steps to reproduce the problem, thank you very much for the report!

diplfranzhoepfinger, 2024-06-26:
hi, 

seems to be a COEX Problem: 

[https://docs.espressif.com/projects/esp-idf/en/v5.2.2/esp32s3/api-guides/coexist.html](https://docs.espressif.com/projects/esp-idf/en/v5.2.2/esp32s3/api-guides/coexist.html)

when BLE Scanner is on, the AP seems not working on ESP32, either Standalone or APSTA: 

[https://github.com/espressif/esp-idf/tree/master/examples/wifi/getting_started/softAP](https://github.com/espressif/esp-idf/tree/master/examples/wifi/getting_started/softAP)

[https://github.com/espressif/esp-idf/tree/master/examples/wifi/softap_sta](https://github.com/espressif/esp-idf/tree/master/examples/wifi/softap_sta)

when BLE off, all good. 

when BLE on, only one working is 

[https://github.com/espressif/esp-idf/tree/master/examples/wifi/softap_sta](https://github.com/espressif/esp-idf/tree/master/examples/wifi/softap_sta)

BUT only when the STA is connected first, then the AP works. 

if STA not connected, AP fail.

hansw123, 2024-06-27:
@diplfranzhoepfinger 
Yes, there are indeed coexisting influences, and in the end, thank you for the report!

Indastri, 2024-07-01:
Closing as completed. If you will have any further problems feel free to reopen this or create a new issue.

Cheers!

closed 2024-07-01 by Indastri