espressif/esp-idf #14077: run itwt sample and services as udp server can't receive packets from udp client

wangzhi15632, 2024-06-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v5.2.1

### Espressif SoC revision.

esp32-c6-devkitc-1 board

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6-DevKitC-1 

### Power Supply used.

USB

### What is the expected behavior?

i add udp server sample code to the itwt sample and setup the service period about 10 seconds. then, i write a udp client demo on linux server ubuntu. ubuntu connect route by ethernet. when i run the udp client demo on ubuntun. most of the time, the device can't recevie the packet from the udp client.
i think that when the device wakeup, it will recive all packets cached by ap.

### What is the actual behavior?

the device rarely receive packets

### Steps to reproduce.

1. use itwt sample and add udp server code.
2. connect to route
3. run udp client on ubuntu
...

Yinqingzhao-espressif, 2024-07-03:
@wangzhi15632 Could you provide a packet capture file?

QingzhaoYin, 2025-07-01:
@wangzhi15632  Hi～， typically，the AP's buffering capability is limited. If you send multiple UDP packets per second or in a short period while the STA is asleep, the AP's buffer may overflow, causing it to drop some packets.