espressif/esp-idf #14079: Http OTA failed using "simple_ota_example"

mahoneinsh, 2024-06-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V5.2.1

### Espressif SoC revision.

ESP32-C6

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

My board

### Power Supply used.

USB

### What is the expected behavior?

I want to download app bin file from http server (pc) into esp32-c6 board through wifi.  

### What is the actual behavior?

I have modified the SDK configuration, created the http server on my pc, and tried to download but failed at the last step, as shown below.
[image]

### Steps to reproduce.

1. ESP32 project configuration
[image]
[image]
2.HTTP server
[image]
[image]
3.Example project build, flash and monitor 
[image]

mahoneinsh, 2024-06-26:
Follow ESP32 FAE's advice, i also mask the following code
[image]

nileshkale123, 2024-07-01:
Hi @mahoneinsh ,

Thank you for bringing this issue to my attention.

I've just tested the program on my end, and it executed successfully. To help troubleshoot your issue, could you please provide some more details? Specifically:

    1. The steps you followed in detail to execute the program
    2. Any specific environment or configuration details (like sdkconfig ) that might be relevant

With this information, I'll be better able to assist you in resolving the problem.

Thank you for your patience and cooperation.

nileshkale123, 2024-07-01:
I reviewed your logs and noticed that the HTTP server and the ESP device are on different networks. The server is on 192.168.159.x, while the ESP device is on 192.168.137.x.

Please ensure both devices are on the same network to resolve the connection issue. Try connecting both devices ESP device and PC where the server is running to the same router.

closed 2024-07-12 by nileshkale123