espressif/esp-idf #14080: correct usage of esp_wifi_get_config

diplfranzhoepfinger, 2024-06-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, 

how to use `esp_wifi_get_config` ? 

after executing the command, how to know if there was a stored config in flash ? 

is this the right way ? 

```
    //Check if Configs are available: 
    
    esp_wifi_get_config(WIFI_IF_AP, &ap_cfg);
    esp_wifi_get_config(WIFI_IF_STA, &sta_cfg);
    
    
    
    
        
	ESP_LOGI(TAG_AP, "wifi_init_softap stored: SSID:%s password:%s channel:%d",
             ap_cfg.ap.ssid, ap_cfg.ap.password, ap_cfg.ap.channel);          
             
    ESP_LOGI(TAG_STA, "wifi_init_station stored. SSID:%s password:%s channel:%d",
             sta_cfg.sta.ssid, sta_cfg.sta.password, sta_cfg.sta.channel);             
    
    
    
```

diplfranzhoepfinger, 2024-06-26:
also discussed here: 

[https://esp32.com/viewtopic.php?t=29199](https://esp32.com/viewtopic.php?t=29199)

ravinder-bb, 2024-06-26:
Default storage in **WIFI_STORAGE_FLASH**

You can manually change storage [Documentation](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_wifi.html#_CPPv420esp_wifi_set_storage14wifi_storage_t)
```
esp_wifi_set_storage(WIFI_STORAGE_FLASH);
``

imdahsiaria90, 2024-07-01:
@diplfranzhoepfinger @ravinder-bb 

I am trying to do the same but for me the contents of wifiStaConfig.sta_cfg.sta.ssid and wifiStaConfig.sta_cfg.sta.password are 0 /empty
I am using IDF v5.0.1

Example code
```
     wifi_config_t wifiStaConfig = {0};

	esp_err_t ret = esp_wifi_get_config(ESP_IF_WIFI_AP, &wifiStaConfig);

	printf("WIFI STA SSID DETECTED: %s\n", wifiStaConfig.sta.ssid);
	printf("WIFI STA PASSWORD DETECTED: %s\n", `wifiStaConfig.sta.password);

```

Any idea what would be wrong ? I have successfully stored it. The reading does not work

mhdong, 2024-07-01:
Hi @imdahsiaria90 
esp_wifi_get_config(WIFI_IF_STA, &wifiStaConfig);

Sherry616, 2024-08-21:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates. 
Thanks for using our Espressif product!

closed 2024-08-21 by Sherry616

diplfranzhoepfinger, 2024-08-21:
@Sherry616 @ravinder-bb 

Still the Question un-answered: 

> after executing the command, how to know if there was a stored config in flash ?
> 

is there any Flag for "Stored" or "non-Stored" ?

mhdong, 2024-08-21:
Hi @diplfranzhoepfinger 
There isn't a specific flag for "Stored" or "non-Stored" in ESP-IDF. When you retrieve the configuration use esp_wifi_get_config, you either get the stored value or a default value if nothing has been stored yet.

nopnop2002, 2025-03-15:
I think esp_wifi_get_config(WIFI_IF_STA, &sta_cfg) returns this value.

```
esp32> nvs_list nvs
namespace 'nvs.net80211', key 'sta.ssid', type 'blob'
namespace 'nvs.net80211', key 'sta.pswd', type 'blob'
```