espressif/esp-idf #14081: ESP32S3 Stuck in Download Boot Mode

Areeb-Mufti, 2024-06-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Power Supply used.

External 3.3V

### What is the expected behavior?

The device to boot in SPI Boot Mode

### What is the actual behavior?

Device boots in Joint Download Boot Mode
[image]

### Steps to reproduce.

1. Device operating normally
2. Remove power to ESP (3.3V) to trigger brownout reset
3. Upon powering up the device, the device gets stuck in Joint Download Boot Mode whereas the Boot Pin

### Debug Logs.

```plain
I (42) boot: ESP-IDF v5.2.2-dirty 2nd stage bootloader
I (42) boot: compile time Jun 26 2024 12:40:45
I (43) boot: Multicore bootloader
I (46) boot: chip revision: v0.2
I (50) boot.esp32s3: Boot SPI Speed : 80MHz
I (55) boot.esp32s3: SPI Mode       : DIO
I (59) boot.esp32s3: SPI Flash Size : 4MB
I (64) boot: Enabling RNG early entropy source...
I (70) boot: Partition Table:
I (73) boot: ## Label            Usage          Type ST Offset   Length
I (80) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (88) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (95) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (103) boot:  3 ota_0            OTA app          00 10 00010000 00180000
I (110) boot:  4 ota_1            OTA app          00 11 00190000 00180000
I (118) boot: End of partition table
I (122) esp_image: segment 0: paddr=00010020 vaddr=3c0e0020 size=48880h (297088) map
I (164) esp_image: segment 1: paddr=000588a8 vaddr=3fc9f200 size=050dch ( 20700) load
I (167) esp_image: segment 2: paddr=0005d98c vaddr=40374000 size=0268ch (  9868) load
I (171) esp_image: segment 3: paddr=00060020 vaddr=42000020 size=d5d14h (875796) map
I (276) esp_image: segment 4: paddr=00135d3c vaddr=4037668c size=18aech (101100) load
I (292) esp_image: segment 5: paddr=0014e830 vaddr=50001000 size=00004h (     4) load
I (292) esp_image: segment 6: paddr=0014e83c vaddr=600fe000 size=00070h (   112) load
I (308) boot: Loaded app from partition at offset 0x10000
I (309) boot: Disabling RNG early entropy source...
I (320) cpu_start: Multicore app
I (330) cpu_start: Pro cpu start user code
I (330) cpu_start: cpu freq: 160000000 Hz
I (330) cpu_start: Application information:
I (333) cpu_start: Project name:     dcarb_d100
I (338) cpu_start: App version:      v1.3.5-rc5-2-gffbe7e9-dirty
I (345) cpu_start: Compile time:     Jun 26 2024 12:40:19
I (351) cpu_start: ELF file SHA256:  144429dac...
I (356) cpu_start: ESP-IDF:          v5.2.2-dirty
I (362) cpu_start: Min chip rev:     v0.0
I (366) cpu_start: Max chip rev:     v0.99
I (371) cpu_start: Chip rev:         v0.2
I (376) heap_init: Initializing. RAM available for dynamic allocation:
I (384) heap_init: At 3FCAB358 len 0003E3B8 (248 KiB): RAM
I (389) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (396) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (402) heap_init: At 600FE070 len 00001F78 (7 KiB): RTCRAM
I (409) spi_flash: detected chip: generic
I (413) spi_flash: flash io: dio
W (417) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (427) sleep: Configure to isolate all GPIO pins in sleep state
I (434) sleep: Enable automatic switching of GPIO sleep configuration
I (441) coexist: coex firmware version: d96c1e51f
I (447) coexist: coexist rom version e7ae62f
I (452) main_task: Started on CPU0
I (462) main_task: Calling app_main()
E (462) BOOT: Download boot mode is enabled (persistent).
I (462) main: FIRMWARE VERSION: v1.3.0, HW_VERSION: REV-F_3.2
I (502) NVS: S_nvs semaphore init
I (532) Simple-ADC: ADC Unit 0 calibration scheme version is Curve Fitting
I (532) Simple-ADC: ADC Unit 0 Calibration Success
I (532) Simple-ADC: ADC Unit 1 calibration scheme version is Curve Fitting
I (542) Simple-ADC: ADC Unit 1 Calibration Success
I (542) gpio: GPIO[10]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (552) gpio: GPIO[20]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (562) PID: Hbridge ADCs initialized
W (562) main: Not a ULP-RISC-V wakeup (cause = 0), initializing it
I (632) iotcore-service: Using DEV Server MQTT Credentials
I (632) iotcore_events: iotcore event loop started
I (632) NVS: S_nvs semaphore init
I (722) pp: pp rom version: e7ae62f
I (722) net80211: net80211 rom version: e7ae62f
I (732) wifi:wifi driver task: 3fcbf584, prio:23, stack:6656, core=0
I (732) wifi:wifi firmware version: 3e0076f
I (732) wifi:wifi certification version: v7.0
I (732) wifi:config NVS flash: enabled
I (732) wifi:config nano formating: disabled
I (742) wifi:Init data frame dynamic rx buffer num: 32
I (742) wifi:Init static rx mgmt buffer num: 5
I (752) wifi:Init management short buffer num: 32
I (752) wifi:Init dynamic tx buffer num: 32
I (762) wifi:Init static tx FG buffer num: 2
I (762) wifi:Init static rx buffer size: 1600
I (762) wifi:Init static rx buffer num: 10
I (772) wifi:Init dynamic rx buffer num: 32
I (772) wifi_init: rx ba win: 6
I (782) wifi_init: tcpip mbox: 32
I (782) wifi_init: udp mbox: 6
I (782) wifi_init: tcp mbox: 6
I (792) wifi_init: tcp tx win: 5760
I (792) wifi_init: tcp rx win: 5760
I (802) wifi_init: tcp mss: 1440
I (802) wifi_init: WiFi IRAM OP enabled
I (802) wifi_init: WiFi RX IRAM OP enabled
I (812) Wifi Manager: Setting Wifi AP mode
I (812) Wifi Manager: ap_config.ap.channel = 3
I (822) Wifi Manager: ap_config.ap.authmode = WIFI_AUTH_OPEN
I (832) Wifi Manager: wifi_init_softap finished.SSID:IOTCORE_348518B10588C2R2 password: pin:0527
W (832) Connectivity: Starting Wifi
I (842) wifi:Set ps type: 2, coexist: 0

I (842) phy_init: phy_version 670,b7bc9b9,Apr 30 2024,10:54:13
I (902) wifi:mode : sta (34:85:18:b1:05:88) + softAP (34:85:18:b1:05:89)
I (902) wifi:enable tsf
I (902) wifi:Total power save buffer number: 16
I (902) wifi:Init max length of beacon: 752/752
I (902) wifi:Init max length of beacon: 752/752
I (912) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
I (912) MQTT: MQTT_EVENT_BEFORE_CONNECT
I (922) gpio: GPIO[9]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (922) RTC: Boot count: 1
E (922) esp-tls: couldn't get hostname for :dcarb-broker-dev.australiaeast.cloudapp.azure.com: getaddrinfo() returns 202, addrinfo=0x0
I (932) RTC: timezone GMT0

E (952) transport_base: Failed to open a new connection: 32769
I (952) RTC: Time is not set yet. Connecting to WiFi and getting time over NTP.
E (962) mqtt_client: Error transport connect
I (962) RTC: timezone GMT0

E (972) MQTT: MQTT_EVENT_ERROR
I (972) RTC: The current date/time in Pakistan is: Thu Jan  1 00:00:00 1970 (0)
E (982) MQTT: MQTT_EVENT_DISCONNECTED
I (932) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1002) gpio: GPIO[42]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1012) gpio: GPIO[48]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (1022) gpio: GPIO[1]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1032) gpio: GPIO[47]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1042) gpio: GPIO[15]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3
I (1042) gpio: GPIO[4]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3
I (1052) main: Hardware Capabilities: ldr 1, led 1, invert waterlvl 0
I (1062) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:3
E (1072) main: Main Bus Vo�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xf (BROWNOUT_RST),boot:0x0 (DOWNLOAD(USB/UART0))
waiting for download
```

### More Information.

I have confirmed that the voltage on the GPIO0 is 3.3V during bootup

granxavi, 2024-06-26:
Have you tried pushing the boot button, hold your finger on it, then push the reset buton and release the reset button without releasing the boot? Or something like that? Sometimes I do that and works again.

ESP-Marius, 2024-06-27:
Is anything connected to GPIO0?

What happens if you reset it again when it enters download mode? Will it enter download mode again or do an SPI boot?

Areeb-Mufti, 2024-06-27:
I have a custom PCB design with no buttons connected to Enable and GPIO0

The enable pin is pulled high via a 10k resistor along with 1uF capacitor

The GPIO0 is left unconnected as it is already pulled high internally

My ESP32S3 is powered through a backup coin cell CR2477

I am continuously monitoring the main power line through ADC and when it drops below a threshold, I switch the controller in deep sleep mode while powered by the backup coin cell.

Sometimes, the main power drops rapidly and the controller does not have enough time to go into deep sleep mode and brownout occurs
About 90% of the time when a brownout is triggered, the device boots up as intended i.e. SPI boot mode and starts operating. Rarely the device gets stuck into the download boot mode and this issue is not resolved until reset the ESP32S3 through the enable pin and power cycle the whole thing. If I only power cycle, ESP32S3 continuously boots in download boot mode

From the boot mode, I have checked that the 1st Stage bootloader is detecting GPIO0 as low and going into download boot mode
I have tried using a external pullup resistor with no effect

I believe the issue is somewhere in the 1st stage bootloader where it is misreading the GPIO0 state

So far I have been able to replicate this issue in 5 different devices on my side so I believe this is not a manufacturing fault

ESP-Marius, 2024-06-27:
> I believe the issue is somewhere in the 1st stage bootloader where it is misreading the GPIO0 state

The strapping pins are latched by HW and the first stage bootloader simply reads this register (`GPIO_STRAPPING`) so there shouldnt be much room for software errors.

I'll take this issue to our digital team and see if they can explain what is happening

chipweinberger, 2024-06-27:
similar issue:

https://github.com/espressif/esptool/issues/970

chipweinberger, 2024-06-27:
and for completeness:

https://github.com/espressif/esp-idf/issues/13783

ihor-ziuba, 2024-06-29:
> Sometimes, the main power drops rapidly and the controller does not have enough time to go into deep sleep mode and brownout occurs

I had the same issue with our custom controller, but brownout and "sticking" occured in the moment of switching between main power supply and reserve battery. The solution was pretty straightforward, just increase capacitor on EN pin up to 10 uF, so esp32s3 will be in reset state a bit longer

o-marshmallow, 2024-09-25:
In my opinion, the issue is with the power supply: the CR2477 battery.

No matter what their capacity (200mAh, 1000mAh, ...), these coin batteries have a very low maximum current peak, ~20mAh, you have to check the datasheet of yours. However, on boot the ESP32 series chips, including the S3, are very likely to have a much higher current peak (100~300mAh). 
As @ESP-Marius said, the strapping pins are latched in hardware, but since the power supply is unstable/not in range, the hardware is not guaranteed to work properly.

Have a try power your S3 with a stable power supply and see if this issue is still here

gm-jiang, 2024-09-25:
@Areeb-Mufti Can you please connect the gpio0 and gpio46 pins to an oscilloscope and then reproduce this issue on your ```Custom Board``` and observe the waveforms (levels) of these two pins at the same time.

gm-jiang, 2024-10-31:
@Areeb-Mufti  If there are no more issues, we will close this ticket.

closed 2024-11-08 by Sherry616