espressif/esp-idf #14083: IRAM is 100% used ESP32-S3

granxavi, 2024-06-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi! I am using esp32-s3 16MR8, IDF 5.2.2 IDE 3.0.0 
I just noticed that after building my project, the console says that I am using 100% of the static IRAM.
I thought it was my code or my sdkconfig, so I ran the hello_world example from expressiff without any change in code or configuration, and still, it says the same.
Am I missing something? Should I worry about it in my production project? Is it a bug?

```
Build complete (0 errors, 0 warnings): C:\User\hello_world\build
C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe C:\Espressif\frameworks\esp-idf-v5.2.2\tools\idf_size.py C:/User/hello_world/build/hello_world.map
Total sizes:
Used static IRAM:   16383 bytes (      1 remain, 100.0% used)
      .text size:   15356 bytes
   .vectors size:    1027 bytes
Used stat D/IRAM:   55523 bytes ( 290333 remain, 16.1% used)
      .data size:   10668 bytes
      .bss  size:    2248 bytes
      .text size:   42607 bytes
Used Flash size :  161315 bytes
           .text:  116083 bytes
         .rodata:   44976 bytes
Total image size:  230973 bytes (.bin may be padded larger)
```

ESP-Marius, 2024-06-27:
@granxavi On S3 there is 32KB of IRAM (only accessible on the instruction bus, not data bus) intended for the cache,  when `CONFIG_ESP32S3_INSTRUCTION_CACHE_16KB` is selected the leftover 16 KB can be used for IRAM.

This is what you see as being 100% used. But the generic D/IRAM which can be used as both IRAM and DRAM is what you actually should be looking at. Here in your hello world project it is only using 16.1% of it.

fhrbata, 2024-06-27:
Hello @granxavi,

@ESP-Marius is correct. The memory usage report from `esp-idf-size` is closely tied to specific chip memory types and layout. On the ESP32-S3, there are various memories including SRAM0 and SRAM1. The memory address range for accessing SRAM1 immediately follows the address range for accessing SRAM0. While SRAM0 can only be accessed via the instruction bus (reported as IRAM), SRAM1 can be accessed via both the data and instruction buses (reported as D/IRAM). For further details, please refer to the ESP32-S3 Technical Reference Manual available at https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf.

Depending on the configuration (`CONFIG_ESP32S3_INSTRUCTION_CACHE_(16|32)KB`), part or all of SRAM0 can be used for cache. If not fully used for cache, it is made available for code. This means that parts of both SRAM0 and SRAM1 are used for code, as the address space is continuous. Note that SRAM1 can be accessed via both instruction and data buses. In the `esp-idf-size` report, the code spans two memory types: IRAM in IRAM0 and DIRAM from IRAM1. Since these are two different memory types, they are reported separately, leading to the 100% usage of IRAM.

This is related to the recent fix https://github.com/espressif/esp-idf-size/commit/88a8b132e1b9eb8f418e4155252eab336e7db871, which aims to correctly handle situations involving DIRAM in the legacy version of `esp-idf-size` used in ESP-IDF prior to v5.3. Unfortunately, the legacy version displays memory types in a fixed order, with the full IRAM shown first for esp32s3, which could be confusing. This should be better handled in the refactored version, which lists memory types based on usage and allows different kinds of sorting. For your information, the refactored version can be enabled for ESP-IDF versions prior to v5.3 by setting the `ESP_IDF_SIZE_NG` environmental variable.

Thank you for your feedback!

closed 2024-06-27 by ESP-Marius