espressif/esp-idf #14086: Problems with RMT reception when Wi-Fi STA mode in use

mjdswan, 2024-06-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.2.2

### Espressif SoC revision.

ESP32C6 v0.0

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom Board

### Power Supply used.

Battery

### What is the expected behavior?

Our application requires coexistence between Wi-Fi STA mode and RMT (RX only).
We are using esp-rainmaker and IR remote to control LED lights driven by LEDC PWM driver.
We want the IR remote to also be usable when the Wi-Fi STA mode is active.

### What is the actual behavior?

With the Wi-Fi function disabled, the IR reception is flawless.
I have used the example from **esp-idf/tree/master/examples/peripherals/rmt/ir_nec_transceiver**, and simply disabled the TX functions.  It works 100% perfectly with an off-the-shelf IR remote.

However, as soon as Wi-Fi is active, the reception becomes very unreliable, I assume because of timing issues.  I can see from the logging that usually there is only a single bit error in the received codes, but it is fairly consistently like that when Wi-Fi is enabled.

I saw there is another issue opened about using RMT to drive LEDs while using Wi-Fi, but the suggestion to use the second core is not helpful here because the C6 is only single core.

I tried to increase the **rx_channel_cfg.mem_block_symbols** to 96, it seemed to help a little, but not enough.

Can you suggest any way around this issue?

### Steps to reproduce.

Enable Wi-Fi STA mode (esp-rainmaker) + RMT RX example.

[ir_nec_transceiver_main.zip](https://github.com/user-attachments/files/16008007/ir_nec_transceiver_main.zip)

suda-morris, 2024-06-27:
How long is the RMT packet you want to receive?

If you only use one RMT channel, then try `rx_channel_cfg.mem_block_symbols = 192;` to use all the memory blocks in the RMT peripheral.

mjdswan, 2024-06-27:
@suda-morris, thanks for the quick reply.

The transmitting remote sends 16-bit address and 16-bit command, so the NEC RX example was set up perfectly.
I'm only using one RX channel

I tried setting `rx_channel_cfg.mem_block_symbols = 192;` but on calling `rmt_new_rx_channel` I get:
```
E (847) rmt: rmt_rx_register_to_group(140): no free rx channels
E (847) rmt: rmt_new_rx_channel(215): register channel failed
```
followed by an abort() due to unhandled error 0x105 returned.  Is there a formula between command lengths and the `mem_block_symbols` I need to follow?

suda-morris, 2024-06-27:
Ah sorry, my bad. Since the only RMT channel 2-3 can be used for RX, so actually it can only use **2** memory clock at most. That is **96** (on esp32c6).

If the protocol is NEC, I think 2 memory block, which contains 96 words, should be sufficient for a NEC packet. The driver should not goes into the interrupt handler too frequently. Weird.

mjdswan, 2024-06-28:
OK thanks, as part of trying the `RMT_ISR_IRAM_SAFE`, I discovered I had commented a line which warned of an unknown NEC frame.  What is actually happening is that something is generating spurious RMT RX noise while the Wi-Fi is up, I can see it after un-commenting the log.  It may be hardware related, so I will check it some more, but thanks again.

closed 2024-06-28 by mjdswan