espressif/esp-idf #14091: esp-idf编译所有的usb ncm例子，usb枚举都不对

linhuikui, 2024-06-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1和v5.2.2

### Espressif SoC revision.

ESP32S2

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32s2

### Power Supply used.

USB

### What is the expected behavior?

[image]

### What is the actual behavior?

[image]

### Steps to reproduce.

使用example目录中的network/sta2eth和peripherals\usb\device\tusb_ncm都不行，都不能正常枚举为ncm

Alvin1Zhang, 2024-06-27:
Thanks for reporting.

roma-jam, 2024-06-27:
Hi @linhuikui, 

To understand the situation better: 

1. Could you provide the debug output log from the device, after uploading the firmware?
2. Is the ESP32 configured with correct ssid? Device should be connected to the WiFi (ssid could be configured via menuconfig, under the `Example Configuration`) 
3. Is there any information changed in device manager, after you insert the device into usb port? 

Thanks.

linhuikui, 2024-06-28:
@roma-jam
[esp32s2_log.txt](https://github.com/user-attachments/files/16023083/esp32s2_log.txt)

这是模块输出的log，我只进去配置了SSID和密码，其它任何东西都没修改；win10设备管理器有图中设备
[image]

roma-jam, 2024-07-08:
Hi @linhuikui,

thanks for additional information.

1. How the USB wire connection is established with the dev kit? Please, check the connection of the USB wire and power supply. Information can be found here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/usb_device.html#:~:text=self%2Dpowered%20devices-,Hardware%20Connection,to%20standard%2DA%20plug).
2. Before you will make any changes in the project, could you run `idf.py set-target esp32s2` again in the example folder and share versions of the components, currently being used during the build? It should look like (maybe versions would be different):
```
-- Building ESP-IDF components for target esp32s2
Processing 3 dependencies:
[1/3] espressif/esp_tinyusb (1.4.4)
[2/3] espressif/tinyusb (0.15.0~10)
[3/3] idf (5.2.2)
```
3. After step 2 (if the versions would be different), could you update the components, by resolving the dependencies again? It could be easily done by removing the managed_component folder from the example. After setting the target via `idf.py set-target esp32s2` new components should be downloaded automatically. 

Thanks.

roma-jam, 2024-07-09:
Hi @linhuikui, 

On Windows, there is no automatically assignment of a NCM driver to USB CDC NCM interface.
To assign the driver to the device, additional steps should be taken (such as manual driver assignment).

Please, refer to the comments on this question for additional information and detailed description of the additional steps: https://learn.microsoft.com/en-us/answers/questions/52386/usb-cdc-ncm-devices-not-assigned-to-usbncm-driver

linhuikui, 2024-07-16:
Hi @roma-jam
不止NCM驱动不能用，选择RNDIS也没有用，从串口调试输出的信息来看，应该是自动配置工具，没有把NCM或RNDIS的描述符信息，编译到程序中。
对于NCM，除了idf.py set-target esp32s2这个动作，其它配置什么都没有修改；对于RNDIS，进入menuconfig选择了RNDIS，其它什么都没有修改。