espressif/esp-idf #14093: Can't use SD Card with PSRAM in ESP32

josesimoes, 2024-06-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.4

### Espressif SoC revision.

ESP32-D0WDQ6 (revision v1.0)

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32 Wrover kit

### Power Supply used.

USB

### What is the expected behavior?

Using SD Card sample successfully.

### What is the actual behavior?

Sample fails to run.

### Steps to reproduce.

When executing the sample out-of-the-box everything runs fine.

After enabling SPIRAM by adding 
```
CONFIG_SPIRAM=y
CONFIG_SPIRAM_BOOT_INIT=y
CONFIG_SPIRAM_IGNORE_NOTFOUND=y
```

to the sdkconfig.defaults file, rebuilding and re-flashing, I get a timeout error.

Attached the verbose log for reference.
[idf log.txt](https://github.com/user-attachments/files/16012363/idf.log.txt)

Here's the relevant part:

```
E (3340) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
D (3350) sdmmc_init: sdmmc_card_init: sdmmc_init_ocr returned 0x107

E (3360) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
V (3360) intr_alloc: esp_intr_free: Disabling int, killing handler
E (3370) example: Failed to initialize the card (ESP_ERR_TIMEOUT). Make sure SD card lines have pull-up resistors in place.
I (3380) main_task: Returned from app_main()
```

I've made this change to double check this exact same issue we're seeing in .NET nanoFramework, following the migration to IDF v5.x.

### Debug Logs.

```plain
[idf log.txt](https://github.com/user-attachments/files/16012363/idf.log.txt)
```

nopnop2002, 2024-07-01:
I don't have an ESP32 with PSRAM, but this might be related.

https://github.com/espressif/esp-idf/issues/13971

haberturdeur, 2024-07-10:
> I don't have an ESP32 with PSRAM, so I can't confirm, but this may be related.
> 
> #13971

That regression was introduced in a later version, so `v5.1.4` should be unaffected. It might be something else...

josesimoes, 2024-07-25:
Turns out that this is indeed working with v5.1.4.
Thanks everyone!

closed 2024-07-25 by josesimoes